dsmrreader/dsmr-reader #220: Meer statistieken

Varsame, 2017-01-06:
Beste Dennis, 
Mooi project heb je inmiddels opgebouwd, ik gebruik het sinds een paar maanden met veel plezier! 
Een suggestie die ik je graag wil doen is om wat meer leuke statistieken te berekenen uit de data die DSMR verzamelt.  De "Statistieken" pagina bevat nu eigenlijk alleen het hoogst, laagst en meest voorkomende verbruik. Ik zat te denken aan gemiddeld verbruik per dag, maand, jaar etc. En ook hoogst, laagst, meest voorkomend, en dan per dag, week etc.. Een beetje zoals de temperatuur-cijfers in het Archief zeg maar. 
 Voor gasverbruik is het met graaddagen lastig om een voorspelling te doen voor een periode in de toekomst, maar voor electriciteit is dat wat minder aan de orde lijkt mij?

dennissiemensma, 2017-01-06:
Fijn om te horen dat je er zoveel plezier aan hebt.

Wat betreft je verzoeken. Ik denk dat daar zeker wel wat mee te doen is. In een eerder stadium was ik van plan om dit soort dingen onder de Trends-pagina te scharen, maar ik ben daar nog niet meer verder gekomen.

Ik zet dit ticket, voor nu, op de lijst van `1.7` om er dan naar te kijken. De eerstvolgende release (`1.6`) zal waarschijnlijk eerst focussen op een hoop redesign van de grafieken en data die beschikbaar is.

dennissiemensma, 2018-10-06:
Ik laat dit voorlopig voor wat het is. Het idee is tof, maar vrij complex om betrouwbaar uit te werken.

closed 2018-10-06 by dennissiemensma