dsmrreader/dsmr-reader #225: Keep track of meter serials in case of smart meter replacement

dennissiemensma, 2017-01-07:
There should be support for checking the source of a reading: it's meter. Because whenever someone's smart meter gets replaced, it would cause the app trouble in calculating the consumption.

I think the application should keep track of the device serials each time a reading is processed. But only in the form of 'last serial read'. 
And whenever it changes ity should do some magic or set some flag so it can be taken into account.

ThinkPadNL, 2017-01-07:
Isn't this a bit of a '[feature creep](https://en.wikipedia.org/wiki/Feature_creep)'? 😉 
How often do you expect a meter to be replaced? I requested a smart meter, my appartment still had the original meter from 1988 when i moved in a few years ago.

dennissiemensma, 2017-01-07:
Well yes and no. I've added it to the backlog three releases forward, so I won't get into this any time soon.

A colleague of mine had his new smart meter replaced after 6 months due to malfunction. But the most common situation will be when someone moves to another home or appartment, which might be more common.

So it's not important at this time, but I think it should be possible.

dennissiemensma, 2017-01-18:
Actually, it happened now to one of the users: #242

hsleewis, 2017-01-19:
I think this will actually happen more often than you think. Liander uses a lot of gasmeters that make an annoying pumping noise and they are replacing them all for Landys + Gyr meters that don't use blades to measure volume. The Landys + Gyr meters measure usage with a ultrasonic tone. 

You could make the app do it automatically based upon serial numbers. Or you can add a setting so a meter change can be added manually.

ThinkPadNL, 2017-01-23:
Don't forget that if you upload your gasmeter readings to Mindergas.nl, you also have to change something there:

> **Ik heb een nieuwe gasmeter gekregen. Hoe kan in dit aanpassen op de site?**
Navigeer naar 'Mijn Gasverbruik -> Meterstanden'. Klik op de link 'Nieuwe gasmeter' en vul de eindstand van de oude gasmeter, de beginstand van de nieuwe gasmeter, en de ingangsdatum in. Klik op 'Opslaan'.
https://mindergas.nl/faq

BBasss, 2018-02-02:
Hoi Dennis,

In de omgeving Amersfoort gaat Stedin nu beginnen met de gasmeterwissels ivm. onveilige meters ( Zelf spreken ze van het "niet voldoet aan onze hoge kwaliteitseisen".......  jaja
Ik weet niet hoeveel blije DSMR reader gebruikers zijn, maar eentje in ieder geval ;-)

Kan ik nog behulpzaam zijn met loggings oid? of heb je al een idee hoe je het kan oplossen.

dennissiemensma, 2018-05-02:
Ik laat deze feature voor wat het is. Het vrij veel werk om dit goed en sluitend in de applicatie te krijgen. Het detecteren van de meterwijziging is vrij eenvoudig. Echter het goed berekenen en weergeven van alle metingen bij zo'n wisseling valt tegen.
Wanneer iemand een andere meter krijgt, wijkt slechts één dag af, die handmatig recht te trekken is. Dat lijkt me voor nu afdoende.

closed 2018-05-02 by dennissiemensma