dsmrreader/dsmr-reader #226: Import of external MySQL data

balk77, 2017-01-07:
I have a year worth of P1 reading data in MySQL which I would like to import in dsmr-reader. Is there some kind of import function?

dennissiemensma, 2017-01-07:
Hi, there is currently no import function. But you might be able to import the data retroactivilty via the API, if you have the raw telegram data.

Could you provide more info about the exact format of the data?

balk77, 2017-01-07:
It is a fairly simple database in MySQL, single table. Below is the first entry of the database and the column names. A bit ugly, so a CSV is also attached. Can I construct a telegram from this?

p1_timestamp	15/01/16 21:04
p1_meterreading_in_1	3305,519
p1_unitmeterreading_in_1	kWh
p1_meterreading_in_2	3527,207
p1_unitmeterreading_in_2	kWh
p1_current_tariff	2
p1_current_power_in	0,26
p1_unit_current_power_in	kW
p1_current_switch_position	1
p1_powerfailures	0
p1_long_powerfailures	0
p1_long_powerfailures_log	
p1_channel_1_id	1
p1_channel_1_type_id	3
p1_channel_1_type_desc	Gas
p1_channel_1_timestamp	15/01/16 22:00
p1_channel_1_meterreading	2869,31
p1_channel_1_unit	m3
p1_channel_1_valveposition	0

[P1mysqldata.txt](https://github.com/dennissiemensma/dsmr-reader/files/691759/P1mysqldata.txt)

dennissiemensma, 2017-01-07:
Thanks for providing input. I've checked your example, but I'm affraid it's currently not possible to import this data. The source telegrams provides some additional fields that are required as well, and can't be reconstructed from the data you provided.

I think the best you can do is to either use the tool you've collected this data with, to view that year of data. Or to export it all as CSV, convert it to Excel and plot some graphs or stats there. 

But regarding to gas consumption you might have an extra option. You could register and connect an account with MinderGas.nl and insert the gas positions of your last year there as well. 
I did that myself as well for any data I collected *before using this project*, by simply retroactivily inserting the gas meter positions on that website, one for each month. It provides a bit insight.

dennissiemensma, 2017-01-07:
Still thanks for reporting your question here. I'd loved to provide a solution for this, but I'm sorry. :]

balk77, 2017-01-20:
Dennis, thanks for making this great product!

I am looking into importing MySQL data from an older database into the Postgresql database by constructing a telegram from the old data. But then I noticed that a telegram does not contain a date stamp. Does the dsmr-reader API allow for receiving telegrams with a date stamp?

That way, I can run a script to take data from the MySQL database, construct a telegram and send it to DSMR-reader via an API call. 🥇

dennissiemensma, 2017-01-20:
Hi, a telegram is the raw data read from the serial port. If you do not have that stored you can not reconstruct it. The current API was only meant for **remote dataloggers** sending the entire telegram.

I do however think that the API should be extended, at some point in the future, so you can just insert the data you want. Because the application [uses these fields](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.5/dsmr_datalogger/models/reading.py#L14), they should be able to be filled by such future API.

I will assign this to the backlog of a future release.

balk77, 2017-01-20:
Just to clarify what I think could be usefull.... :-)

Currently, the API receives its telegram via this:
`response = requests.post(
    'http://YOUR-DSMR-URL/api/v1/datalogger/dsmrreading',
    headers={'X-AUTHKEY': 'YOUR-DSMR-API-AUTHKEY'},
    data={'telegram': telegram_string},
)`

I think it is usefull to extend the API with a timestamp field, for instance like this. Syntax may be wrong.
`response = requests.post(
    'http://YOUR-DSMR-URL/api/v1/datalogger/dsmrreading',
    headers={'X-AUTHKEY': 'YOUR-DSMR-API-AUTHKEY'},
    data={'telegram': telegram_string , 'timestamp' : timestamp},
)`

balk77, 2017-01-27:
I noticed indeed; I have written a script that reads all 400000 measurements in the MySQL database, creates a telegram out of it and sends it to the API. I found out that sending "old" data to a running database causes the backend to complain; does it expect incoming readings to be newer than what is in the database? Workaround: import old readings in a fresh database (A) while database (B) registers new measurements on another computer, wait 15 hours for the import, export the database B to database A. Use (A) as master database.

I will share the script, although it is very specific for my situation.

dennissiemensma, 2017-01-28:
I'm not sure what it will do, but I think it won't work retroactively. Inserting old telegrams should be possible, but I'm not sure whether the background processes them as well.
It depends on what the exact error is the backend is 'complaining' about. You can check out the `logs/dsmrreader.log` file inside the project dir. It might be CRC checks, which can be disabled if needed.

If you have the possibility to run a separate/second instance of dsmrreader and it's database (VM on PC?), you can experiment easily by inserting directly into the `dsmr_datalogger_dsmrreading` database table.
When you have both old and new readings inserted (from recent backup + script), you can truncate these tables: `dsmr_consumption_electricityconsumption`, `dsmr_consumption_gasconsumption`, `dsmr_stats_daystatistics` and `dsmr_stats_hourstatistics`. Finally, set the `processed` value of all records in the `dsmr_datalogger_dsmrreading` to `False`.
The backend process should then start processing all readings again, from the start. 

*But you should only experiment with this when you have a separate environment to test with.*

dennissiemensma, 2017-05-04:
@balk77 in case you're still using dsmrreading and are looking for a way to import the data. There is now a much easier way using the new API (#230), included in the latest release.

Please do note that I **do not know** whether importing historic data will be picked up by the data processor correctly. You might want to **stop** the ``dsmr_backend`` process until you've importer all data.

There also is a last resort [info buble in the docs](https://dsmr-reader.readthedocs.io/en/latest/api.html#id2) about this. I hope this solves your need for importing existing data.

closed 2017-05-04 by dennissiemensma

balk77, 2017-05-06:
Dennis, je bent een held! :-) bedankt voor alle moeite die je hier in steekt!