dsmrreader/dsmr-reader #227: UTC timezone naar CET timezone en daarop een query

dbmnl, 2017-01-08:
Ik heb in #196 een query gebruikt om de dagstanden die met het telegram tussen 08:00:00 en 08:00:10 binnen komen weer te geven. Vanmorgen kwam ik er echter achter dat om 08:30 er nog geen nieuwe meterstand te vinden was toen ik de query afvuurde. 
Ik ben de DB in gegaan en zie dat je de waardes altijd wegschrijft op UTC tijd. Geen probleem dacht ik, ik pas wel even de value aan naar CET en query er dan mee, maar ik kreeg nog steeds de waardes terug krijgen van 09:00... Dus heb ik een kolom toegevoegd om te kijken of ik nu gek ben maar ook al zeg ik in mijn where clause dat hij op de CET select moet queryen blijft hij maar terugkomen op de tijd van 09:00.... 

Ik heb al het 1 en ander uitgezocht op internet, maar ik kan geen duidelijk antwoord vinden hoe ik nu ervoor kan zorgen dat ik met 08:00 CET kan zoeken in de database in plaats van op de UTC waarde wat hij nu doet.

`sudo -u postgres psql dsmrreader -c "select distinct read_at::date, (read_at::time AT TIME ZONE 'CET') as tijd, delivered_1, delivered_2 from dsmr_consumption_electricityconsumption where (read_at::time AT TIME ZONE 'CET' between time '08:00:00' and '08:00:10') order by read_at::date desc;"`

`  read_at   |    tijd     | delivered_1 | delivered_2
------------+-------------+-------------+-------------
 2017-01-08 | 09:00:06+01 |     465.983 |     355.570
 2017-01-07 | 09:00:03+01 |     463.121 |     355.570
 2017-01-06 | 09:00:09+01 |     462.433 |     351.995
 2017-01-05 | 09:00:05+01 |     460.951 |     349.199
 2017-01-04 | 09:00:02+01 |     460.413 |     346.380
 2017-01-03 | 09:00:08+01 |     459.818 |     342.926
 2017-01-02 | 09:00:04+01 |     459.330 |     341.525
 2017-01-01 | 09:00:04+01 |     448.953 |     341.282
 2016-12-30 | 09:00:07+01 |     440.059 |     336.547
 2016-12-29 | 09:00:04+01 |     438.550 |     330.858
 2016-12-28 | 09:00:10+01 |     437.676 |     324.973
 2016-12-27 | 09:00:06+01 |     436.937 |     317.567
 2016-12-26 | 09:00:03+01 |     425.071 |     317.167
 2016-12-25 | 09:00:09+01 |     410.191 |     317.167
 2016-12-24 | 09:00:05+01 |     393.098 |     317.167
 2016-12-23 | 09:00:01+01 |     392.202 |     313.392
 2016-12-22 | 09:00:08+01 |     390.411 |     310.608
 2016-12-21 | 09:00:04+01 |     389.757 |     307.400
 2016-12-20 | 09:00:10+01 |     389.214 |     303.642
 2016-12-19 | 09:00:07+01 |     388.694 |     300.601
 2016-12-18 | 09:00:03+01 |     378.652 |     300.014
 2016-12-17 | 09:00:09+01 |     373.599 |     300.014
 2016-12-16 | 09:00:06+01 |     372.953 |     297.630
 2016-12-15 | 09:00:02+01 |     372.433 |     295.385
 2016-12-14 | 09:00:08+01 |     370.674 |     292.790
 2016-12-13 | 09:00:04+01 |     370.173 |     289.998
 2016-12-12 | 09:00:01+01 |     369.577 |     286.719
(27 rows)
`

dennissiemensma, 2017-01-08:
Ik heb even gekeken hoe Django dat doet. Je kunt in je verbinding naar de database aangeven in welke tijdzone je zit. Dit kan met de query: `set time zone 'Europe/Amsterdam'`.

Gecombineerd met de query:
```
sudo -u postgres psql dsmrreader -c "set time zone 'Europe/Amsterdam' ; select distinct read_at::date, read_at, delivered_1, delivered_2 from dsmr_consumption_electricityconsumption where read_at::time between time '08:00:00' and '08:00:10' order by read_at::date desc;"
```

Zie het tweede veld, `read_at`. in het resultaat. Als het goed is krijg je nu de goede gegevens terug. 

Werkt dat voor je?

dbmnl, 2017-01-08:
Jup werkt als een trein! Ik dacht al dat ik iets mistte. Ik dacht dat het omzetten van de values voldoende was, maar blijkbaar niet. Tnx voor je hulp weer Dennis :)

`sudo -u postgres psql dsmrreader -c "set time zone 'Europe/Amsterdam' ; select distinct read_at::date, delivered_1, delivered_2 from dsmr_consumption_electricityconsumption where read_at::time between time '08:00:00' and '08:00:10' order by read_at::date desc;"`

>   read_at   | delivered_1 | delivered_2
> ------------+-------------+-------------
>  2017-01-08 |     465.915 |     355.570
>  2017-01-07 |     462.994 |     355.570
>  2017-01-06 |     462.433 |     351.938
>  2017-01-05 |     460.951 |     349.139
>  2017-01-04 |     460.413 |     346.314
>  2017-01-03 |     459.818 |     342.866
>  2017-01-02 |     459.330 |     341.465
>  2017-01-01 |     448.887 |     341.282
>  2016-12-31 |     440.765 |     341.282
>  2016-12-30 |     440.059 |     336.462
>  2016-12-29 |     438.550 |     330.793
>  2016-12-28 |     437.676 |     324.899
>  2016-12-27 |     436.937 |     317.490
>  2016-12-26 |     424.675 |     317.167
>  2016-12-25 |     408.750 |     317.167
>  2016-12-24 |     392.944 |     317.167
>  2016-12-23 |     392.202 |     313.323
>  2016-12-22 |     390.411 |     310.548
>  2016-12-21 |     389.757 |     307.332
>  2016-12-20 |     389.214 |     303.576
>  2016-12-19 |     388.694 |     300.153
>  2016-12-18 |     378.570 |     300.014
>  2016-12-17 |     373.539 |     300.014
>  2016-12-16 |     372.953 |     297.569
>  2016-12-15 |     372.433 |     295.318
>  2016-12-14 |     370.674 |     292.649
>  2016-12-13 |     370.173 |     289.882
>  2016-12-12 |     369.577 |     286.649
> (28 rows)
>

closed 2017-01-08 by dbmnl