dsmrreader/dsmr-reader #229: Error in measurement of returned energy

KoenRutten, 2017-01-11:
Thanks a lot for your work on this project. I had a lot of fun installing and using my first Raspberry ever. 
Now I come across some measurements I can't explain: when we use our water kettle or oven, the graph for 'Recent electricity returned' gives an unexpected spike. I think that measurements of the first phase and measurements for energy returned are switched:  
- we don't have solar panels
- we have a 3-phase connection and use all of them 
- no energy consumption has been registered for phase l1  
- we hava a Kaifa (E0026? MA304C?)

Any suggestions on how to correct this?

[image]

dennissiemensma, 2017-01-11:
Hi, thank you for reporting this issue. I've not seen such thing before.

Could you post a dump here of one of the telegrams read? Please make sure to **remove** any serials of the devices used.

`tail -n 50 /var/log/supervisor/dsmr_datalogger.log`

The serial numbers usually are lines with around 30 digits and start with either `0-0:96.1.1` or `0-1:96.1.0`.

Depending on your power installation and devices used, some phases are only used on full power. I.e. while having all the electric burners in your kitchen active simultaneously (at high power). Did you try that as well?

KoenRutten, 2017-01-12:
I've attached two files with. The first is during 'normal' power consumption. The  second was read while our oven and water kettle was switched on. 
[slimmemeter_stroom_laag.txt](https://github.com/dennissiemensma/dsmr-reader/files/702620/slimmemeter_stroom_laag.txt)
[slimmemeter_stroom_hoog3.txt](https://github.com/dennissiemensma/dsmr-reader/files/702621/slimmemeter_stroom_hoog3.txt)
[image]

dennissiemensma, 2017-01-12:
Thanks for your data as it's **very helpful**.

This is from your high usage snapshot:
```
# currently used (!?)
1-0:1.7.0(00.000*kW)

# currently returned (!?)
1-0:2.7.0(00.409*kW)

# phases L+ 1, 2, 3
1-0:21.7.0(00.000*kW)
1-0:41.7.0(02.008*kW)
1-0:61.7.0(00.007*kW)

# phases L- 1, 2, 3
1-0:22.7.0(02.425*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
```

So It's basicly subtracting consumption somehow. Also according to the telegram, you meter positions at the time of reading were:

```
1-0:1.8.1(000410.263*kWh)  # low tariff consumed
1-0:1.8.2(000510.152*kWh)  # high tariff consumed

1-0:2.8.1(000037.034*kWh)  # low tariff returned
1-0:2.8.2(000071.248*kWh)  # high tariff returned
```

This basically means that you are having 'free' power whenever you kettle is active. So either the data is corrupt or the smart meter is broken or wrongly connected.

**Could you check manually on your smart meter whether the meter positions above are correct?** You can run the `tail` command again to have those values refreshed.

I've attached the DSMR-specs as well, below. You can lookup the line codes above to double check, but I think I'm reading it correctly...

[Dutch Smart Meter Requirements v5.0.2 Final P1.pdf](https://github.com/dennissiemensma/dsmr-reader/files/702741/Dutch.Smart.Meter.Requirements.v5.0.2.Final.P1.pdf)

dennissiemensma, 2017-01-12:
And for your reference, this is how it looks at my home, also no solar panels and three phases:

```
# currently used
1-0:1.7.0(07.057*kW)

# currently returned
1-0:2.7.0(00.000*kW)

# phases L+ 1, 2, 3
1-0:21.7.0(01.852*kW)
1-0:41.7.0(01.938*kW)
1-0:61.7.0(03.265*kW)

# phases L- 1, 2, 3
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
```

KoenRutten, 2017-01-12:
Haha! The readings for the total of low/high tariff returned correspond exactly with the readings on our "slimme" meter! I've found out that it's our oven which is "producing" electricity.
... I'll be baking a lot the coming months.

dennissiemensma, 2017-01-12:
Well that is kinda hilarious indeed! It's up to you about what to do with it (regarding reporting it to your power grid operator).

**Please note that**:
- Both your energy supplier and power grid operator **can** read your smart meter remotely **any time** (even though they say they do it only 6 times a year). It's unknown whether they perform any cross checks regarding *power delivered and power paid for*. But, I think actually they do at some point, for example to find any weed plantations using illegal power taps.

- There is a big chance that whenever they find out, you'll most likely have to pay for that consumption retroactively. The 'difference' at the moment is only around 100 kWh, which is about 20 Euro's. But that difference *will* increase each day.

So I'm not telling you what to do, but please mind the notes above. :]

I'll close this issue for now, as it's not caused by a bug in this project. Still thanks for reporting it though.

closed 2017-01-12 by dennissiemensma

KoenRutten, 2017-01-13:
I will contact Stedin or our energy supplier.

KoenRutten, 2017-01-13:
Contacted Stedin: they confirm that our Slimme meter is probably not connected in the correct way. They will come over and fix it in the near future.