dsmrreader/dsmr-reader #233: Piek+dal Tarief  combineren in overzichten

Roeliooo, 2017-01-16:
Op de archief /trend/ etc pagina totaalverbuik electra optelsom Piek/Dal tarief weergeven. Nu Niet Duidelijk Hoeveel het Totaalverbruik Per Dag Is.

dennissiemensma, 2017-01-17:
Bedankt voor je ticket. In principe kun je het wel zelf optellen (lage getallen), maar ik zal kijken of ik een totaal telletjes kan maken bij de redesign van de pagina's.

Roeliooo, 2017-01-19:
Zou heel fijn zijn :)

Helloitsme, 2017-04-01:
Zou wellicht ook een idee zijn om in plaats van 2 lijnen in de grafiek te gebruiken, een stacked column grafiek te maken, dan zie je zowel de details als het totaal goed. Vooral op de archief pagina, waar je op maand niveau zowel dal als piek tarief hebt zou dat handig zijn

dennissiemensma, 2017-04-02:
Bedankt voor je suggestie, wellicht is dat wel een goede optie. Ik zal tzt meerdere formaten uitproberen en kijken welke het duidelijkste is.

dennissiemensma, 2017-09-20:
Dit is voor een deel opgepakt in de pull request #341 van helmo. Het hervormen van de interface is een apart iets, waar ik sowieso wel een release aan kwijt ben. Voor nu zou dit afdoende moeten zijn.

closed 2017-09-20 by dennissiemensma