dsmrreader/dsmr-reader #234: DSMR reader stop met meten na x tijd

eterpstra, 2017-01-17:
Hi Dennis,

Sinds ik de upgrade heb uitgevoerd en de database heb teruggezet, stopt DSMR met meten / uitlezen van de slimme meter. Je ziet dan bij het status overzicht ook dat de metingen niet lukken (alles wordt rood)

Als ik de Pi echter reboot, dan worden de metingen wel allemaal ingelaten die hij daarvoor niet heeft kunnen meten? Dat is dus wel erg raar..

Heb je enig idee wat dit kan zijn?

dennissiemensma, 2017-01-17:
- Welke versie draai je, de laatste?
- Wordt alles rood in de Status-pagina of alleen het onderdeel van metingen?
- Na hoeveel tijd 'stopt' die met meten?

- Wanneer hij stopt, wil je eens kijken in de logfiles? 

Dat laatste kan vrij makkelijk met `tail -f /var/log/supervisor/dsmr_datalogger.log`, die houdt de logfile in de gaten en toont telkens nieuwe regels.
Als de datalogger geen fouten laat zien, bekijk dan het backend-proces eens: `tail -f /var/log/supervisor/dsmr_backend.log`

eterpstra, 2017-01-18:
Ik zal het in de gaten houden. Het gekke is dat hij de oude data wel weer toevoegd aan DSMR Reader (nieuwste versie draai ik nu)

Dit is voor de reboot:

https://s30.postimg.org/luw1ul1a9/Capture1.png

Na reboot is alles ''tijdelijk'' weer goed:

https://s27.postimg.org/uptcm1awj/Capture2.png

 tail -f /var/log/supervisor/dsmr_backend.log:

2017-01-18 07:24:15.163497+01:00: Calling backend services
Command completed. Sleeping for 5 second(s)...

2017-01-18 07:24:20.293949+01:00: Calling backend services
Command completed. Sleeping for 5 second(s)...

2017-01-18 07:24:25.433713+01:00: Calling backend services
Command completed. Sleeping for 5 second(s)...

eterpstra, 2017-01-18:
Ik zie dat hij sinds 9 uur gestopt is.  Dit is nu de output log:

`tail -f /var/log/supervisor/dsmr_backend.log
Command completed. Sleeping for 5 second(s)...

2017-01-18 09:00:04.006265+01:00: Calling backend services
 - Processed reading: 2017-01-18 08:59:01.761903+01:00.
 - Processed reading: 2017-01-18 08:59:12.256925+01:00.
 - Processed reading: 2017-01-18 08:59:22.751806+01:00.
 - Processed reading: 2017-01-18 08:59:33.246936+01:00.
 - Processed reading: 2017-01-18 08:59:43.739674+01:00.
 - Processed reading: 2017-01-18 08:59:54.233577+01:00.
 - Performing temperature reading at Buienradar.
`

dennissiemensma, 2017-01-18:
- Ok, maar wat is de output erna dan? Of houdt het hele process ermee op? 
- Wat zie je met `sudo supervisorctl status`, draait alles nog?
- Alle fouten worden hier gelogd, wil je kijken of daar iets in staat: `sudo su - dsmr` en dan `cat logs/django.log`

Het lijkt erop dat de `datalogger` en `webinterface` het goed doen, maar het `backend` proces hangt of stopt of iets. De `datalogger` schrijft alleen metingen weg en de applicatie verwerkt ze dan met `backend` proces (dat zijn die 'onverwerkte metingen'). En die kunnen met terugwerkende kracht gedaan worden (by design), dus daarom zie je dit gedrag. Alleen zou ik graag willen weten wat er precies gebeurt met het `backend` proces. 

Je kunt ook nog de hele log bekijken in `/var/log/supervisor/dsmr_backend.log`.

eterpstra, 2017-01-18:
Alle supervisor scripts hebben allemaal de status RUNNING. Dit lijkt dus goed.

django.log:

```
[2017-01-18 00:15:32,270] WARNING @ base | Not Found: /clientaccesspolicy.xml
[2017-01-18 02:17:08,779] ERROR @ exception | Invalid HTTP_HOST header: '/var/tmp/gunicorn--dsmr_webinterface.socket:'. The domain name provided is not valid according to RFC 1034/1035.
[2017-01-18 02:57:49,268] ERROR @ exception | Invalid HTTP_HOST header: '/var/tmp/gunicorn--dsmr_webinterface.socket:'. The domain name provided is not valid according to RFC 1034/1035.
[2017-01-18 04:42:21,944] WARNING @ base | Not Found: /w00tw00t.at.blackhats.romanian.anti-sec:)
[2017-01-18 04:42:22,164] WARNING @ base | Not Found: /phpMyAdmin/scripts/setup.php
[2017-01-18 04:42:22,390] WARNING @ base | Not Found: /phpmyadmin/scripts/setup.php
[2017-01-18 04:42:22,613] WARNING @ base | Not Found: /pma/scripts/setup.php
[2017-01-18 04:42:22,833] WARNING @ base | Not Found: /myadmin/scripts/setup.php
[2017-01-18 04:42:23,059] WARNING @ base | Not Found: /MyAdmin/scripts/setup.php
[2017-01-18 04:42:23,277] WARNING @ base | Not Found: /PHPMYADMIN/scripts/setup.php
[2017-01-18 04:42:23,500] WARNING @ base | Not Found: /mysqladmin/scripts/setup.php
[2017-01-18 04:42:23,723] WARNING @ base | Not Found: /SQL/scripts/setup.php
[2017-01-18 13:55:40,657] ERROR @ exception | Invalid HTTP_HOST header: '/var/tmp/gunicorn--dsmr_webinterface.socket:'. The domain name provided is not valid according to RFC 1034/1035.
[2017-01-18 17:35:10,805] WARNING @ base | Not Found: /script
[2017-01-18 17:35:10,888] WARNING @ base | Not Found: /jenkins/script
[2017-01-18 17:35:21,038] WARNING @ base | Not Found: /jmx-console
[2017-01-18 17:35:21,303] WARNING @ base | Not Found: /manager/html
[2017-01-18 17:35:21,529] WARNING @ base | Not Found: /msd
[2017-01-18 17:35:22,651] WARNING @ base | Not Found: /mySqlDumper
[2017-01-18 17:35:22,731] WARNING @ base | Not Found: /msd1.24stable
[2017-01-18 17:35:22,806] WARNING @ base | Not Found: /msd1.24.4
[2017-01-18 17:35:22,882] WARNING @ base | Not Found: /mysqldumper
[2017-01-18 17:35:23,967] WARNING @ base | Not Found: /MySQLDumper
[2017-01-18 17:35:24,069] WARNING @ base | Not Found: /mysql
[2017-01-18 17:35:24,145] WARNING @ base | Not Found: /sql
[2017-01-18 17:35:24,221] WARNING @ base | Not Found: /phpmyadmin
[2017-01-18 17:35:24,304] WARNING @ base | Not Found: /phpMyAdmin
[2017-01-18 17:35:24,390] WARNING @ base | Not Found: /mysql
[2017-01-18 17:35:24,477] WARNING @ base | Not Found: /sql
[2017-01-18 17:35:24,551] WARNING @ base | Not Found: /myadmin
[2017-01-18 17:35:27,634] WARNING @ base | Not Found: /phpMyAdmin-4.2.1-all-languages
[2017-01-18 17:35:27,712] WARNING @ base | Not Found: /phpMyAdmin-4.2.1-english
[2017-01-18 17:35:37,805] WARNING @ base | Not Found: /sqlite/main.php
[2017-01-18 17:35:40,882] WARNING @ base | Not Found: /SQLite/SQLiteManager-1.2.4/main.php
[2017-01-18 17:35:43,962] WARNING @ base | Not Found: /SQLiteManager-1.2.4/main.php
[2017-01-18 17:35:44,045] WARNING @ base | Not Found: /sqlitemanager/main.php
[2017-01-18 19:07:38,093] WARNING @ base | Not Found: /favicon.ico
[2017-01-18 19:08:08,688] WARNING @ base | Not Found: /check_proxy
```

Het dsmr_backend.log heb ik geupload naar Onedrive omdat deze 4mb groot is:

https://1drv.ms/u/s!AqT4eIHVip2fgZltLjE-QtUKNQ6KjA

dennissiemensma, 2017-01-18:
Ik zie deze melding rond het tijdstip die je aangeeft:

```
2017-01-18 09:00:04.006265+01:00: Calling backend services
... <knip> ...
Detected signal #15, exiting on next run...
Detected signal #15, exiting on next run...
```

Dat is een `SIGTERM` die (zoals ik lees op google) ofwel komt door de applicatie zelf (maar de log zegt niet) of door een externe `kill` van het proces, door bijvoorbeeld het OS.

- Zou je willen kijken wat supervisor precies aangeeft als die stopt? `sudo supervisorctl status`
- Wil je eens kijken of er nu (gevulde) error logs zijn in de supervisor log dir? ` ls -lh /var/log/supervisor/`

Ze heten iets als `dsmr_backend-stderr---supervisor-xxxxxx`.

eterpstra, 2017-01-18:
Ik zie dat er geen data meer is gevuld rond 20:00 uur. Als ik nu SV status bekijk:
```
dsmr_backend                     RUNNING    pid 786, uptime 3:15:25
dsmr_datalogger                  RUNNING    pid 783, uptime 3:15:25
dsmr_webinterface                RUNNING    pid 779, uptime 3:15:25
```

```
dsmr_backend-stderr---supervisor-onGEFK.log
dsmr_datalogger-stderr---supervisor-ICbNfm.log
dsmr_webinterface-stderr---supervisor-JNVN3M.log
```

Deze drie zijn allemaal leeg.  Bedoel je die?

dennissiemensma, 2017-01-18:
Ja die bedoel ik. Het is nu lastig te zeggen of het proces crasht of dat het OS hem om een of andere reden afschiet.

Op het moment dat hij weer gestopt is kun je denk ik dit weer doen. Hopelijk toont die dan een oorzaak:
- `sudo supervisorctl status`
- `sudo supervisorctl tail dsmr_backend`

Jeltel, 2017-01-19:
Ik had hier last van in combi met pi hole en één enthousiast apparaat. Dan stopte de webinterface er ook mee.
Is er niet een ander proces dat druk bezig is?

eterpstra, 2017-01-24:
Ik heb het inmiddels opgelost.. het probleem zit hem in het terugzetten van de oude database om 1 of andere reden.. Ik heb nu al een aantal dagen geen backup teruggezet en de metingen blijven nu netjes doorlopen...

Helaas pindakaas voor de oude data, maar ik kan er mee leven :)

dennissiemensma, 2017-01-24:
Oke goed en ook jammer om te horen. Maar als je er vrede mee hebt verder dan prima. Bedankt voor de terugkoppeling.

closed 2017-01-24 by dennissiemensma