dsmrreader/dsmr-reader #242: Verkeerde waarde bij wisselen gasmeter

hsleewis, 2017-01-18:
Mijn gasmeter is zojuist verwisseld en de nieuwe staat weer op 0, DSMR-Reader ziet nu een verbruik van -553.10 vandaag.

Het zou mooi zijn als je aan kan geven wat de uiteindelijke stand van de oude meter was en wanneer hij verwisseld is zodat de waardes kloppen.

dennissiemensma, 2017-01-18:
Bedankt voor het melden van dit issue. Helaas heb ik er niet direct een oplossing voor. De situatie had ik al wel voorzien in #225, maar dus nog niet opgelost. 
Ik ga deze situatie in dat ticket oplossen, dus wellicht wil je die eventueel volgen. Zolang alle metingen bewaard blijven is dit later altijd nog met terugwerkende kracht op te lossen, al begrijp ik dat de situatie vervelend is.

Zou je wellicht willen kijken of die vanaf morgen weer 'normale' dagstanden weergeeft? Dan heb je wellicht voor de andere dagen weer 'normaal' inzicht, voor nu.

hsleewis, 2017-01-19:
Thanks! Hij geeft vandaag wel weer normale dagstanden en kosten aan.

dennissiemensma, 2017-01-19:
Top, dat is goed om te horen. 

Ik heb zojuist een nieuwe patch uitgebracht waardoor je (en anderen later met hetzelfde probleem) zelf de statistieken kan aanpassen. Anders gaan je maand- en jaartotalen (in archief) namelijk alsnog naar de knoppen door die afwijking.

Nadat je geüpgrade hebt naar `v1.5.5` kun je in de admin interface de dag (gisteren) opzoeken bij `Dagstatistieken` en `Uurstatistieken`. De absurde getallen kun je daar op nul zetten of zelf berekenen. Ik verwacht je dat je slechts 1 dag en 1 uur hoeft te wijzigen.

Voor de rest is het afwachten op de definitieve fix in 225. Daar zal wel enige tijd overheen gaan.

closed 2017-01-19 by dennissiemensma

hsleewis, 2017-01-21:
Als ik een post deploy doe blijf ik nog steeds op v1.5.4

dennissiemensma, 2017-01-21:
``Post-deploy`` hoef je alleen maar te gebruiken om de huidige versie te herladen, wat eigenlijk nooit nodig is.

Voor upgraden [zie de docs hier](http://dsmr-reader.readthedocs.io/nl/development/faq.html#how-can-i-update-my-application).

hsleewis, 2017-01-21:
Sjonge, ik zat te slapen, dank je wel! :)