dsmrreader/dsmr-reader #245: Grafiek gasverbruik doet wat vreemd na aantal uur geen nieuwe data

ThinkPadNL, 2017-01-18:
Ik zit even te kijken naar m'n verbruik, maar volgens mij komt e.e.a. niet overeen. Nu is het zo dat de metingen er een paar uur uit hebben gelegen, zou het kunnen dat e.e.a daardoor niet meer synchroon loopt?
[image]
[image]

Als ik de punten in de grafiek optel (vanaf 0:00 uiteraard) kom ik op 1,9m3. De tekstweergave-indicator zal wel kloppen, maar die grafiek doet een beetje raar op deze manier 😄

dennissiemensma, 2017-01-18:
Ja klopt, alle meetpunten zijn event-gebaseerd, dus als er geen data is maakt die ook geen datapunt en weet de grafiek ook niet dat er iets mist. 

Dit valt denk ik onder dezelfde noemer als #198 en dan mag de grafiek rekening houden met lege punten.

m2day, 2017-05-13:
Ik denk dat het komt omdat, in [dsmr-reader/dsmr_stats/services.py](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_stats/services.py) de functie  create_hourly_statistics slechts naar het actuele verbruik op het hele uur kijkt in plaats van gedurende het uur. 

In het geval van meters die meerdere resultaten per uur teruggeven zoals bij mijn DSMR v5 klopt de grafiek dan niet. Ik zou verwachten dat de berekening voor het verbruik dan gelijk zou moeten zijn aan die van elektriciteit. Dus stand aan het eind van het uur - stand aan het begin van het uur.

Dus iets in de trant van:
```
if gas_readings.exists():
	gas_start = gas_readings[0] 
	gas_end = gas_readings[gas_readings.count() - 1] 
	creation_kwargs['gas'] = gas_end.delivered - gas_start.delivered
```

in plaats van:
```
if gas_readings.exists(): 
	# Gas readings are unique per hour anyway. 
	creation_kwargs['gas'] = gas_readings[0].currently_delivered 
```

dennissiemensma, 2017-05-14:
Bedankt voor je input. Ik denk dat dat nog een stukje nazorg is, dat ik ben vergeten om te bouwen bij het ondersteunen van DSMR v5.

dennissiemensma, 2017-09-20:
Even onderzocht, test kunnen maken om te reproduceren en de bovenstaande voorgestelde oplossing doorgevoerd. Bedankt voor het melden/uitwerken.

closed 2017-09-20 by dennissiemensma