dsmrreader/dsmr-reader #246: Feature request: add support for Pushover

tpjanssen, 2017-01-24:
Dear Dennis,

Is it possible to add support for Pushover? My guess is that it would be fairly simple to do so.

Regards.

dennissiemensma, 2017-01-24:
Hi, I asume you refer to the notification service Pushover? (Googled it)

tpjanssen, 2017-01-25:
Sorry, for the incomplete request, but yes, that's what I would like to see implemented.

Thanks.

dennissiemensma, 2018-06-02:
Dit is nodig als vervanger van Notify My Android. De service werkt helaas wel iets anders, namelijk net zoals Dropbox. Gebruikers moeten zelf een applicatie registreren en dan zowel een API-key als user-token invoeren.

dennissiemensma, 2018-06-05:
Ja, klopt. En die komt vrij snel, ik probeer ze wat kleiner te houden tegenwoordig.

dennissiemensma, 2018-06-05:
Bij deze is v1.18 uitgebracht.