dsmrreader/dsmr-reader #249: Req: Add iOS icon for Bookmark

mbnn, 2017-01-31:
When possible, it would be amazing if you could add a 'transparent, 114x114px icon' that will be used when people bookmark the dsmr site on their iphone/ipad:

<link rel="apple-touch-icon" href="icon.png"/> is the only line needed, could simply use the DSMR reader icon on the correct format and it's done :)

dennissiemensma, 2017-01-31:
I will see what I can do about that. All templates, and icons and stuff used in this project are free-to-use/non-commercial. I'm no designer but I can try to see whether I get this working.

dennissiemensma, 2017-01-31:
Some overlap from #250, quoted from @emilkwaaitaal:

> I'm launching DSMR Reader from my homescreen on iPhone and iPad (save webpage to homescreen in Safari browser). There are possibilities to add a homescreen icon and also possibilities to launch full screen (without browser controls) to make it really look like a native iOS app.
> 
> See official documentation from Apple on how to get this working:
> https://developer.apple.com/library/content/documentation/AppleApplications/Reference/SafariWebContent/ConfiguringWebApplications/ConfiguringWebApplications.html
> 
> Would be great if this can be implemented to further improve the user experience on iOS devices

dennissiemensma, 2017-02-14:
I've added the icon and meta tag in fc5308c0438ef8116fa9d7d836733b18c5373153 for the next v1.6 release. 

Since I'm not a designer and I do not have a hi-res version of the icon, the quality suffers a bit, but it's better better than having nothing.

Thanks again for reporting.

closed 2017-02-14 by dennissiemensma

TopdRob, 2017-02-15:
Hi Dennis,

http://realfavicongenerator.net is a great tool for creating favicons for different platforms in the correct formats.

Best regards,
Robert

dennissiemensma, 2017-02-15:
Thanks for sharing. I've looked up the [original source](http://www.flaticon.com/free-icon/eco-energy_25013) of the favicon, and it seems there is a higher resolution available as well. So I'll update it once more.

dennissiemensma, 2017-02-15:
Updated via f7fba90dc821ef3842e22400f35fed1da2b100b4