dsmrreader/dsmr-reader #251: Buienradar Uncaught exception

perryodk, 2017-01-31:
Beste Dennis,

Allereerst bedankt voor je tijd die je in dit mooie project steekt! Ik gebruik het naar volle tevredenheid, echter eerder vandaag is het " dsmr_backend" proces onverwachts gestopt.

"dsmr_backend" proces logging:

2017-01-31

 10:59:54.046850+01:00: Calling backend services
Command completed. Sleeping for 5 second(s)...

2017-01-31 10:59:59.079288+01:00: Calling backend services
 - Processed reading: 2017-01-31 10:59:57.959628+01:00.
Command completed. Sleeping for 5 second(s)...

2017-01-31 11:00:04.098207+01:00: Calling backend services
 - Performing temperature reading at Buienradar.
 >>> Uncaught exception :: 'NoneType' object has no attribute 'text'
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_weather>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 29, in run
    raise current_response

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 227, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_weather/apps.py", line 20, in _on_backend_called_signal
    dsmr_weather.services.read_weather()

  File "/home/dsmr/dsmr-reader/dsmr_weather/services.py", line 47, in read_weather
    temperature = temperature_element.text
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 29, in run
    raise current_response
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 227, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_weather/apps.py", line 20, in _on_backend_called_signal
    dsmr_weather.services.read_weather()

  File "/home/dsmr/dsmr-reader/dsmr_weather/services.py", line 47, in read_weather
    temperature = temperature_element.text
CommandError: ['  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 29, in run\n    raise current_response\n\n  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 227, in send_robust\n    response = receiver(signal=self, sender=sender, **named)\n\n  File "/home/dsmr/dsmr-reader/dsmr_weather\n    temperature = temperature_element.text\n']
Starting INFINITE command loop...

Na het handmatig stoppen en opnieuw starten van het "dsmr_backend" proces middels supervisorctl worden de onverwerkte telegrammen verwerkt.
Opvallend is onderstaande dik gedrukte tekst in de logging:

- Processed reading: 2017-01-31 15:12:18.424425+01:00.
 - Processed reading: 2017-01-31 15:12:28.444812+01:00.
 **- Performing temperature reading at Buienradar.**
Command completed. Sleeping for 5 second(s)...

2017-01-31 20:08:09.096306+01:00: Calling backend services
 - Processed reading: 2017-01-31 15:12:38.468654+01:00.
 - Processed reading: 2017-01-31 15:12:48.489663+01:00.
 - Processed reading: 2017-01-31 15:12:58.515101+01:00.

Heb je enig idee waardoor dit veroorzaakt kan zijn?

dennissiemensma, 2017-01-31:
Bedankt voor het melden van dit issue inclusief alle info. 

Ik denk dat hier de [API van Buienradar](http://api.buienradar.nl/) even niet beschikbaar was. Het is alleen niet de bedoeling dat die daarop stuk gaat, dus ik zal in een latere release een bettere oplossing voor maken.

dennissiemensma, 2017-02-14:
Ik heb een aanpassing gemaakt in 6d92eee0ebf2c3e9f3f779b20a5ef3583a18cf8d waardoor een fout het proces niet meer zou moeten stoppen. Dit gaat mee in de aankomende 1.6 release. Nogmaals bedankt voor het melden.

closed 2017-02-14 by dennissiemensma