dsmrreader/dsmr-reader #252: Change database credentials and web root path

bsmeding, 2017-01-31:
Hi i have a server running ISPCONFIG3 for the web hosting (based on nginx), now i have created a MySQL database from within the control panel, and also a stong password.
Also the base path of the www directory is different and i see that the script (./manage.py collectstatic --noinput)  is looking for /var/www/dsmrreader/

where can i change:

- database name
- database user
- database password
- root of www directory

bsmeding, 2017-01-31:
found the database credentials in : dsmrreader/provisioning/django/mysql.py
edited and rerun the copy and ./mirate.py

but still need the path of the www files to change, now run on error by : ./manage.py collectstatic --noinput
:[Errno 13] Permission denied: '/var/www/dsmrreader'

dennissiemensma, 2017-01-31:
Somewhere during the installation guide you executed this command:
```
cp dsmrreader/provisioning/django/mysql.py dsmrreader/settings.py
```

It copied a default/template config to the application with default credentials. 

(!) Please note that `dsmrreader/provisioning/django/mysql.py` is **not** used by the application, it's a template.

You can edit the copied `dsmrreader/settings.py` file (inside the project), to alter the credentials used. 

Regarding the static files path, it's [located in some other settings file](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/config/production.py#L14), included by the one above. But you can easily **override** it by editing the same  `dsmrreader/settings.py` file.

Just add a new line there with `STATIC_ROOT = '/whatever/path/you/need'`

**Please note that the files are loaded once in memory on start**. So you'd have to either restart Supervisor processed or (even better), gracefully reload them with the `reload.sh` script located in the root of the project folder. Then the changes should take affect.

bsmeding, 2017-01-31:
Hi Dennis, 
many thanks for the quick response and the very fancy DSMR reader you have build! 

I also found out myself some of the settings to change.

I now have changed settings in  : 

Change MySQL credentials in :
* nano dsmrreader/provisioning/django/mysql.py
* nano dsmrreader/provisioning/mysql/mysql-backup.sh 
Change Web root in  :
* nano dsmrreader/config/production.py
* nano dsmrreader/provisioning/ngingx/dsmr-webinterface

but the webinterface is not running, i had changed the dsmr-webinterface to another port (8085), but still see the old path is trying to load:

`2017/01/31 21:58:36 [error] 31941#0: *102 open() "/var/www/dsmrreader/static/dsmr_frontend/js/chartjs/Chart-2.4.0.min.js" failed (2: No such file or directory),`

A had to change the dsmr-webinterface file so that the uptream is removed and the proxy_pass is changed to 
`proxy_pass http://unix:/var/tmp/gunicorn--dsmr_webinterface.socket;`

then there is something showing up, but without the styles loaded.

will look into this further in the next couple of days..

dennissiemensma, 2017-01-31:
You will have to revert some of the changes, as most files you've edited are **under version control**. You can view them by doing `sudo su - dsmr` and then `git status` in the project directory. It **should** say that there are no changes.

I'm not sure what git will do *when you will update to a new version of dsmrreader in the future*. There is a possibility that it will mark all those files as conflicted and you'll manually have to resolve them.

Therefor you should only update the **copies** you've made, which are:
* `dsmrreader/settings.py`
* `/etc/supervisor/conf.d/dsmr-reader.conf`
*  Please copy this one somewhere else before editing: `dsmrreader/provisioning/mysql/mysql-backup.sh`

Regarding Ngix, if you want to use a port instead of a socket, you should remove the `unix:///` directive in it. 
Do you even need to change the www-path? If you want change it you should change it in the Nginx vhost as well.

bsmeding, 2017-02-01:
I've reverded the changes  I made to the files and made the changes to the files you said.

Also changed de nginx file, uncommented the 3 lines of the upstream part, now it is starting, only no data received yet and every menu button is throwing a error. When looking at the log it will still connect to the upstream.

`2017/02/01 16:16:22 [crit] 26118#0: *1 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstream, client: 192.168.111.141, server: _, request: "GET /admin/ HTTP/1.1", upstream: "http://unix:///var/tmp/gunicorn--dsmr_webinterface.socket:/admin/", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:16:25 [crit] 26118#0: *1 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstream, client: 192.168.111.141, server: _, request: "GET /archive HTTP/1.1", upstream: "http://unix:///var/tmp/gunicorn--dsmr_webinterface.socket:/archive", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *1 open() "/var/www/dsmrreader/static/dsmr_frontend/css/bootstrap.min.css" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/css/bootstrap.min.css HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *2 open() "/var/www/dsmrreader/static/dsmr_frontend/css/font-awesome.min.css" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/css/font-awesome.min.css HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *8 open() "/var/www/dsmrreader/static/dsmr_frontend/css/ionicons.min.css" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/css/ionicons.min.css HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *9 open() "/var/www/dsmrreader/static/dsmr_frontend/css/lato-font.css" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/css/lato-font.css HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *1 open() "/var/www/dsmrreader/static/dsmr_frontend/js/jquery-3.1.1.min.js" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/js/jquery-3.1.1.min.js HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *10 open() "/var/www/dsmrreader/static/dsmr_frontend/css/style.css" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/css/style.css?r=2 HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *11 open() "/var/www/dsmrreader/static/dsmr_frontend/css/dsmrreader.css" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/css/dsmrreader.css HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *2 open() "/var/www/dsmrreader/static/dsmr_frontend/js/bootstrap.min.js" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/js/bootstrap.min.js HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *8 open() "/var/www/dsmrreader/static/dsmr_frontend/js/Director/app.js" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/js/Director/app.js?v=2 HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
2017/02/01 16:17:24 [error] 26118#0: *9 open() "/var/www/dsmrreader/static/dsmr_frontend/js/chartjs/Chart-2.4.0.min.js" failed (2: No such file or directory), client: 192.168.111.141, server: _, request: "GET /static/dsmr_frontend/js/chartjs/Chart-2.4.0.min.js HTTP/1.1", host: "192.168.111.6:8085", referrer: "http://192.168.111.6:8085/"
`

Think the installation on a already in use server is not that easy. Will look into this further when I have a spare Pi readdy.

dennissiemensma, 2017-02-01:
Could you paste your Nginx vhost here? Please redact any personal data such as a domain name (if any).

bsmeding, 2017-02-02:
Hi Dennis,

The config from nginx is:
`##upstream dsmr-webinterface {
##    server ## unix:///var/tmp/gunicorn--dsmr_webinterface.socket fail_timeout=0;
##}

server {
    listen      8085;
    server_name _;  # No hostname available? Just use the underscore for wildcard matching. :]

##      # Uncomment these lines to force additional authentication on application access.
##  # Do not forget to generate a password first in '/etc/nginx/htpasswd' and then RELOAD nginx!
##    auth_basic "Restricted application";
##    auth_basic_user_file /etc/nginx/htpasswd;

    location /static {
        alias /var/www/dsmrreader/static;
    }

    location / {
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Host $http_host;
        proxy_redirect off;
        proxy_pass http://dsmr-webinterface;
    }
}

`

dennissiemensma, 2017-02-02:
Ah I see, you don't need to comment the `upstream dsmr-webinterface` block. I'm not sure how much you know about running Django, but here's an explaination anyway.

Nginx should just pass all requests on /static/* to a folder where all the static files are at. This must **not** be the application installation folder, but a random static folder outside the application, accessible to both Nginx and `dsmr` user (of whatever user is running the app). 
The 'random static folder' will need to be set in the application as well, using the `STATIC_ROOT` setting. The `collectstatic` command will **copy** all static files into that folder.

The rest of the request in Nginx should go to `/`, which will proxy to Django which is running in the Supervisor process.

* Do you need to change the way the `upstream` is working? Because it now uses a local socket, but it can use a local port as well.

* Does `alias /var/www/home.local/web/dsmrreader/static;` point to the installation path of the dsmr-reader application? If so, that will *not work* (see above).

bsmeding, 2017-02-03:
I've removed the whole upstream part

Some more of my setup right now:

in /home/dsmr/dsmr-reader is the git clone of the application
i've changed the settings.py and added the root path of my website folder : 
```
DATABASES = {
    'default': {
        'ENGINE': 'django.db.backends.mysql',
        'NAME': 'c1dsmrreader',  # Database name.
        'USER': '<<MYUSER>>',
        'PASSWORD': '<<MYPASSWORD>>,
        'HOST': 'localhost',
        'CONN_MAX_AGE': 60,
        'STATIC_ROOT': '/var/www/home.local/web/dsmrreader/static/',
    }
}
```

Virtual Env path : /home/dsmr/.virtualenvs/dsmrreader

Web folder : /var/www/home.local/web/dsmrreader/static/
```
root@srv-01:/var/www/home.local/web/dsmrreader/static# ls -al
drwxr-xr-x 5 dsmr client1 4096 Jan 31 21:28 .
drwxr-xr-x 3 dsmr client1 4096 Jan 31 20:28 ..
drwxr-xr-x 6 dsmr dsmr    4096 Jan 31 21:28 admin
drwxr-xr-x 3 dsmr dsmr    4096 Jan 31 21:28 colorfield
drwxr-xr-x 8 dsmr dsmr    4096 Jan 31 21:28 dsmr_frontend
```

The web folder right are normally web5:client1  (this is done by the ISPCONFIG admin portal)

also the proxy_pass part which redirect the "/"to dsmr-webinterface

```
upstream dsmr-webinterface {
    server unix:///var/tmp/gunicorn--dsmr_webinterface.socket fail_timeout=0;
}

server {
    listen      8085;
    server_name _;  # No hostname available? Just use the underscore for wildcard matching. :]

##      # Uncomment these lines to force additional authentication on application access.
##  # Do not forget to generate a password first in '/etc/nginx/htpasswd' and then RELOAD nginx!
##    auth_basic "Restricted application";
##    auth_basic_user_file /etc/nginx/htpasswd;

    location /static {
        alias /var/www/home.local/web/dsmrreader/static;
    }

    location / {
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Host $http_host;
        proxy_redirect off;
        proxy_pass http://dsmr-webinterface;
    }
}
```

Now i get the error 
```2017/02/03 12:22:46 [crit] 13248#0: *1 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstream, client: 192.168.111.141, server: _, request: "GET / HTTP/1.1", upstream: "http://unix:///var/tmp/gunicorn--dsmr_webinterface.socket:/", host: "192.168.111.6:8085"```

the file is not created for webinterface:
```
root@srv-01:/var/tmp# ls -al
total 32
drwxrwxrwt  3 root root 4096 Feb  3 12:25 .
drwxr-xr-x 16 root root 4096 Mar  2  2016 ..
-rw-r--r--  1 dsmr dsmr    5 Feb  3 12:25 dsmrreader--dsmr_backend.pid
-rw-r--r--  1 dsmr dsmr    5 Feb  3 11:48 dsmrreader--dsmr_datalogger.pid
```

in the supervisor is see error 
```
root@srv-01:/var/www/home.local/web/dsmrreader# sudo supervisorctl
dsmr_backend                     STARTING   
dsmr_datalogger                  RUNNING    pid 28644, uptime 0:35:20
dsmr_webinterface                FATAL      Exited too quickly (process log may have details)
supervisor> status
dsmr_backend                     RUNNING    pid 16076, uptime 0:00:05
dsmr_datalogger                  RUNNING    pid 28644, uptime 0:36:33
dsmr_webinterface                FATAL      Exited too quickly (process log may have details)
supervisor> 

```

The backend and logger are changing from RUNNING, STOPPING, STARTING in a loop, the webinterface is error, when in look at the log i see that is is missing the module 'dsmrreader'...

```
    return self.load_wsgiapp()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/wsgiapp.py", line 52, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/util.py", line 357, in import_app
    __import__(module)
ImportError: No module named 'dsmrreader'
[2017-02-03 11:48:26 +0100] [28503] [INFO] Worker exiting (pid: 28503)
[2017-02-03 11:48:26 +0100] [28497] [INFO] Shutting down: Master
[2017-02-03 11:48:26 +0100] [28497] [INFO] Reason: Worker failed to boot.
[2017-02-03 11:48:30 +0100] [28506] [INFO] Starting gunicorn 19.6.0
[2017-02-03 11:48:30 +0100] [28506] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (28506)
[2017-02-03 11:48:30 +0100] [28506] [INFO] Using worker: sync
[2017-02-03 11:48:30 +0100] [28588] [INFO] Booting worker with pid: 28588
[2017-02-03 11:48:30 +0100] [28588] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/arbiter.py", line 557, in spawn_worker
    worker.init_process()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/workers/base.py", line 126, in init_process
    self.load_wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/workers/base.py", line 136, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/wsgiapp.py", line 65, in load
    return self.load_wsgiapp()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/wsgiapp.py", line 52, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/util.py", line 357, in import_app
    __import__(module)
ImportError: No module named 'dsmrreader'
[2017-02-03 11:48:30 +0100] [28588] [INFO] Worker exiting (pid: 28588)
[2017-02-03 11:48:31 +0100] [28506] [INFO] Shutting down: Master
[2017-02-03 11:48:31 +0100] [28506] [INFO] Reason: Worker failed to boot.
```

Do i need to re-register / install the module? or is it a conflict of python?

dennissiemensma, 2017-02-04:
Well, Nginx is throwing the error because the webinterface process crashes in Supervisor. But I can not think of an obvious reason while it crashes anyway.

Did you happen to make any changes to the Supervisor config? You could try to execute [the command manually](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.5/dsmrreader/provisioning/supervisor/dsmr-reader.conf#L45). There might be something wrong with the Python path or directory it's trying to run it from.

bsmeding, 2017-02-08:
When i try to run manually i get the same error

```
root@srv-01:/usr/bin# /usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterfaces.socket --pid /var/tmp/gunicorn--dsmr_webinterfaces.pid dsmrreader.wsgi
[2017-02-08 15:28:24 +0100] [8745] [INFO] Starting gunicorn 19.6.0
[2017-02-08 15:28:24 +0100] [8745] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterfaces.socket (8745)
[2017-02-08 15:28:24 +0100] [8745] [INFO] Using worker: sync
[2017-02-08 15:28:24 +0100] [8809] [INFO] Booting worker with pid: 8809
[2017-02-08 15:28:24 +0100] [8809] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/arbiter.py", line 557, in spawn_worker
    worker.init_process()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/workers/base.py", line 126, in init_process
    self.load_wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/workers/base.py", line 136, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/wsgiapp.py", line 65, in load
    return self.load_wsgiapp()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/wsgiapp.py", line 52, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/util.py", line 357, in import_app
    __import__(module)
ImportError: No module named 'dsmrreader'
[2017-02-08 15:28:24 +0100] [8809] [INFO] Worker exiting (pid: 8809)
[2017-02-08 15:28:24 +0100] [8745] [INFO] Shutting down: Master
[2017-02-08 15:28:24 +0100] [8745] [INFO] Reason: Worker failed to boot.
root@srv-01:/usr/bin# 

```

is it possible that python 2.7.9 which is also installed could be a problem?

dennissiemensma, 2017-02-08:
You have to run it from the appropriate directory as well, using all the settings the Supervisor config uses.

* Run as `dsmr` user.
* Run from `/home/dsmr/dsmr-reader/`

The process passes the argument `dsmrreader.wsgi` to the Gunicorn command, which translates to the path `dsmrreader/wsgi.py`, relative to the directory it should execute in. 

The latter should exist if you've checkout out the git clone at `/home/dsmr/dsmr-reader/`, making the full path `/home/dsmr/dsmr-reader/dsmrreader/wsgi.py`. It seems that it cannot find the path specified, and thus `no module dsmrreader`.

Can you check whether the path exists? And whether you get the same error when you cd to the project root folder and try again?

dennissiemensma, 2017-02-14:
@bsmeding did you find the time yet to test/check the information mentioned in the comment above?

bsmeding, 2017-02-15:
Hi Dennis,

sorry i was busy with other things.
just tried it with user dsmr

```
(dsmrreader)dsmr@srv-01:~/dsmr-reader$ /usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterfaces.socket --pid /var/tmp/gunicorn--dsmr_webinterfaces.pid dsmrreader.wsgi
[2017-02-15 15:27:12 +0100] [7135] [INFO] Starting gunicorn 19.6.0smr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix[2017-02-15 15:27:12 +0100] [7135] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterfaces.socket (7135)
[2017-02-15 15:27:12 +0100] [7135] [INFO] Using worker: sync
[2017-02-15 15:27:12 +0100] [7265] [INFO] Booting worker with pid: 7265
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:20 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:21 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:21 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:21 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:21 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:27:21 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
[2017-02-15 15:28:11 +0100] [7135] [INFO] Handling signal: winch
^C[2017-02-15 15:28:40 +0100] [7135] [INFO] Handling signal: int
[2017-02-15 15:28:40 +0100] [7265] [INFO] Worker exiting (pid: 7265)
[2017-02-15 15:28:40 +0100] [7135] [INFO] Shutting down: Master
(dsmrreader)dsmr@srv-01:~/dsmr-reader$ 

```

still the file gives an error in /var/log/nginx/error.log:
```
2017/02/15 15:28:34 [crit] 19411#0: *55858 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstr$
2017/02/15 15:28:34 [crit] 19411#0: *55858 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstr$
```

but the file is created, only with 0 bytes in it

```
drwxrwxrwt  3 root root 4096 Feb 15 15:30 .
drwxr-xr-x 16 root root 4096 Mar  2  2016 ..
-rw-r--r--  1 root root  608 Jan 31 15:42 dtb.log
-rw-r--r--  1 root root  993 Jan 31 15:42 dtb.log.errors
-rw-r--r--  1 dsmr dsmr    6 Feb 15 15:30 gunicorn--dsmr_webinterfaces.pid
srwxrwxrwx  1 dsmr dsmr    0 Feb 15 15:30 gunicorn--dsmr_webinterfaces.socket
drwxr-xr-x  2 root root 4096 Jul  6  2016 openvpn_as
-rw-r-----  1 root root  275 Jul 18  2016 rain.tmp
root@srv-01:/var/tmp# 
```

dennissiemensma, 2017-02-15:
The file actually should be zero Bytes, because it's a socket (see the `s` in the permissions).

Also, I think the problem is in the file name, because you've spawned it with an additional `s` on the end:

```
gunicorn--dsmr_webinterfaces.socket   <<< extra S

vs.

gunicorn--dsmr_webinterface.socket
```

You should remove the S and try again.

bsmeding, 2017-02-16:
Hi Dennis,

whow I think everything is working now! I've added the S to the nginx file as the webinterfaces.socket is how it is in your template configuration file [https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/provisioning/supervisor/dsmr-reader.conf#L45](url)

I will wait till tomorrow to get some data visible

dennissiemensma, 2017-02-16:
Great, you should see electricity data within minutes and gas (if available) within hours. Let me know whether it's working correctly tomorrow.

The variable is actually called "`%(program_name)s`", but it doesn't matter, as long as they are in sync. Configs aren't updated with new releases anyway, only for new installations. :]

bsmeding, 2017-02-17:
Yes i have received data, but the process of python3 is consuming 100 percent CPU.

```
top - 22:09:32 up 16 days,  6:28,  2 users,  load average: 1.04, 2.16, 2.42
Tasks: 198 total,   3 running, 195 sleeping,   0 stopped,   0 zombie
%Cpu(s): 12.7 us,  4.0 sy, 42.9 ni, 39.4 id,  1.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:   2062116 total,  1768752 used,   293364 free,   157436 buffers
KiB Swap:  4177916 total,   254068 used,  3923848 free.   925940 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                                                       
**11970 dsmr      30  10   46988  31136   9896 R  86.0  1.5   0:02.95 python3**       
```

```
top - 22:08:37 up 16 days,  6:27,  2 users,  load average: 1.17, 2.40, 2.51
Tasks: 190 total,   3 running, 187 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1.2 us,  3.0 sy, 26.1 ni, 69.5 id,  0.0 wa,  0.0 hi,  0.2 si,  0.0 st
KiB Mem:   2062116 total,  1747844 used,   314272 free,   157276 buffers
KiB Swap:  4177916 total,   254068 used,  3923848 free.   925800 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                                                       
**11519 dsmr      30  10   19896  15296   7228 R  26.7  0.7   0:00.81 python3**                                                                                                       
11327 root      20   0    5260   2956   2508 R   1.3  0.1   0:00.29 top  
```

dennissiemensma, 2017-02-17:
Can you see which command is consuming it? Python 3 is used by multiple commands, such as the `backend` and `datalogger`. You can identify it by looking up the PID in `ps faux` output.

The commands in Supervisor are `nice`'d by the way, as you can see as well (`26.1 ni`, `42.9 ni`). They should backoff a bit when other processes are requesting CPU as well.

Is you server connected directly to the smart meter at your home or do you use the API?

If you are using the latter (API), then you should disable the `dsmr_datalogger` process in Supervisor, because it will constantly attempt to read a non-existing serial connection..

bsmeding, 2017-02-21:
It is mostly the backend consuming cpu

```
dsmr     10108  1.7  1.1  42428 24388 ?        SN   Feb17  99:36  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogg
dsmr     17777  0.0  0.6  18768 13132 ?        SN   Feb19   0:58  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn -
dsmr     17780  0.0  0.9  31276 19728 ?        SN   Feb19   0:04  |   \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunico
dsmr     19384 15.3  1.3  44648 28796 ?        SN   14:40   0:22  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_backend
```

I will close this Issue as the starting issue is solved. For the cpu part i will check if i can move some other process to another server. This server is in the 'meterkast' so directly connected to the P1 interface

closed 2017-02-21 by bsmeding

dennissiemensma, 2017-02-21:
Thanks for the info. 

The application shouldn't take that much CPU, except when it's creating a database backup (and compressing it). You could check the logs of the backend process in `/var/log/supervisor/` to see what it's doing. Most of the time it should just be sleeping.