dsmrreader/dsmr-reader #253: Unable to succesfully run step 8.

joopy, 2017-02-02:
following your installation manual: http://dsmr-reader.readthedocs.io/nl/latest/installation.html#application-configuration-setup

Step 7: Install as (dsmrreader)dsmr@raspberrypi user went fine no error's.
Running step 8: result in error:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/psycopg2/__init__.py", line 164, in connect
    conn = _connect(dsn, connection_factory=connection_factory, async=async)
django.db.utils.OperationalError: FATAL:  password authentication failed for user "dsmrreader"
FATAL:  password authentication failed for user "dsmrreader"

(dsmrreader)dsmr@raspberrypi:~/dsmr-reader $

joopy, 2017-02-02:
dsmrreader/settings.py alter password when you use your own db password... Mayby state this is your manual?

closed 2017-02-02 by joopy

dennissiemensma, 2017-02-02:
You've closed the issue, did you managed to fix it yourself?

I'm lacking some info here, are you pointing out that you are using an existing database?  I will improve the docs by explaining some settings via #232 later.

joopy, 2017-02-03:
Step 1.
sudo sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"

I used my own password..