dsmrreader/dsmr-reader #254: Data verwerking

knetsooj, 2017-02-02:
Kon niet vinden waar ik evt vragen kan stallen dus dan maar hier.
Ik maak al enkele weken tevreden gebruik van deze applicatie en alles was tot dusver goed.
nu keek ik vandaag naar de interface en zag ik het volgende:

[image]

Wat kan er voor zorgen dat de data verwerking niet meer loopt?
een reboot van de RPI deed het verwerken van telegrammen weer starten.

dennissiemensma, 2017-02-02:
Bedankt voor het melden. Het kan door een bug komen of een databasebackup. Van de week crashte het ook al bij iemand om deze reden: #251

Kun je zien of je een foutmelding kan terugvinden in de logs van de dataverwerking? Die staan in `/var/log/supervisor/dsmr_backend.log`.

knetsooj, 2017-02-03:
Helaas, ik maak voor de logging gebruik van tmpfs om mijn SD te besparen. Er is daar dus niets terug te zien van voor de reboot.

dennissiemensma, 2017-02-04:
Oke. Mocht het weer voorkomen, dan kun je in ieder geval in die logs kijken voor een mogelijke oorzaak.

dennissiemensma, 2017-02-14:
@knetsooj ik heb in #251 een mogelijke fix gemaakt voor een van de oorzaken van dit probleem. Dat gaat mee bij de volgende v1.6 release. Mocht in de tussentijd toch nog zo'n fout ervaren en meer informatie hebben, dan hoor ik het graag. :]

closed 2017-02-14 by dennissiemensma

knetsooj, 2017-03-17:
@dennissiemensma na een weekje te zijn weggeweest zag ik zojuist dat de telegrammen al 4,5 dag niet meer verwerkt worden. in de backend log zie ik geen foutmeldingen,  laatste bericht:
***
2017-03-13 10:00:02.648887+01:00: Calling backend services
 - Processed reading: 2017-03-13 09:59:00+01:00.
 - Processed reading: 2017-03-13 09:59:11+01:00.
 - Processed reading: 2017-03-13 09:59:31+01:00.
 - Processed reading: 2017-03-13 09:59:41+01:00.
 - Processed reading: 2017-03-13 09:59:51+01:00.
 - Performing temperature reading at Buienradar.
***
Hierna is er voor de dsmr_backend.log geen nieuwe logging meer aangemaakt. Status in de supervisor is nog steeds running.

dennissiemensma, 2017-03-18:
Bedankt voor het melden. Betekent dit dat je onverwerkte telegrammen hebt, of heeft de applicatie helemaal niets meer gemeten? (zie Status pagina)

knetsooj, 2017-03-18:
Ik heb voor bijna 5 dagen aan onverwerkte telegrammen,  na service restart was #280 na een aantal uren het gevolg, ik weet echter niet in hoeverre dit aan elkaar gerelateerd was en of dit als twee individuele issues moet worden behandeld

dennissiemensma, 2017-03-18:
Ik denk dat het losse dingen zijn. Zojuist is versie v1.6 uitgebracht en die verhelpt sowieso een probleem dat door foute data van Buienradar veroorzaakt kan worden.