dsmrreader/dsmr-reader #256: dsmr_webinterface                FATAL      Exited too quickly (process log may have details)

eterpstra, 2017-02-03:
Hoi,

Sinds gister kon ik dsmrreader webinterface niet meer laden.  Supervisor geeft de volgende error aan bij status:

dsmr_webinterface                FATAL      Exited too quickly (process log may have details)

reread commando lukt helaas niet. Ik krijg dan de volgende melding:
No config updates to processes

Hoe kan ik ervoor zorgen dat de webinterface weer gaat werken?

dennissiemensma, 2017-02-04:
Je kunt het proces weer proberen te starten met de `sudo supervisorctl` tool. Eventueel direct een command meegeven:

* `sudo supervisorctl start dsmr_webinterface`

De `reread` gebruik je alleen als je wijzigingen aanbrengt in de configuratie (eenmalig bij installatie).

dennissiemensma, 2017-02-14:
@eterpstra is dit gelukt? Draait het weer naar behoren?

closed 2017-02-26 by dennissiemensma