dsmrreader/dsmr-reader #259: dummy data to test

reflexpnt, 2017-02-14:
Denniss, it is possible that you can supply silly read data to be incorporated into the database. I have run the Debian 8 server but in my country I do not have a DSMR to perform tests and I am interested in seeing how your project works. I would like to try to base myself on your project to achieve a server (with django) but for CAN bus readings.
(i'm sorry by my poor english)
thank you very much.

dennissiemensma, 2017-02-14:
Hi, there actually is a hidden [dsmr_fake_datasource](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.5/dsmr_datalogger/management/commands/dsmr_fake_datasource.py) management command for this. I haven't used it for quite a while, but I think it should still work. It generates random data, based on some silly rules.

You would have to replace the default `dsmr_datalogger` command/app (in supervisor) by this one.

**Please note that you should DISABLE CRC-checking in the datalogger settings as well**. Because the telegrams the fake command generates, do not have a valid CRC signature.

The `--help` argument displays:
```
  --run-once            Forces single run, overriding Infinite Command mixin
  --ack-to-mess-up-my-data
                        Required option to acknowledge you that you WILL mess
                        up your data with this.
  --with-gas            Include gas consumption
  --with-electricity-returned
                        Include electricity returned (solar panels)
```

The `--ack-to-mess-up-my-data` is mandatory, you can optionally use `--with-gas` and `--with-electricity-returned`.

Please let me know whether this helps you, I'm 99% sure it should still work, without CRC checking.

dennissiemensma, 2017-02-14:
In addition, you can find some (anonimized) production data/examples [in the tests as well](https://github.com/dennissiemensma/dsmr-reader/tree/v1.5.5/dsmr_datalogger/tests/datalogger). For example in `test_landisgyr350_dsmr42.py`.

reflexpnt, 2017-02-14:
dennissiemensma , Thank you very much for the quick response. I'm going to study / try these steps and tell you. (I'm new to django too.)
Again thousand thanks.
Fernando

reflexpnt, 2017-02-14:
dennissiemensma , I disable CRC check under admin wep page and I try this:

reflex@Magi:~$ sudo cp /etc/supervisor/conf.d/dsmr-reader.conf /etc/supervisor/conf.d/dsmr-reader.conf.BKP
reflex@Magi:~$ sudo nano /etc/supervisor/conf.d/dsmr-reader.conf 
reflex@Magi:~$ sudo supervisorctl
dsmr_backend                     RUNNING    pid 825, uptime 0:09:57
dsmr_datalogger                  RUNNING    pid 824, uptime 0:09:57
dsmr_webinterface                RUNNING    pid 823, uptime 0:09:58
supervisor> reread
dsmr_datalogger: disappeared
dsmr_fake_datasource: available
supervisor> update
dsmr_datalogger: stopped
dsmr_datalogger: removed process group
dsmr_fake_datasource: added process group
supervisor> status
dsmr_backend                     RUNNING    pid 825, uptime 0:11:27
dsmr_fake_datasource             RUNNING    pid 1870, uptime 0:00:47
dsmr_webinterface                RUNNING    pid 823, uptime 0:11:28
supervisor> 
reflex@Magi:~$ ^C
reflex@Magi:~$ 

/etc/supervisor/conf.d/dsmr-reader.conf
...
#################################################################################
# Has the solely task of polling your meter, to perform and store any readings. #
# stats, backups and any (optional) third party integrations.                   #
#################################################################################

[program:dsmr_fake_datasource]
command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/ds$
pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=20
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5
...
...
...

anyway in the dashboard i getNo data found. Please make sure the datalogger is up and running and the backend process is running as well.
NOTIFICATIONS
   View page   Welkom bij DSMR-reader! Controleer eerst de instellingen in de Configuratie-pagina!
   View page   Je kunt de status van de metingen bekijken op de Status-pagina.
   View page   Vanaf nu is het mogelijk om de elektriciteitstarieven samen te voegen tot enkeltarief. Dit komt van pas wanneer je energieleverancier je slechts één tarief in rekening brengt voor zowel piek als dal. Je kunt dit inschakelen in de interfaceconfiguratie bij "Voeg tarieven samen"
   View page   Het is nu ook mogelijk om je gasmeterstanden automatisch door te geven aan je persoonlijke Mindergas.nl-account! Bekijk de FAQ in de documentatie voor meer informatie over hoe je een account aanmaakt en/of koppelt aan de applicatie.
   View page   Vanaf nu is er ondersteuning voor het versturen van push notificaties naar je telefoon, met daarin je dagelijkse verbruik. Zie voor meer informatie de documentatie / FAQ.

dennissiemensma, 2017-02-15:
Since you're new to Django, I think you should run the command manually first.

This supervisor line you pasted was either cut at some point or it's missing some stuff at the end:
```
command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/ds$
```

You should execute it manually first:

```
sudo su - dsmr
./manage.py dsmr_fake_datasource --ack-to-mess-up-my-data
```

Please make sure that you have both enabled the virtualenv and changed the working dir into the project root ([as instructed here](http://dsmr-reader.readthedocs.io/en/latest/installation.html#virtualenv))

If it did write something to the database, you should be able to see something on the Status page. You could also check the Django admin for the dsmrreading model. It's located at: `/admin/dsmr_datalogger/dsmrreading/`

Regarding the notifications, they are not translated, so you can safely discard them (click the red X) unless you can understand Dutch as well.

Please let me know whether you see anything happening at all.

reflexpnt, 2017-02-15:
Dennis, Sorry to bother you again, but I'm very interested in learning from
your project (in addition to other tutorials).

I have tried the following:

*(dsmrreader) dsmr@Magi:~/dsmr-reader$ ./manage.py dsmr_fake_datasource
--ack-to-mess-up-my-data*
Starting INFINITE command loop...
CommandError: Intended usage is NOT production! Only allowed when DEBUG =
True

Then to try, I comment out the lines (lines 46 & 47) from the
dsmr_fake_datasource.py file:

    def run(self, **options):
        *""" InfiniteManagementCommandMixin listens to handle() and calls
run() in a loop. *
*        if not settings.DEBUG:*
*            raise CommandError(_('Intended usage is NOT production! Only
allowed when DEBUG = True'))"""*

        if not options.get('acked_warning'):
            raise CommandError(_('Intended usage is NOT production! Force
by using --ack-to-mess-up-my-data'))

        self._inject(options)

and try again but give a lot messages (errors ?) and in
the /admin/dsmr_datalogger/dsmrreading/  nothing change.

thank again,regards
Fernando

*(dsmrreader) dsmr@Magi:~/dsmr-reader$ python3 ./manage.py
dsmr_fake_datasource --ack-to-mess-up-my-data*
Starting INFINITE command loop...
--------------------------------
2017-02-15 23:20:06.362292+01:00
with gas: False
with electricity returned: False
--------------------------------

/XMX5LGBBFFB123456789

1-3:0.2.8(40)
0-0:1.0.0(170215232006W)
0-0:96.1.1(FAKE-FAKE-FAKE-FAKE-FAKE)
1-0:1.8.1(003356.341*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(002685.073*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(01.401*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00003)
0-0:96.7.9(00000)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00001)
1-0:52.32.0(00002)
1-0:72.32.0(00003)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.000*kW)
[... 144 lines cut ...]
line 250, in execute
    self.errorhandler(self, exc, value)
  File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/connections.py",
line 42, in defaulterrorhandler
    raise errorvalue
  File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py",
line 247, in execute
    res = self._query(query)
  File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py",
line 411, in _query
    rowcount = self._do_query(q)
  File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py",
line 374, in _do_query
    db.query(q)
  File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/connections.py",
line 270, in query
    _mysql.connection.query(self, query)
django.db.utils.IntegrityError: (1048, "Column 'electricity_delivered_1'
cannot be null")
(dsmrreader) dsmr@Magi:~/dsmr-reader$

dennissiemensma, 2017-02-16:
You should not alter the code, as it will cause conflicts on future updates.

* Regarding the `DEBUG` setting, just add a line `DEBUG = True` to your config in `dsmrreader/settings.py`.

* As for the management command, it seems to break somewhere so I have to debug it. I hoped it would still work. I will look into this shortly, after I finish another ticket.

reflexpnt, 2017-02-16:
Dennis, Thanks a lot . Don't worry about this ticket.  The project run very
well and is more than enough to know the basics. Dummy data is not
required. (I could also simulate them with an electronic board through
serial communication).

Just one more question to guide me. The idea is that the tasks of
'uploading' data to the DB, is Supervisor who takes care of it? (Or
Supervisor 'calls' any application written in Python, which in this case?)

Thank you. regards
Fernando

dennissiemensma, 2017-02-16:
Well, I just happen to finish fixing thefake data command for you, it should work properly now. 

Since you're not running a production environment I've fixed it in the branch for the upcoming release. You can however check that one out easily, since (once again) you're not running production data anyway.

This should update it to the latest development version:

* `sudo su - dsmr`

* Please check wether you've reverted all your local changes first. This command should say something like *nothing to commit, working directory clean*: `git status`

* `git pull`

* `git checkout -b development---v1.6 origin/development---v1.6`

* `./upgrade.sh` (this might take a few seconds as it does lots of stuff)

* The script should end with something like ` --- Deployed version: 1.6b6`

* Now manually executing this command should no longer fail: `./manage.py dsmr_fake_datasource --ack-to-mess-up-my-data --with-gas --with-electricity-returned`

* If that works, you can have Supervisor run it.

As for your question regarding Supervisor: You can use this tool to run any kind of commands, and keep them running. You could also run something similar with system daemons. 
It's used a lot for Python/Django. The [configuration](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/provisioning/supervisor/dsmr-reader.conf) simply spawns three processes. One for the webserver (Gunicorn), one for the datalogger (django management command) and one for antoher backend process (also django management command).

dennissiemensma, 2017-02-16:
I almost forgot, but the fake data (telegrams) are now also signed with a valid CRC checksum, so they should validate even with the CRC-check-setting enabled. See 2440c94ddfd1d67981b70316c74f4681604a9ced for the changes.

reflexpnt, 2017-02-16:
Dennis, Thanks a lot, The system works great. I have a lot to learn.

I was studying a little code, but an idea that I still do not interpret is
if the readings (telegram) are stored in the data base or in files on the
disk (/var/tmp/dsmrreader--%(program_name)s.pid)

regards
Fernando

dennissiemensma, 2017-02-16:
The readings [are stored in the database](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.5/dsmr_datalogger/services.py#L250) indeed. All the data is, except for the logfiles and static files (CSS, etc).

closed 2017-02-26 by dennissiemensma