dsmrreader/dsmr-reader #262: Mindergas Api

DaanvanderKleij, 2017-02-19:
Hello Dennis, 

Great project love the detailed documentation and functionality.

Mindergas contact David La Hei indicates it should be possible to load historical measurements, he was never contacted to discuss the matter and he is open to make this possible.

Secondly the method in which retries are handled is being commented as shown below, based on my manual entries I need to delete this duplicate date on the mindergas page as duplicates can't be stored:

Started POST "/api/gas_meter_readings" for 82.217.***.*** at 2017-02-19 03:49:14 +0100
Processing by Api::GasMeterReadingsController#create as */*
  Parameters: {"date"=>"2017-02-18", "reading"=>"2267.452", "gas_meter_reading"=>{"date"=>"2017-02-18", "reading"=>"2267.452"}}
Completed 422 Unprocessable Entity in 32ms (ActiveRecord: 13.7ms)

Started POST "/api/gas_meter_readings" for 82.217.***.***  at 2017-02-19 04:49:15 +0100
Processing by Api::GasMeterReadingsController#create as */*
  Parameters: {"date"=>"2017-02-18", "reading"=>"2267.452", "gas_meter_reading"=>{"date"=>"2017-02-18", "reading"=>"2267.452"}}
Completed 422 Unprocessable Entity in 52ms (ActiveRecord: 26.8ms)

Started POST "/api/gas_meter_readings" for 82.217.***.***  at 2017-02-19 05:49:20 +0100
Processing by Api::GasMeterReadingsController#create as */*
  Parameters: {"date"=>"2017-02-18", "reading"=>"2267.452", "gas_meter_reading"=>{"date"=>"2017-02-18", "reading"=>"2267.452"}}
Completed 422 Unprocessable Entity in 39ms (ActiveRecord: 19.2ms)

David's remarks: 

"Ieder uur wordt geprobeerd om dezelfde meterstand te uploaden. Retry zou moeten stoppen na een '201 Created' response.
Ook heeft het geen zin om te retry-en bij een '401 Unauthorized' of een '422 Unprocessable Entity'. Dezelfde request zal blijven
falen. Op zich is het niet zo'n probleem, maar echt netjes is het niet.

Dat de applicatie het uploaden van oude meterstanden niet ondersteunt heeft niets met het beleid van MinderGas te maken.
Op zich heb ik daar geen bezwaar tegen, er is mij nooit iets gevraagd.

Wellicht kan je dat ook doorgeven, ze kunnen eventueel contact met mij opnemen.

Met vriendelijke groet,
David La Hei
MinderGas.nl"

dennissiemensma, 2017-02-19:
Bedankt voor je ticket en goed om te horen dat je zo tevreden bent.

Wat betreft je vragen:

Het klopt dat ik geen contact heb gehad met MinderGas zelf wat betreft historische metingen. Mijn aanname over hun beleid komt rechtstreeks van de [API-pagina daar].(https://www.mindergas.nl/member/api).

> Er zijn een paar voorwaarden verbonden aan het gebruik van de API:
> - ...
> - Alleen normaal gebruik van de API: **Het uploaden van een meterstand 1x per 1 of meer dagen.**
> - ...

Ik lees dat zelf heel duidelijk als maximaal 1 meting per dag. Dus op dat moment zag ik geen mogelijkheid om metingen met terugwerkende kracht te ondersteunen, om te voorkomen dat ik de voorwaarde zou schenden voor alle gebruikers.
Voor mijn eigen situatie heb ik van alle eerdere gasstanden (6 maanden), telkens 1 gasstand van het begin van elke maand ingevoerd op MinderGas, zodat de historische trend in ieder geval zichtbaar is.

Wat betreft de foutcodes die je krijgt. Bedoel je dat de applicatie nu geen nieuwe gasmetingen meer upload, ondanks dat je meterstanden verder lopen?

Ik zou de 422-foutcode kunnen toestaan, maar doorgaans zou het niet moeten uitmaken, omdat hij bij een volgend gasverbruik weer verder moet gaan (tenzij je zelf hogere meterstanden op de website hebt ingevoerd). 
Het punt is dat de 422-foutcode zeer generiek is voor allerlei situaties volgens MinderGas, dus ik kan het niet altijd als goed of fout markeren:

> 422 Unprocessable Entity 
> Er is een validatiefout opgetreden. Dit kan komen door:
> 
> Er is al een meterstand voor de opgegeven datum.
> De meterstand is geen getal.
> De meterstand is kleiner dan de vorige meterstand.
> De meterstand is groter dan de volgende meterstand.
> De datum ligt in de toekomst.
> De datum ligt voor 31 december 2005.
> Tip: Je kan achterhalen wat er mis is door de meterstand handmatig in te voeren op de website. De foutmelding wordt op de website getoond.

Laat maar weten waar je exact tegenaan loopt.

DaanvanderKleij, 2017-02-19:
De situatie was dat ik manueel via de mindergas app al een meterstand had ingevoerd voor de 18e. De dsmr tool bleef het proberen om de eerste meterstand te plaatsen op dezelfde datum, waar dubbele invoer niet mogelijk is (hoeveel  retries zijn mogelijk, gaat het door tot in het oneindige?). Met het verwijderen van de meterstand  op de 18e is het verholpen.  Ook ik heb manueel historische meterstanden ingevoerd, als het te automatiseren is zou dat een mooie add on zijn. Wellicht bulk import middels csv?

dennissiemensma, 2017-02-19:
Bedankt voor de toelichting. Ik zal kijken of ik in een toekomstige release hier een melding of iets van kan tonen. Doorgaans zal het uploaden van de gasstanden de dag erna weer goed gaan.

Wat betreft het in bulk uploaden van de historische meterstanden. Ik denk dat dat eerder iets voor MinderGas zelf is. Momenteel houd ik namelijk niet bij welke dagen allemaal geupload zijn en MinderGas lijkt daar ook geen API-ondersteuning voor te hebben (deze ondersteunt alleen het invoeren van gegevens, niet uitlezen).

dennissiemensma, 2018-03-05:
Afgelopen jaar zijn er al kleine verbeteringen doorgevoerd voor de koppeling met Mindergas. Niet specifiek voor dit issue, maar doorgaans krijg ik ook vrij weinig tickets over de koppeling met Mindergas. Dus voor nu laat ik dit even voor wat het is.

closed 2018-03-05 by dennissiemensma

rm78, 2019-12-16:
Ik gebruik de mindergas app in Homey.
De aangemaakte flow stuurt de meterstand niet door.
Als: tijd is na 00:01
Dan: verstuur waarde #gasmeter 
Als ik de flow test stuurt hij af en toe wel door. 

In de app juiste token ingevoerd, vinkje aan after midnight  maximum delay 12
[image]

dennissiemensma, 2019-12-16:
@rm78 ik denk dat je het beste even een los issue kan aanmaken. Je reageert nu op een oude, gesloten issue.

Verder mis ik nog wat achtergrond informatie en dat wordt ook gevraag bij een nieuw issue.