dsmrreader/dsmr-reader #263: Support for Youless P1 meter (LAN)

mbnn, 2017-02-20:
Hello,

It would be awesome if support for the Youless could be integrated in DSMR Reader.
There is a current version of the meter, but the new version (LS120) will have both energy and gas measurement through P1 port. The Youless connects to the LAN so is available from anywhere on the local network. http://wiki.td-er.nl/index.php?title=YouLess]

https://gathering.tweakers.net/forum/list_messages/1494214

dennissiemensma, 2017-02-20:
Thanks for your suggestion. To keep this project manageble I won't integrate direct support for external devices. 

I do however plan to add API support in the 1.7 release. Besides, I might create an easy API client as well and maybe some sort of (example) script to have the integration you request. 
Actually, it would be very similar to #187, which requested ESP support.

dennissiemensma, 2017-09-18:
Reconsidered it, but I won't add support for other input. Since `v1.7` there is an new version of the API available. 

Inserting data from other sources, such as Youless, can be done [using the v2 API(http://dsmr-reader.readthedocs.io/nl/latest/api.html#api-v2-restful-api).

closed 2017-09-18 by dennissiemensma