dsmrreader/dsmr-reader #264: Enabling dropbox creates a continuous datastream from approx 3 / 4 Mbps.

JeroenTielen, 2017-02-21:
I've enabled the Dropbox integration for backups. But that options creates a data stream from approx 3 to 4 Mbps. When I remove the API key this stream is gone. Is this by design? I would suspect the dropbox backup would run once a day (for example). 

[image]

In the screenshot (PRTG monitoring) is clearly visible when I disabled the dropbox integration. ;)

dennissiemensma, 2017-02-21:
Thank you for reporting this issue, as I haven't noticed this myself.

It's not by  design and I asume that there should only be traffic when it's uploading a (new) backup. I'm [using the vendor Dropbox client for Python](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.5/dsmr_backup/services/dropbox.py#L62), but I will see if I can reproduce it and resolve.

dennissiemensma, 2017-02-21:
I'm not seeing the same issue here (I think). Do you see the connections in netstat as well? 
You can view all connections with: `sudo netstat -plant`. 
It should be visible in the `Foreign Address` column. If you can not find it, it might be filtered with: `sudo netstat -plant | grep python`.

----
Another cause could be the backend keep trying to upload the same file. Could you check whether that happens? I think you should be able to see that in `/var/log/supervisor/dsmr_backend.log`. Normally it will just keep saying something like: 
```2017-02-21 19:19:40.443943+01:00: Calling backend services
Command completed. Sleeping for 5 second(s)...
```

When doing Dropbox related stuff, it should look like ` - Uploading file to Dropbox: `. Or any Dropbox/backup related message.

JeroenTielen, 2017-02-21:
Holy moly what a NOOB mistake I just did. I copied the API Key and not the generated access token. Now it is working ;) Sorry Dennis for this.

dennissiemensma, 2017-02-21:
No thank you for checking it out, as this still is a bug. 

I will schedule this ticket for a later release, to add some better error messages in the GUI or via the notification/message system in the dashboard.

JeroenTielen, 2017-02-22:
True ;) So my noob actions actually found a minor bug ;)

dennissiemensma, 2018-07-06:
Fix gemaakt voor de volgende release. Wanneer het API-token ongeldig is, komt hiervan een melding op het dashboard. Daarnaast wordt het API-token weggehaald, zodat die het niet blijft proberen (het heeft immers geen zin).

dennissiemensma, 2018-07-06:
Zit in `development`.

closed 2018-07-06 by dennissiemensma

dennissiemensma, 2018-07-11:
Meegenomen in de laatste release.