dsmrreader/dsmr-reader #265: Teruglevering optioneel verrekenen

pyrocumulus, 2017-02-26:
Het zou leuk zijn om de teruggeleverde kWh te kunnen verrekenen. Op diverse plaatsen zie je de leveringskosten maar bij teruglevering ontbreekt dit allemaal. Op de volgende plaatsen zou het nuttig zijn:

1. Dashboard - actuele teruglevering
2. Dashboard - kosten voor kWh vandaag
3. Archief - kosten voor kWh per dag

Nu is het waarschijnlijk lastig - vanwege die prachtige salderingsregeling - om zeker te weten wat voor bedrag je krijgt voor een kWh op enig moment in het jaar. Maar voor de meeste huishoudens zal een schatting middels de leveringskosten eerst wel volstaan. Alleen huishoudens die gegarandeerd op jaarbasis meer opwekken dan gebruiken zijn problematisch, omdat voor het verschil een veel lager tarief uitgekeerd wordt.

Zodra de salderingsregeling geheel afgeschaft is, is het allemaal weer duidelijk. Maar tot die tijd zou bovenstaande wel leuk kunnen zijn. Het is nu een beetje 'gek' om te zien dat twee dagen waarop ik evenveel verbruikt heb maar waarbij ik op één dag hetzelfde ook nog aanzienlijk teruggeleverd heb, de kosten voor die dag gelijk zijn.

P.S.: heel fancy dat je alles omtrent teruglevering verbergt tenzij er ook echt teruggeleverd wordt! Ik schrok na het overstappen op dsmr even omdat ik dacht dat het niet ondersteunt werd ;)

dennissiemensma, 2017-02-26:
Bedankt voor je uitgebreide verzoek. E.e.a. voor terugleveringkosten staat op de planning via #163.

Hoe dat exact gaat werken weet ik nog niet. Ook omdat, zoals je zelf al aangeeft, een deel gesaldeerd wordt.

dennissiemensma, 2017-09-28:
Deels opgepakt in #163. Zie #343 voor een screenshot. Ik heb helaas geen oplossing voor ná saldering, maar ik denk dat dat ook te complex wordt. Dus dat laat ik voor wat het is.

Wanneer je straks dezelfde prijzen instelt voor teruglevering, heb je effectief saldering (tot aan de salderingsgrens, daarna klopt het niet meer).

closed 2017-09-28 by dennissiemensma