dsmrreader/dsmr-reader #266: Data processing is lagging behind: 1 day, 14 hours ago

mkruiver, 2017-02-28:
While electricity usage is up to date, suddenly the processing of gas usage stopped working.

From the status page:

Most recent reading: now

   TELEGRAM PROCESSING 
Unprocessed telegrams: 14

   ELECTRICITY STATISTICS 
FOR DASHBOARD DISPLAY.
Data processing is on schedule: 37 seconds ago
   GAS STATISTICS 
FOR DASHBOARD DISPLAY.
Data processing is lagging behind: 1 day, 14 hours ago

dennissiemensma, 2017-02-28:
Hi, can you check whether your Supervisor logfiles state any error? I would expect any errors in the `dsmr_backend` logs. You could check the `django.log` files as well, in the `logs` directory, located inside the project root.

dennissiemensma, 2017-02-28:
And can you see whether the gas meter position on the telegrams is updated (it changes every hour)? The line containing the gas position [looks similar to this](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/tests/datalogger/test_landisgyr350_other_dsmr42.py#L42). You can see the telegrams inside the `dsmr_datalogger` process logfile or directly in Supervisor.

mkruiver, 2017-03-01:
Gas readings processing has resumed somwhere last night and the FEB 28 meter position was successfully uploaded to mindergas as well. I am only missing the FEB 27 position.
I don't see any errors in the backend or datalogger logs... The gas positions are reflected in the telegrams from the last two days.

I see one error in Django, but not related to any readings:

[2017-01-10 20:34:01,284] WARNING @ base | Not Found: /favicon.ico
[2017-01-10 20:48:35,816] WARNING @ base | Not Found: /favicon.ico
[2017-01-12 19:57:43,944] WARNING @ base | Not Found: /favicon.ico
[2017-01-14 18:47:07,336] WARNING @ base | Not Found: /favicon.ico
[2017-01-19 20:03:54,461] WARNING @ base | Not Found: /favicon.ico
[2017-02-28 13:38:53,326] WARNING @ base | Not Found: /favicon.ico

mkruiver, 2017-03-01:
I also see three logfiles I am not able to open:

pi@raspberrypimeterkast:/var/log/supervisor $ ls
dsmr_backend.log
dsmr_backend.log.1
dsmr_backend.log.2
dsmr_backend.log.3
dsmr_backend.log.4
dsmr_backend.log.5
**dsmr_backend-stderr---supervisor-ZFRVf3.log**
dsmr_datalogger.log
dsmr_datalogger.log.1
dsmr_datalogger.log.2
dsmr_datalogger.log.3
dsmr_datalogger.log.4
dsmr_datalogger.log.5
**dsmr_datalogger-stderr---supervisor-P5Stf6.log**
dsmr_webinterface.log
**dsmr_webinterface-stderr---supervisor-6gVkWu.log**
supervisord.log

dennissiemensma, 2017-03-01:
Thanks for looking into this. You might be able to see what gas readings were logged that day. Can you tell me whether you use MySQL or PostgreSQL so I can prepare a query for you?

And if I'm reading this correctly, the gas consumption is 'back to normal' now? As if nothing happened?

mkruiver, 2017-03-02:
I would have to check but I am fairly certain I'm using PostgreSQL.

The gas consumption is "back to normal" starting Feb 28, so according to
DSMR-reader, there was 0 consumption on Feb 27, which is not true....

Mindergas.nl shows the following:

Datum
Meterstand
1 mrt 2017
455,502
[image: Delete] <https://www.mindergas.nl/member/gas_meter_readings/1113696>
28 feb 2017
449,646
[image: Delete] <https://www.mindergas.nl/member/gas_meter_readings/1112211>
26 feb 2017
437,447
[image: Delete] <https://www.mindergas.nl/member/gas_meter_readings/1109200>
25 feb 2017
431,364

dennissiemensma, 2017-03-03:
Can you check whether the meter positions of that day were recorded properly? This should display it:

```
sudo -u postgres psql dsmrreader -c "select distinct read_at, delivered from dsmr_consumption_gasconsumption where read_at::text like '2017-02-27%' order by read_at desc limit 100;"
```

mkruiver, 2017-03-11:
No data was recorded on the 27th....

pi@raspberrypimeterkast:~ $ sudo -u postgres psql dsmrreader -c "select
>> distinct read_at, delivered from dsmr_consumption_gasconsumption where
>> read_at::text like '2017-02-2%' order by read_at desc limit 100;"
>
>

>         read_at         | delivered
>>
>> ------------------------+-----------
>>
>>  2017-02-28 23:55:09+01 |   449.646
>>
>>  2017-02-28 23:50:07+01 |   449.646
>>
>>  2017-02-28 23:45:05+01 |   449.646
>>
>>  2017-02-28 23:40:04+01 |   449.646
>>
>>  2017-02-28 23:35:13+01 |   449.646
>>
>>  2017-02-28 23:30:12+01 |   449.646
>>
>>  2017-02-28 23:25:04+01 |   449.646
>>
>>  2017-02-28 23:20:11+01 |   449.646
>>
>>  2017-02-28 23:15:08+01 |   449.646
>>
>>  2017-02-28 23:10:09+01 |   449.646
>>
>>  2017-02-28 23:05:07+01 |   449.646
>>
>>  2017-02-26 23:05:04+01 |   437.447
>>
>>  2017-02-26 23:00:07+01 |   437.447
>>
>>  2017-02-26 22:55:02+01 |   437.447
>>
>>  2017-02-26 22:50:01+01 |   437.447
>>
>>  2017-02-26 22:45:10+01 |   437.447
>>
>>  2017-02-26 22:40:06+01 |   437.447
>>
>>  2017-02-26 22:35:10+01 |   437.447
>>
>>  2017-02-26 22:30:03+01 |   437.447
>>
>>  2017-02-26 22:25:04+01 |   437.447
>>
>>  2017-02-26 22:20:06+01 |   437.447
>>
>>  2017-02-26 22:15:04+01 |   437.447
>>
>>  2017-02-26 22:10:07+01 |   437.447
>>
>>  2017-02-26 22:05:03+01 |   437.447
>>
>>  2017-02-26 22:00:01+01 |   437.447
>>
>>  2017-02-26 21:55:08+01 |   437.447
>>
>>  2017-02-26 21:50:04+01 |   437.447
>>
>>  2017-02-26 21:45:05+01 |   437.447
>>
>>  2017-02-26 21:40:05+01 |   437.447
>>
>>  2017-02-26 21:35:07+01 |   437.447
>>
>>  2017-02-26 21:30:03+01 |   437.437
>>
>>  2017-02-26 21:25:10+01 |   437.349
>>
>>  2017-02-26 21:20:09+01 |   437.331
>>
>>  2017-02-26 21:15:09+01 |   437.277
>>
>>  2017-02-26 21:10:05+01 |   437.221
>>
>>  2017-02-26 21:05:02+01 |   437.186
>>
>>  2017-02-26 21:00:03+01 |   437.138
>>
>>  2017-02-26 20:55:08+01 |   437.107
>>
>>
>>
>

dennissiemensma, 2017-03-11:
Can you see any other data when looking in the Archive page for the 27th? Such as electricity.

mkruiver, 2017-03-12:
Yes, elec use was recorded on the 27th in the archive page...

dennissiemensma, 2017-03-13:
Can you check the source readings stored in the database for that day? They only contain the gas meter position, so they should be changing during the day (if you consumed gas).

This will display the unique gas positions of that day:

```
sudo -u postgres psql dsmrreader -c "select extra_device_delivered from dsmr_datalogger_dsmrreading where timestamp::text like '2017-02-27%' group by extra_device_delivered limit 100;"
```

Query might take a while.

mkruiver, 2017-03-16:
sudo -u postgres psql dsmrreader -c "select extra_de
vice_delivered from dsmr_datalogger_dsmrreading where timestamp::text like
'2017
-02-27%' group by extra_device_delivered limit 100;"
 extra_device_delivered
------------------------
                437.447
(1 row)

dennissiemensma, 2017-03-16:
Thanks, it seems that there were **no** differences in the gas meter positions that day. So that explains the lack of data.
This is as close as you can look into the data source (since the raw telegrams are gone), so I can't debug any more for you regarding this issue.

Did the gas consumption/display resume normally the day after?

mkruiver, 2017-03-16:
Yes it did resume and has been working since Feb 28.

By means of slimmemetermanager I do know that meter positions did actually change during the day and have been shared with the eletric meter...

[image]

dennissiemensma, 2017-03-16:
Thanks for the additional information and I'm sorry that there seem to be differences in the data. 

Then I have absolutely no idea why the telegrams did not report a change in gas meter positions (*to  dsmr-reader*).  Even when it's lagging behind in processing the data, it should be able to parse the gas meter positions retroactivily.
The query you executed displays all unique gas meter positions it has captured. You can run it for another day for comparison, but there is nothing I can do in this case. 

I.e. the day after:
```
sudo -u postgres psql dsmrreader -c "select extra_de
vice_delivered from dsmr_datalogger_dsmrreading where timestamp::text like
'2017
-02-28%' group by extra_device_delivered limit 50;"
```

Please let me know when you **do** see this happening again. If possible, try to preserve some of the `dsmrreader.log` files in the `logs` directory (when they are not rotated already due to the many telegrams you receive).

closed 2017-03-16 by dennissiemensma