dsmrreader/dsmr-reader #268: Data preservation/backups (readthedocs.io)

WatskeBart, 2017-02-28:
Maybe add a part on how to setup a USB drive for [data preservation/backups](https://dsmr-reader.readthedocs.io/en/latest/application.html#data-preservation-backups).

I did the following, using a FAT32 formatted USB drive;

Create a new data directory
`sudo mkdir /data`

Check which device i'm using
`sudo blkid`

Mount the device to /data (in my case sda1)
`sudo mount /dev/sda1 /data`

Test if backup script is working
`sudo sh /home/dsmr/dsmr-reader/dsmrreader/provisioning/postgresql/psql-backup.sh`

Check if backup is written
`ls /data` 
a **backup-postgres-[day of week].sql.gz** file should be created

Make sda1 automatically mount on each reboot
`sudo nano /etc/fstab`

Add the following line below the other devices (i'm using a FAT32 formatted device so i set the flag to **vfat**)
`/dev/sda1 /mnt /vfat defaults 0 0`
Press Alt+X, Press Y, Press Enter

Add the backup script to cron.daily
`sudo nano dsmr`

Add the following lines
```
#!/bin/sh

sh /home/dsmr/dsmr-reader/dsmrreader/provisioning/postgresql/psql-backup.sh
```
Press Alt+X, Press Y, Press Enter

Change permissions to make it executable
`sudo chmod +x dsmr`

Test the cron job
`sudo run-parts -v /etc/cron.daily/`

Maybe it's something to add to the documentation.

dennissiemensma, 2017-03-01:
Thanks for your effort in describing this! I will add it to the docs in an upcoming release.

janvanhelvoort, 2017-04-29:
Note:

The commands for the cron.daily need to execute in the `/etc/cron.daily` directory

dennissiemensma, 2017-04-29:
Thank you for the additional information.

janvanhelvoort, 2017-05-05:
after modification of the file `/etc/fstab`, after a reboot of my pi, it doesn't boot.
Use these staps: [Link](http://www.techjawab.com/2013/06/how-to-setup-mount-auto-mount-usb-hard.html)

dennissiemensma, 2018-01-16:
Alles overnemen werd me wat te veel voor de pagina, daarom heb ik er een verwijzing in opgenomen naar het verhaal hier, voor de volgende release. Nogmaals bedankt.

closed 2018-01-16 by dennissiemensma

T1rnanog, 2019-12-21:
Hello Dennis,
I would really like to install the backup to usb drive.
Following the readme got me to this topic. My knowlege goes as far as copy pasting the above lines.
I'm afraid to start this because of the comment from 'janvanhelvoort' the 'Use these staps: Link' the link isn't working anymore. 

Is it possible to update it so i can follow it? 
Thank you

dennissiemensma, 2019-12-22:
@T1rnanog you can try using the Internet Archive, sometimes the page is still available there: 

https://web.archive.org/web/20190516134732/http://www.techjawab.com/2013/06/how-to-setup-mount-auto-mount-usb-hard.html

barrymossel, 2021-01-05:
> Make sda1 automatically mount on each reboot
> `sudo nano /etc/fstab`
> 
> Add the following line below the other devices (i'm using a FAT32 formatted device so i set the flag to **vfat**)
> `/dev/sda1 /mnt /vfat defaults 0 0`
> Press Ctrl+X, Press Y, Press Enter

Shouldn't that be:

`/dev/sda1 /data /vfat defaults 0 0`

**(/data instead of /mnt)**