dsmrreader/dsmr-reader #269: x-as gasgrafiek geeft rare waarden aan

mkruiver, 2017-03-01:
Het laatste label op de x-as van mijn gasgrafiek is altijd 3:02 AM, terwijl die voor stroom wel normaal werkt.
[image]

dennissiemensma, 2017-03-01:
Ik denk dat dit mogelijk gerelateerd is aan je andere ticket over de missende gasmetingen. Kun je aangeven welk soort database je gebruikt? En ook welke versie van DSMR-reader? (staat rechtsbovenin)

mkruiver, 2017-03-02:
Dit issue bestond al voor de missende gasmetingen, op 1.5.4.
Na update naar 1.5.5 hetzelfde.
Op Postgres.
Ik heb wel al eerder eens mislukkende mindergas.nl updates gehad (zoals
gerapporteerd) alsook een gasmeter die de eerste twee weken nog geen
gasstanden doorgaf via Modbus. Gasmeter werkt nu goed, gasstand wordt ook
gereflecteerd in de telegrams.

dennissiemensma, 2017-03-03:
Hmmm het zou ook kunnen dat de volgorde van de data niet lekker loopt, maar dat is een aanname. Kun je eens kijken wat de laatste gasdata is, in twee varianten? Eentje op basis van datum+tijd en eentje op ID. Ze zouden gelijk moeten lopen.

*Tijd*
```
sudo -u postgres psql dsmrreader -c "select id, read_at, delivered, currently_delivered from dsmr_consumption_gasconsumption order by read_at desc limit 24;"
```
*ID*
```
sudo -u postgres psql dsmrreader -c "select id, read_at, delivered, currently_delivered from dsmr_consumption_gasconsumption order by id desc limit 24;"
```

mkruiver, 2017-03-11:
Het probleem zit hem duidelijk in de tijd...!

ID is goed:

>  id   |        read_at         | delivered | currently_delivered
>
> -------+------------------------+-----------+---------------------
>
>  17556 | 2017-03-11 08:40:04+01 |   503.148 |               0.000
>
>  17555 | 2017-03-11 08:35:06+01 |   503.124 |               0.477
>
>  17554 | 2017-03-11 08:30:07+01 |   503.100 |               0.000
>
>  17553 | 2017-03-11 08:25:08+01 |   503.045 |               0.000
>
>  17552 | 2017-03-11 08:20:10+01 |   503.004 |               0.000
>
>  17551 | 2017-03-11 08:15:12+01 |   502.944 |               0.000
>
>  17550 | 2017-03-11 08:10:11+01 |   502.931 |               0.000
>
>  17549 | 2017-03-11 08:05:04+01 |   502.905 |               0.000
>
>  17548 | 2017-03-11 08:00:02+01 |   502.882 |               0.000
>
>  17547 | 2017-03-11 07:55:10+01 |   502.843 |               0.000
>
>  17546 | 2017-03-11 07:50:07+01 |   502.818 |               0.000
>
>  17545 | 2017-03-11 07:45:08+01 |   502.746 |               0.000
>
>  17544 | 2017-03-11 07:40:05+01 |   502.681 |               0.000
>
>  17543 | 2017-03-11 07:35:06+01 |   502.647 |               0.000
>
>  17542 | 2017-03-11 07:30:03+01 |   502.633 |               0.000
>
>  17541 | 2017-03-11 07:25:06+01 |   502.579 |               0.000
>
>  17540 | 2017-03-11 07:20:07+01 |   502.559 |               0.000
>
>  17539 | 2017-03-11 07:15:11+01 |   502.542 |               0.000
>
>  17538 | 2017-03-11 07:10:08+01 |   502.462 |               0.000
>
>  17537 | 2017-03-11 07:05:10+01 |   502.442 |               0.000
>
>  17536 | 2017-03-11 07:00:07+01 |   502.360 |               0.000
>
>  17535 | 2017-03-11 06:55:06+01 |   502.338 |               0.000
>
>  17534 | 2017-03-11 06:50:10+01 |   502.207 |               0.000
>
>  17533 | 2017-03-11 06:45:03+01 |   502.064 |               0.000
>
> (24 rows)
>
>
>

Maar tijd klopt niet:

  id   |        read_at         | delivered | currently_delivered

-------+------------------------+-----------+---------------------

  2470 | 2054-12-18 03:02:01+01 |   144.197 |               0.000

  5901 | 2024-07-08 17:25:36+02 |   239.844 |               0.000

 12914 | 2024-04-03 07:35:11+02 |   402.487 |               0.000

 17556 | 2017-03-11 08:40:04+01 |   503.148 |               0.000

 17555 | 2017-03-11 08:35:06+01 |   503.124 |               0.477

 17554 | 2017-03-11 08:30:07+01 |   503.100 |               0.000

 17553 | 2017-03-11 08:25:08+01 |   503.045 |               0.000

 17552 | 2017-03-11 08:20:10+01 |   503.004 |               0.000

 17551 | 2017-03-11 08:15:12+01 |   502.944 |               0.000

 17550 | 2017-03-11 08:10:11+01 |   502.931 |               0.000

 17549 | 2017-03-11 08:05:04+01 |   502.905 |               0.000

 17548 | 2017-03-11 08:00:02+01 |   502.882 |               0.000

 17547 | 2017-03-11 07:55:10+01 |   502.843 |               0.000

 17546 | 2017-03-11 07:50:07+01 |   502.818 |               0.000

 17545 | 2017-03-11 07:45:08+01 |   502.746 |               0.000

 17544 | 2017-03-11 07:40:05+01 |   502.681 |               0.000

 17543 | 2017-03-11 07:35:06+01 |   502.647 |               0.000

 17542 | 2017-03-11 07:30:03+01 |   502.633 |               0.000

 17541 | 2017-03-11 07:25:06+01 |   502.579 |               0.000

 17540 | 2017-03-11 07:20:07+01 |   502.559 |               0.000

 17539 | 2017-03-11 07:15:11+01 |   502.542 |               0.000

 17538 | 2017-03-11 07:10:08+01 |   502.462 |               0.000

 17537 | 2017-03-11 07:05:10+01 |   502.442 |               0.000

 17536 | 2017-03-11 07:00:07+01 |   502.360 |               0.000

(24 rows)

dennissiemensma, 2017-03-11:
Hmm heel vreemd, zoiets heb ik nog nooit gezien. Wellicht dat er toch ooit corrupte data of iets is binnengekomen. 

Hoe dan ook, de oplossing is om de foute gegevens te verwijderen, gezien ze de sortering van de applicatie in de wat gooien. Gezien het er maar 3 zijn, is dat nog wel makkelijk te doen:

```
sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = 2470 limit 1;"

sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = 5901 limit 1;"

sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = 12914 limit 1;"
```

Wellicht goed om even ervoor te zorgen dat je een goede database backup hebt, ondanks dat er weinig mis kan gaan, want de query verwijdert maximaal 1 record per keer (limit 1).

dennissiemensma, 2017-03-11:
Ik zie trouwens dat je elke 5 minuten een gasmeting hebt. Ik dacht dat de applicatie dat nog niet toestond (ik heb er daarom ook iets voor gemaakt in v1.6), maar dat werkt dus al goed verder?

mkruiver, 2017-03-11:
Ze komen goed binnen maar de grafiek laat ze hooguit één keer per uur zien.

dennissiemensma, 2017-03-12:
Ok bedankt voor de informatie daarover :]. 

Werkt de fix met de queries uit mijn eerdere comment?

mkruiver, 2017-03-12:
Er lijkt een syntax error in de query te zitten:

>  sudo -u postgres psql dsmrreader -c "DELETE from
> dsmr_consumption_gasconsumption where id = 5901 limit 1;"
> ERROR:  syntax error at or near "limit"
> LINE 1: ...rom dsmr_consumption_gasconsumption where id = 5901 limit 1;
>                                                                ^
> pi@raspberrypimeterkast:~ $
>

pyrocumulus, 2017-03-13:
Ik heb ditzelfde probleem sindskort. De 'nieuwste' tijd in mijn gas-grafiek laat nu permanent 12:06 zien. Ik denk dat het wel eens met datacorruptie te maken zou kunnen hebben. Bij mij was laatst één van de dsmr processen blijkbaar omgevallen waardoor ik ruim duizend telegrammen achter liep. Volgens mij gaat het sinds dat moment mis met de weergave. Ik heb overigens ook elke 5 minuten gasmetingen dankzij DSMR v5 maar de dataverwerking is inderdaad eenmaal per uur en dus de grafiek zelf ook.

Zal vanavond die queries eens draaien om te kijken of de oorzaak hetzelfde is.

dennissiemensma, 2017-03-13:
@mkruiver excuses, ik was in de war met MySQL, waarbij een `limit 1` wel is toegestaan. Voor PostgreSQL is het dezelfde query, zonder de limit:

```
sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = 2470;"

sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = 5901;"

sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = 12914;"
```

dennissiemensma, 2017-03-13:
@pyrocumulus grote kans dat daar hetzelfde aan de hand is. Je kunt de foute ID's verwijderen met:

```
sudo -u postgres psql dsmrreader -c "DELETE from dsmr_consumption_gasconsumption where id = ID-VAN-GASMETING;"
```

Waarbij `ID-VAN-GASMETING` uiteraard het ID getal is.

pyrocumulus, 2017-03-13:
Inderdaad, zelfde probleem. Twee records met een datum ver in de toekomst:

id | read_at | delivered | currently_delivered	
---|--------|-----------|------------
4039 | 2057-01-12 11:06:14 | 342.116 | 0.000	
5176 | 2043-04-08 07:39:40 | 369.966 | 0.000	
5479 | 2017-03-13 20:35:02 | 375.095 | 0.000	
5478 | 2017-03-13 20:30:03 | 375.095 | 0.000	

Waar haal je die datum en tijd vandaan; uit het telegram? Ik kan me niet voorstellen dat pi/python dat ineens verzint namelijk :) Anyhow, ik heb beide weggegooid en nu is de weergave weer in orde!

Bedankt voor de hulp.

dennissiemensma, 2017-03-14:
Ze komen uit het telegram. 

Het is mogelijk dat data corrupt is geraakt tijdens het transport over de kabel, echter als het goed is zit er sinds december CRC-checking in die standaard aanstaat voor DSMR v4+. Dat is een soort checksum die bevestigt of het bericht nog intact is.
Alleen is het wel erg toevallig dat drie verschillende personen, die ook nog eens wat op Github hebben achtergelaten én DSMR v5 hebben, er allemaal in zekere zin last van hebben. Helaas slaat de applicatie de oorspronkelijke telegrammen slechts kort op, anders hadden we daar in kunnen kijken. Voor (nu) laat ik het voor wat het is dan. Ook gezien het geringe aantal berichten waar het over gaat...

mkruiver, 2017-03-16:
Dit heeft inderdaad gewerkt, de grafiek is weer normaal, bedankt!

dennissiemensma, 2017-03-16:
Oke bedankt voor het melden, ik had dit zelf nog niet eerder gezien!

mkruiver, 2017-04-13:
Dit issue is opnieuw opgedoken na update van 1.5.5 naar 1.6.1.

[image]

[image]

Ondanks dat er op dit moment geen gas wordt verbruikt, denkt DSMR reader dat er 40m3 geleverd wordt elke 5 minuten.

  id   |        read_at         | delivered | currently_delivered
-------+------------------------+-----------+---------------------
 26596 | 2017-04-13 21:30:00+02 |   568.342 |              40.216
 26595 | 2017-04-13 21:25:00+02 |   568.342 |              40.216
 26594 | 2017-04-13 21:20:00+02 |   568.342 |              40.216
 26593 | 2017-04-13 21:15:00+02 |   568.342 |              40.216
 26592 | 2017-04-13 21:10:00+02 |   568.342 |              40.216
 26591 | 2017-04-13 21:05:00+02 |   568.342 |              40.216
 26590 | 2017-04-13 21:00:00+02 |   568.342 |              40.216
 26589 | 2017-04-13 20:55:00+02 |   568.342 |              40.216
 26588 | 2017-04-13 20:50:00+02 |   568.342 |              40.216
 26587 | 2017-04-13 20:45:00+02 |   568.342 |              40.216
 26586 | 2017-04-13 20:40:00+02 |   568.342 |              40.216
 26585 | 2017-04-13 20:35:00+02 |   568.342 |              40.216
 26584 | 2017-04-13 20:30:00+02 |   568.342 |              40.216
 26583 | 2017-04-13 20:25:00+02 |   568.342 |              40.216
 26582 | 2017-04-13 20:20:00+02 |   568.342 |              40.216
 26581 | 2017-04-13 20:15:00+02 |   568.342 |              40.216
 26580 | 2017-04-13 20:10:00+02 |   568.342 |              40.216
 26579 | 2017-04-13 20:05:00+02 |   568.342 |              40.216
 26578 | 2017-04-13 20:00:00+02 |   568.342 |              40.216
 26577 | 2017-04-13 19:55:00+02 |   568.342 |              40.216
 26576 | 2017-04-13 19:50:00+02 |   568.342 |              40.216
 26575 | 2017-04-13 19:45:00+02 |   568.342 |              40.216
 26574 | 2017-04-13 19:40:00+02 |   568.342 |              40.216
 26573 | 2017-04-13 19:35:00+02 |   568.342 |              40.216
(24 rows)

  id   |        read_at         | delivered | currently_delivered
-------+------------------------+-----------+---------------------
 19369 | 2070-03-16 13:38:03+01 |   528.126 |               0.000
 26596 | 2017-04-13 21:30:00+02 |   568.342 |              40.216
 26595 | 2017-04-13 21:25:00+02 |   568.342 |              40.216
 26594 | 2017-04-13 21:20:00+02 |   568.342 |              40.216
 26593 | 2017-04-13 21:15:00+02 |   568.342 |              40.216
 26592 | 2017-04-13 21:10:00+02 |   568.342 |              40.216
 26591 | 2017-04-13 21:05:00+02 |   568.342 |              40.216
 26590 | 2017-04-13 21:00:00+02 |   568.342 |              40.216
 26589 | 2017-04-13 20:55:00+02 |   568.342 |              40.216
 26588 | 2017-04-13 20:50:00+02 |   568.342 |              40.216
 26587 | 2017-04-13 20:45:00+02 |   568.342 |              40.216
 26586 | 2017-04-13 20:40:00+02 |   568.342 |              40.216
 26585 | 2017-04-13 20:35:00+02 |   568.342 |              40.216
 26584 | 2017-04-13 20:30:00+02 |   568.342 |              40.216
 26583 | 2017-04-13 20:25:00+02 |   568.342 |              40.216
 26582 | 2017-04-13 20:20:00+02 |   568.342 |              40.216
 26581 | 2017-04-13 20:15:00+02 |   568.342 |              40.216
 26580 | 2017-04-13 20:10:00+02 |   568.342 |              40.216
 26579 | 2017-04-13 20:05:00+02 |   568.342 |              40.216
 26578 | 2017-04-13 20:00:00+02 |   568.342 |              40.216
 26577 | 2017-04-13 19:55:00+02 |   568.342 |              40.216
 26576 | 2017-04-13 19:50:00+02 |   568.342 |              40.216
 26575 | 2017-04-13 19:45:00+02 |   568.342 |              40.216
 26574 | 2017-04-13 19:40:00+02 |   568.342 |              40.216
(24 rows)

mkruiver, 2017-04-13:
Het lijkt gefixt door ID 19369 uit de db te verwijderen en vervolgens wat gas usage te triggeren.

Hier zie je de spookreading ontstaan na de update:

 26302 | 2017-04-12 21:00:00+02 |   568.342 |              40.216
 26301 | 2017-04-12 20:55:00+02 |   568.342 |              40.216
 26300 | 2017-04-12 20:50:00+02 |   568.342 |              40.216
 26299 | 2017-04-12 20:45:00+02 |   568.342 |              40.216
 26298 | 2017-04-12 19:45:01+02 |   568.342 |               0.000
 26297 | 2017-04-12 19:39:59+02 |   568.342 |               0.000
 26296 | 2017-04-12 19:35:02+02 |   568.342 |               0.000
 26295 | 2017-04-12 19:30:04+02 |   568.342 |               0.000

En nu lijkt hij weg te zijn (hiervoor was er sinds de update 0 gas verbruikt):

 id   |        read_at         | delivered | currently_delivered
-------+------------------------+-----------+---------------------
 26599 | 2017-04-13 21:45:00+02 |   568.585 |               0.183
 26598 | 2017-04-13 21:40:00+02 |   568.402 |               0.060
 26597 | 2017-04-13 21:35:00+02 |   568.342 |              40.216
 26596 | 2017-04-13 21:30:00+02 |   568.342 |              40.216

mkruiver, 2017-04-13:
Overigens kwam ook voor het triggeren van een beetje gasverbruik de gasstand wel gewoon goed door in de telegrammen:

/ISK5\2M550T-1011

1-3:0.2.8(50)
0-0:1.0.0(170413211146S)
0-0:96.1.1(-)
1-0:1.8.1(000230.768*kWh)
1-0:1.8.2(000215.860*kWh)
1-0:2.8.1(000108.129*kWh)
1-0:2.8.2(000353.355*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.074*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00005)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(228.4*V)
1-0:52.7.0(231.8*V)
1-0:72.7.0(230.4*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.034*kW)
1-0:41.7.0(00.006*kW)
1-0:61.7.0(00.034*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0()
0-1:24.2.1(170413211002S)(00568.342*m3)

dennissiemensma, 2017-04-13:
Bedankt voor het melden! Het blijft raar dat deze rare datums blijven voorkomen.

Ik zal voor het berekenen van het gasverbruik t.z.t. een fix maken. Want hij vraagt eigenlijk elke keer de meeste recent (vorige) meting op en dat is met `19369 | 2070-03-16 13:38:03+01 | 528.126 | 0.000` natuurlijk altijd die foute. Het is vrij makkelijk om de applicatie geen metingen (ver) in de toekomst toe te laten staan in dit geval.

Daarnaast ben ik nog steeds benieuwd hoe het telegram er uitziet van die foute meting. Gezien het onmogelijk is dat zelf handmatig te spotten, zal ik kijken of ik daar ook iets voor kan maken. Want ik neem aan dat je gewoon CRC-checks aan hebt staan?

mkruiver, 2017-04-13:
Ja crc staat aan en het aantal rejected telegrams staat nog op 12, net als
voor de update.

FutureCow, 2017-04-15:
Hier zelfde probleem, al een keer eerder gehad en verwijderd. Maar ondertussen staan er weer 2 verkeerde in. Na verwijderen is de grafiek weer goed!
heb nog gekeken of ik telegram kon vinden. De foute telegram is van 3 dagen geleden, helaas staat deze niet meer in de logs.

Vescatur, 2017-04-23:
Hier het zelfde probleem 2 waardes in de toekomst en een onjuiste grafiek, heb de 2 toekomst waardes verwijderd en nu is de grafiek weer juist. Heb ook een  DSMR v5 meter en gebruik DSMR Reader versie 1.6.1.

mkruiver, 2017-04-23:
Shit, bij mij is het sinds zaterdagochtend​ 7.20AM weer.... Toekomstige
waarde verwijderd en reboot was genoeg

dennissiemensma, 2017-04-23:
Gezien dit steeds vaker en regelmatig lijkt voor te komen, zal ik dit meenemen in een tussentijdse patch. Bedankt voor het melden.

dennissiemensma, 2017-04-23:
Ik heb een mogelijke oplossing uitgebracht in patch `v1.6.2`. 

Willen jullie updaten en kijken of het daarná weer voorkomt? De applicatie staat nu geen telegrammen meer toe die verder dan 24 uur in de toekomst zijn. Mocht zo'n telegram wél binnenkomen, dan wordt die gelogd in `logs/django.log` (wellicht goed om daar na een paar weken te kijken).

Ik hoor graag wanneer blijkt dat dit het niet oplost. :]

closed 2017-04-23 by dennissiemensma

mkruiver, 2017-04-23:
Bedankt voor de snelle actie! Ik heb geüpdatet, keep you posted!

pyrocumulus, 2017-04-23:
Hier ook meteen een update uitgevoerd. Bedankt! Ik hou het in de gaten :)

Edwink18, 2017-04-27:
Woensdag toch weer een verkeerde waarde vandaag geupdate naar 1.6.2 maar had ook al een backup lopen van de P1 berichten.  Het lijkt er op dat de datum niet helemaal correct overkomt. ik heb een paar 
berichten toegevoegd maar dit gaat 5min door/

1-0:22.7.0(01.280*kW)^M
1-0:42.7.0(01.506*kW)^M
1-0:62.7.0(01.307*kW)^M
0-1:24.1.0(003)^M
0-1:96.1.0(4730303332353635353031393236343136)^M
0-1:24.2.1(170426125003S)(01023.855*m3)^M
!0013^M
Command completed. Sleeping for 0.25 second(s)...

/ISK5\2M550T-1011^M
^M
1-3:0.2.8(50)^M
0-0:1.0.0(170426125505S)^M
0-0:96.1.1(4530303334303036373536333733343136)^M
1-0:1.8.1(001245.419*kWh)^M
1-0:1.8.2(000825.615*kWh)^M
1-0:2.8.1(000191.226*kWh)^M
1-0:2.8.2(000795.499*kWh)^M
0-0:96.14.0(0002)^M
1-0:1.7.0(00.000*kW)^M
1-0:2.7.0(04.110*kW)^M
0-0:96.7.21(00006)^M
0-0:96.7.9(00005)^M
1-0:99.97.0()^M
1-0:32.32.0(00000)^M
1-0:52.32.0(00000)^M
1-0:72.32.0(00000)^M
1-0:32.36.0(00001)^M
1-0:52.36.0(00001)^M
1-0:72.36.0(00001)^M
0-0:96.13.0()^M
1-0:32.7.0(232.3*V)^M
1-0:52.7.0(231.9*V)^M
1-0:72.7.0(231.3*V)^M
1-0:31.7.0(005*A)^M
1-0:51.7.0(006*A)^M
1-0:71.7.0(005*A)^M
1-0:21.7.0(00.000*kW)^M
1-0:41.7.0(00.000*kW)^M
1-0:61.7.0(00.000*kW)^M
1-0:22.7.0(01.280*kW)^M
1-0:42.7.0(01.508*kW)^M
1-0:62.7.0(01.308*kW)^M
0-1:24.1.0(003)^M
0-1:96.1.0(4730303332353635353031393236343136)^M
0-1:24.2.1(681027002758S)(01023.855*m3)^M
!CD75^M
Command completed. Sleeping for 0.25 second(s)...

/ISK5\2M550T-1011^M
^M
1-3:0.2.8(50)^M
0-0:1.0.0(170426125506S)^M
0-0:96.1.1(4530303334303036373536333733343136)^M
1-0:1.8.1(001245.419*kWh)^M
1-0:1.8.2(000825.615*kWh)^M
1-0:2.8.1(000191.226*kWh)^M
1-0:2.8.2(000795.500*kWh)^M
0-0:96.14.0(0002)^M
1-0:1.7.0(00.000*kW)^M
1-0:2.7.0(04.114*kW)^M
0-0:96.7.21(00006)^M
0-0:96.7.9(00005)^M
1-0:99.97.0()^M
1-0:32.32.0(00000)^M
1-0:52.32.0(00000)^M
1-0:72.32.0(00000)^M
1-0:32.36.0(00001)^M
1-0:52.36.0(00001)^M
1-0:72.36.0(00001)^M
0-0:96.13.0()^M
1-0:32.7.0(232.6*V)^M
1-0:52.7.0(232.1*V)^M
1-0:72.7.0(231.3*V)^M
1-0:31.7.0(005*A)^M
1-0:51.7.0(006*A)^M
1-0:71.7.0(005*A)^M
1-0:21.7.0(00.000*kW)^M
1-0:41.7.0(00.000*kW)^M
1-0:61.7.0(00.000*kW)^M
1-0:22.7.0(01.283*kW)^M
1-0:42.7.0(01.508*kW)^M
1-0:62.7.0(01.305*kW)^M
0-1:24.1.0(003)^M
0-1:96.1.0(4730303332353635353031393236343136)^M
0-1:24.2.1(681027002758S)(01023.855*m3)^M
!F2D0^M
Command completed. Sleeping for 0.25 second(s)...

dennissiemensma, 2017-04-27:
Bedankt voor je melding. Ik begrijp niet helemaal wat je bedoelt, maar je kunt eventuele foute metingen verwijderen via de queries in de eerdere comments (let op, andere ID's!).

Als het goed is komen er geen vage metingen meer voor sinds je update naar de laatste versie. Mocht je hulp nodig hebben hoor ik het graag.

Edwink18, 2017-04-27:
Dennis,

Meer als ter kennisgeving, de fout lijkt dus in de gasmeting datum te
zitten.

Edwin.

dennissiemensma, 2017-04-27:
Je zou over een paar dagen kunnen kijken of de nieuwste versie zo'n afwijking heeft gelogd. Deze zou dan in `logs/django.log` moeten staan met een melding als `Discarded telegram with future timestamp`.

Edwink18, 2017-04-27:
Ik hou het in de gate.