dsmrreader/dsmr-reader #270: External access

WatskeBart, 2017-03-03:
Another possible addition to the docs. ([public-webinterface-warning](https://dsmr-reader.readthedocs.io/nl/latest/application.html#public-webinterface-warning))

Instead of have people copy pasting from a website, why not generate the htpasswd file on the Pi itself. No need for a website and the possible trust issue's people might experience.

I did the following;

Install the Apache2 utilities, which contains the tool **htpasswd**
`sudo apt-get install apache2-utils`

Run htpasswd to generate a new htpasswd file inside the nginx directory
`sudo htpasswd -c /etc/nginx/htpasswd YOUR-USERNAME`
_You will be prompted to enter your desired password twice_

Reload the nginx service
`sudo services nginx reload`

Alternatively, if you like multiple users to access the webpage you can do the following;

`sudo htpasswd /etc/nginx/htpasswd ANOTHER-USERNAME` **_notice the -c is not used this time_**
_This will update the current htpasswd file and add another user_

Off course reload the nginx service again
`sudo services nginx reload`

dennissiemensma, 2017-03-05:
Thank you for the contribution. I will see if I can add it to the FAQ later.

dennissiemensma, 2017-09-18:
I've added this to the docs for the upcoming release, thanks again.

closed 2017-09-18 by dennissiemensma