dsmrreader/dsmr-reader #271: IntegrityError: (1048, "Column 'electricity_delivered_1' cannot be null") when using API

michielvisser, 2017-03-04:
I've installed DSMRreader version 1.5.5 (latest version as of today) and am trying to get data into it by using the API. However, whatever I try I always end up with the same error:

IntegrityError: (1048, "Column 'electricity_delivered_1' cannot be null")

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/handlers/base.py", line 185, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/decorators/csrf.py", line 58, in wrapped_view
    return view_func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 68, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 88, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_api/views.py", line 39, in post
    dsmr_reading = dsmr_datalogger.services.telegram_to_reading(data=post_form.cleaned_data['telegram'])

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 250, in telegram_to_reading
    new_reading = DsmrReading.objects.create(**reading_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 399, in create
    obj.save(force_insert=True, using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 796, in save
    force_update=force_update, update_fields=update_fields)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 824, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 908, in _save_table
    result = self._do_insert(cls._base_manager, using, fields, update_pk, raw)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 947, in _do_insert
    using=using, raw=raw)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 1045, in _insert
    return query.get_compiler(using=using).execute_sql(return_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1054, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/mysql/base.py", line 115, in execute
    six.reraise(utils.IntegrityError, utils.IntegrityError(*tuple(e.args)), sys.exc_info()[2])

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/mysql/base.py", line 110, in execute
    return self.cursor.execute(query, args)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py", line 250, in execute
    self.errorhandler(self, exc, value)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/connections.py", line 42, in defaulterrorhandler
    raise errorvalue

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py", line 247, in execute
    res = self._query(query)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py", line 411, in _query
    rowcount = self._do_query(q)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/cursors.py", line 374, in _do_query
    db.query(q)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/MySQLdb/connections.py", line 270, in query
    _mysql.connection.query(self, query)

First I tried on a CentOS 7 server and it gave the eror. As I had some doubts about my installation I tried an Raspberry Pi 2 with a fresh installation of Raspbian Jessie (2017-03-02) and followed the instructions to the letter. This gave the same error. Even when I try the example telegram (and disable check CRC in the admin) the above error is raised.

dennissiemensma, 2017-03-05:
Hi, thank you for reporting your issue. The operating system should not matter. The error you are experiencing is related to the telegram lacking some data. 

Could you post a sample of the telegram you are sending to the API? 

Please make sure to **remove** any serials of the devices used. The serial numbers usually are lines with around 30 digits and start with either `0-0:96.1.1` or `0-1:96.1.0`.

michielvisser, 2017-03-05:
Hi Dennis,

Thanks for the quick reply. Here in the telegram I'm trying to send:

/XMX5LGBBFF**********

1-3:0.2.8(42)
0-0:1.0.0(170304123043W)
0-0:96.1.1(**********************************)
1-0:1.8.1(003009.835*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(002616.606*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.216*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(150707080827S)(0000000644*s)(150505163026S)(0000004222*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:21.7.0(00.217*kW)
1-0:22.7.0(00.000*kW)
!

For testing I've created the following cURL line from this:

curl http://192.168.50.45/api/v1/datalogger/dsmrreading -d 'telegram=/XMX5LGBBFF**********\r\n\r\n1-3:0.2.8(42)\r\n0-0:1.0.0(170304123043W)\r\n0-0:96.1.1(**********************************)\r\n1-0:1.8.1(003009.835*kWh)\r\n1-0:2.8.1(000000.000*kWh)\r\n1-0:1.8.2(002616.606*kWh)\r\n1-0:2.8.2(000000.000*kWh)\r\n0-0:96.14.0(0001)\r\n1-0:1.7.0(00.216*kW)\r\n1-0:2.7.0(00.000*kW)\r\n0-0:96.7.21(00006)\r\n0-0:96.7.9(00002)\r\n1-0:99.97.0(2)(0-0:96.7.19)(150707080827S)(0000000644*s)(150505163026S)(0000004222*s)\r\n1-0:32.32.0(00000)\r\n1-0:32.36.0(00000)\r\n0-0:96.13.1()\r\n0-0:96.13.0()\r\n1-0:31.7.0(001*A)\r\n1-0:21.7.0(00.217*kW)\r\n1-0:22.7.0(00.000*kW)\r\n!\n' -H 'X-AUTHKEY: auth-key-goes-here'

Maybe you can spot something that is missing?

At this moment, after quite some thinking this morning, I'm a bit suspicious about the method I'm using to retrieve the telegrams (this one is retrieved using an Arduino with a sketch based on the API scripts in the documentation). At the moment, I'm building another cable for plugging into the smartmeter and will try to retrieve a telegram directly with an USB to serial cable to see whether there are any differences.

Will keep you posted on that one.

michielvisser, 2017-03-05:
OK, I figured it out. It is not a problem with the application but with the method of data collection... At least sort of.

I've modified the collection of the telegrams to include the checksum at the end and now the API is willing to take my telegrams and it spits out a 201 Created instead of a 500 Internal Server Error (i'm running the development branch now). This is independent on activating or deactiving 'Verify CRC' in the datalogger configuration.  

Problem solved. Now I know what the problem is I can have the benefit of hindsight. This issue kept me puzzled for two days... :-(

Sidetrack: I've made notes on how to install a CentOS 7 machine with the DSMR reader and setting it up correctly (there are quite a few differences to the RPi installation with Ubuntu). Maybe I can contribute these notes somewhere for the benefit of the next person who want to do this? The same goes for the Arduino sketch I'm using to collect telegrams and sending them to the API.

dennissiemensma, 2017-03-05:
Thank you for providing the clarification and good to hear that you've solved it yourself already. 

Any documentation for other platforms is always welcome. You might want to create a new issue for that and attach or insert the notes you've taken. I cannot process it immediately but I will queue it for a future release. :]

closed 2017-03-05 by dennissiemensma