dsmrreader/dsmr-reader #272: Dashboard - weergave huidig verbruik bij smalle weergave

pyrocumulus, 2017-03-06:
Bovenin het Dashboard wordt het actuele verbruik of opgewekte vermogen getoond. Dit wordt vooraf gegaan door een label dat de tijd sinds de laatste meting laat zien. Of is het de tijd tot de volgende update? Dat is me nooit geheel duidelijk.

Hoe dan ook: dit label varieert qua tekst van 'nu' tot 'over een seconde' of variaties van dat laatste. Doordat de tekst daartussen varieert neemt dit soms heel weinig ruimte in en andere keren veel meer. Bij smalle weergaven - zoals op een mobiel - zorgt dat ervoor dat de statistieken de hele tijd verspringen tussen verschillende posities, zoals:

[image]

Is een beetje storend als je de actuele statistieken even probeert te volgen. Wellicht dat het tekst label een vaste breedte kan krijgen.

dennissiemensma, 2017-03-07:
Bedankt voor het melden van dit issue. Ik zal ervoor zorgen dat het label een vaste breedte krijgt, zoals je voorstelt. :]

pyrocumulus, 2017-03-22:
Nog een kleine aanvulling hierop. Met #207 heb je een klein probleem opgelost met slimme meters die qua tijd een beetje voorlopen. Dat werkt nu in 1.6 inderdaad goed op de status pagina.

Maar ik heb het idee dat het bovenstaande label (dat verspringt) nog wel die 'negatieve' tijd weergeeft, kan dat kloppen? Aangezien er bij mij bijna altijd 'over X seconde' staat, in plaats van iets in het verleden. Is het een idee om daar ook het minimum van de telegramtijd/systeemtijd te nemen en in deze casus gewoon 'nu' te laten zien (of in elk geval, dat wat je toont als het echt een recente meting is)?

dennissiemensma, 2017-03-22:
Bedankt voor de tip, dat lijkt me zeker een goede verbetering als aanvulling op de eerste.

dennissiemensma, 2017-09-20:
Ik heb hiervoor een fix gemaakt. Het label heeft straks een vaste breedte en toont tevens geen toekomstige tijd meer.

closed 2017-09-20 by dennissiemensma