dsmrreader/dsmr-reader #273: Docker

xirixiz, 2017-03-07:
Perhaps of use to you or others. I've created a Docker environment for dsmr-reader.

Github:
https://github.com/xirixiz/dsmr-reader-docker

Docker hub:
https://hub.docker.com/r/xirixiz/dsmr-reader-docker/

dennissiemensma, 2017-03-08:
Thank you for sharing your setup! 

* Does it also contain a database or do you intend to only run the application with it? One of the scripts refers to `dsmrdb`.

* I see you manually install the latest version of `psycopg2` in `startup.sh`. It will work fine for 99% of the versions, but you might consider using `/dsmrreader/provisioning/requirements/postgresql.txt`, which contains a pinned version. Because it's also taken into account when the unittests in Travis run.

xirixiz, 2017-03-09:
- Docker Hub is indeed only containing te image with the dsmr-reader application. The Github project contains  a docker-compose.yml file, which also contains a postgres container. https://github.com/xirixiz/dsmr-reader-docker/blob/master/docker-compose.yaml

- I just modified psycopg2 bij replacing it with dsmrreader/provisioning/requirements/postgresql.txt, as you adviced. Cheers!

dennissiemensma, 2017-03-09:
Thanks! In a future release I will have the docs refer to your Docker setup for anyone interested in it :]

xirixiz, 2017-03-10:
Now it's possible to use the development branch besides the master branch.

- DEV uses port 8885.
- DEV uses a seperate Docker volume.
- DEV uses the development branch of this project.

Jeltel, 2017-06-23:
I'm going to try this when I have time. I'll install it on a synology Nas.

dennissiemensma, 2017-09-20:
I've added a reference to your Docker project on top of the Installation page, for next release. Once again thanks.

closed 2017-09-20 by dennissiemensma