dsmrreader/dsmr-reader #275: v1.6 release candidate

dennissiemensma, 2017-03-08:
The upcoming `v1.6` release contains quite some refactoring to support DSMR v5, which allows readings every second and updates gas readings every few minutes. I've constrained the grouping density to still one minute, unless grouping has been disabled at all.

Since I can only test DSMR v4 here, I've have to see if any DSMR v5 user wants to try the current release candidate and search for any bug I've overlooked. I'll test for DSMR v4 regression, since it should **not** be working different at all.

Also, quite some dependencies such as pyserial and Dropbox were updated. So I'll have to check those as well.

dennissiemensma, 2017-03-08:
The current release candidate is the latest version of the `development` branch.

WatskeBart, 2017-03-08:
Is there a quick way to switch between a development branch and a master branch?

dennissiemensma, 2017-03-08:
It's quite easy, but it's a one-way switch due to database migrations. I'll run it myself first to ensure everything works the same and tomorrow, after 24 hours, I'll help you switch. Please remind me.

dennissiemensma, 2017-03-09:
Okay, so I ran it here against my own DSMR v4 smart meter for a day now. All the thing I use (Dropbox, weather, Mindergas, phone notifications) still seem to work, so it should be safe enough.

dennissiemensma, 2017-03-09:
You can switch to the latest release candidate by switching to the `development` branch. Please note that you cannot switch back to the `master`, until the dev branch has been merged back to the master (the actual release).

* So start with making sure you have a recent database back-up.
* `sudo su - dsmr`
* `git fetch`
* `git checkout -b development origin/development` (or `git checkout development` when you already checked out the dev branch in the past, git will in that case state an error about an existing branch)
* `./upgrade.sh` (alias for the existing `./deploy.sh`)
* It should update requirements, static files and have some database migrations.
* Version should be now something like `v1.6rc3`

FutureCow, 2017-03-11:
I've updated to the new 1.6rc3 without a problem. My smartmeter is DSMR 5.0.
Problems so far:
- The gas graph on the dashboard look a bit wrong, at the end i see wrong times (5:26,11:29,23:11)
- The gas graph also doesn't change when you click forward/backward with the arrows (only electricity does)
- The gas graph values looks a bit wrong (almost the same all the time)
- The bottom of the sidebar is missing a picture and also is a bit out of alignement
- At the status page looking for new version is mixed up
- When you open the compare page, you already see 2 turning cirkels, like it is loading something. But it's only waiting for you to fill something in...

This was just a quick check!

FutureCow, 2017-03-11:
Just found out another problem. The reader can't handle a telegram every second. Because i'm using Node-Red i get a bunch of unread telegrams inside node-red. Because of this, DSMR-reader is getting behind (10 minutes in a couple of hours).
I didn't have any problem with DSMR 1.5

dennissiemensma, 2017-03-12:
Hi, thanks **a lot** for trying this out and reporting back so much in detail. Could you attach some screenshots regarding the 'weird looks' of your first comment? It will help me greatly in understanding what you mean.

Regarding your second comment, thanks for debugging that already, as I did seem to forget that the application sleeps for a full 5 seconds by default. I can fix that quickly in a new release candidate.

dennissiemensma, 2017-03-12:
I've decreased the sleep duration to a single second now. You can update it by using the `./upgrade.sh` script again, as `dsmr` user.

Can you check whether that resolves the 'lagging' issue for you? And that the system load isn't affected too much by it?

FutureCow, 2017-03-12:
Thanks for the update. I'm seeing the update every seconds, but i still got the same problem. Now after 10 minutes its already 15 seconds behind realtime. Also the processing looks a bit weird in the log:
`
2017-03-12 17:46:00.068555+01:00: Calling backend services
 - Processed reading: 2017-03-12 17:45:00+01:00.
 - Processed reading: 2017-03-12 17:45:01+01:00.
 - Processed reading: 2017-03-12 17:45:02+01:00.
 - Processed reading: 2017-03-12 17:45:03+01:00.
 - Processed reading: 2017-03-12 17:45:04+01:00.
 - Processed reading: 2017-03-12 17:45:05+01:00.
 - Processed reading: 2017-03-12 17:45:06+01:00.
 - Processed reading: 2017-03-12 17:45:07+01:00.
 - Processed reading: 2017-03-12 17:45:08+01:00.
 - Processed reading: 2017-03-12 17:45:09+01:00.
 - Processed reading: 2017-03-12 17:45:10+01:00.
 - Processed reading: 2017-03-12 17:45:11+01:00.
 - Processed reading: 2017-03-12 17:45:12+01:00.
 - Processed reading: 2017-03-12 17:45:13+01:00.
 - Processed reading: 2017-03-12 17:45:14+01:00.
 - Processed reading: 2017-03-12 17:45:15+01:00.
 - Processed reading: 2017-03-12 17:45:16+01:00.
 - Processed reading: 2017-03-12 17:45:17+01:00.
 - Processed reading: 2017-03-12 17:45:18+01:00.
 - Processed reading: 2017-03-12 17:45:19+01:00.
 - Processed reading: 2017-03-12 17:45:20+01:00.
 - Processed reading: 2017-03-12 17:45:21+01:00.
 - Processed reading: 2017-03-12 17:45:22+01:00.
 - Processed reading: 2017-03-12 17:45:23+01:00.
 - Processed reading: 2017-03-12 17:45:24+01:00.
 - Processed reading: 2017-03-12 17:45:25+01:00.
 - Processed reading: 2017-03-12 17:45:26+01:00.
 - Processed reading: 2017-03-12 17:45:27+01:00.
 - Processed reading: 2017-03-12 17:45:28+01:00.
 - Processed reading: 2017-03-12 17:45:29+01:00.
 - Processed reading: 2017-03-12 17:45:30+01:00.
 - Processed reading: 2017-03-12 17:45:31+01:00.
 - Processed reading: 2017-03-12 17:45:32+01:00.
 - Processed reading: 2017-03-12 17:45:33+01:00.
 - Processed reading: 2017-03-12 17:45:34+01:00.
 - Processed reading: 2017-03-12 17:45:35+01:00.
 - Processed reading: 2017-03-12 17:45:36+01:00.
 - Processed reading: 2017-03-12 17:45:37+01:00.
 - Processed reading: 2017-03-12 17:45:38+01:00.
 - Processed reading: 2017-03-12 17:45:39+01:00.
 - Processed reading: 2017-03-12 17:45:40+01:00.
 - Processed reading: 2017-03-12 17:45:41+01:00.
 - Processed reading: 2017-03-12 17:45:42+01:00.
 - Processed reading: 2017-03-12 17:45:43+01:00.
 - Processed reading: 2017-03-12 17:45:44+01:00.
 - Processed reading: 2017-03-12 17:45:45+01:00.
 - Processed reading: 2017-03-12 17:45:46+01:00.
 - Processed reading: 2017-03-12 17:45:47+01:00.
 - Processed reading: 2017-03-12 17:45:48+01:00.
 - Processed reading: 2017-03-12 17:45:49+01:00.
 - Processed reading: 2017-03-12 17:45:50+01:00.
 - Processed reading: 2017-03-12 17:45:51+01:00.
 - Processed reading: 2017-03-12 17:45:52+01:00.
 - Processed reading: 2017-03-12 17:45:53+01:00.
 - Processed reading: 2017-03-12 17:45:54+01:00.
 - Processed reading: 2017-03-12 17:45:55+01:00.
 - Processed reading: 2017-03-12 17:45:56+01:00.
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:01.720497+01:00: Calling backend services
 - Processed reading: 2017-03-12 17:45:57+01:00.
 - Processed reading: 2017-03-12 17:45:58+01:00.
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:02.772519+01:00: Calling backend services
 - Processed reading: 2017-03-12 17:45:59+01:00.
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:03.813836+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:04.836974+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:05.859827+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:06.884718+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:07.909250+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:08.936891+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:09.963383+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:10.992765+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:12.020507+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2017-03-12 17:46:13.045670+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

etc...
`

dennissiemensma, 2017-03-13:
Well, what it does (by default), is grouping all readings by minute. So it kinda makes sense, because it will 'await' the minute marker and then group all readings for that minute. Therefor it will never be realtime, with the exception of when it will lag several minutes (or more) behind.

And when I look at your instance of the application (thanks for that) the input data seems a bit off. Kan you tell me more about that node-red you are using? Is that some kind of buffer that pushes data against the API of the dsmr-reader?

dennissiemensma, 2017-03-13:
I also see some weird data in your gas graph, which looks like invalid data similar to #269. You can run this query to see if there is any data off:

```
sudo -u postgres psql dsmrreader -c "select id, read_at, delivered, currently_delivered from dsmr_consumption_gasconsumption order by read_at desc limit 24;"
```

(asumes postgresql)

dennissiemensma, 2017-03-14:
I've updated the dev-branch with some additional logging in the dsmr_backend process. It should now state how many readings (max 128) it attempts to compact/process. Maybe that will tell us more about the exact flow it's having on your end. 

I'm really sad that I have a DSMR v4 smart meter here at home and can't reproduce it myself. I'd also like to know how you are exactly inserting data through the node-red you were talking about.

FutureCow, 2017-03-14:
I've updated to RC5. The log:

2017-03-14 23:44:06.932496+01:00: Calling backend services
 - Trying to process 1 reading(s), capped at 128 max per run
 - Processed reading: 2290544 @ 2017-03-14 22:14:36+01:00 (2.237 kW)
2017-03-14 22:44:06.955316+00:00: Command completed. Sleeping for 1 second(s)...

2017-03-14 23:44:07.955726+01:00: Calling backend services
 - Trying to process 1 reading(s), capped at 128 max per run
 - Processed reading: 2290545 @ 2017-03-14 22:14:37+01:00 (2.238 kW)
2017-03-14 22:44:07.981085+00:00: Command completed. Sleeping for 1 second(s)...

2017-03-14 23:44:08.982459+01:00: Calling backend services
 - Trying to process 1 reading(s), capped at 128 max per run
 - Processed reading: 2290546 @ 2017-03-14 22:14:38+01:00 (2.237 kW)
2017-03-14 22:44:09.008162+00:00: Command completed. Sleeping for 1 second(s)...

2017-03-14 23:44:10.012497+01:00: Calling backend services
 - Trying to process 1 reading(s), capped at 128 max per run
 - Processed reading: 2290547 @ 2017-03-14 22:14:39+01:00 (2.239 kW)
2017-03-14 22:44:10.085189+00:00: Command completed. Sleeping for 1 second(s)...

2017-03-14 23:44:11.088490+01:00: Calling backend services
 - Trying to process 1 reading(s), capped at 128 max per run
 - Processed reading: 2290548 @ 2017-03-14 22:14:40+01:00 (2.242 kW)
2017-03-14 22:44:11.110910+00:00: Command completed. Sleeping for 1 second(s)...

So it is still trying to read only 1? I can try to flush the buffer.

Node-Red is doing the following:
ttyUSB0 (original P1 cable) is red bij Node-Red.
Node-Red is pushing every telegram to a socat tunnel that is ending to ttyUSB20
DSMR-Reader is using ttyUSB20 for reading the telegrams.

But Node-Red is queuing the telegrams, a normal serial-port doesn't (so you will miss some)
So every minute you read 60 telegrams, but the que is 61, it will add every minute 1 telegram. Now after 2 day's the que is already 1 hour of telegrams.

dennissiemensma, 2017-03-15:
I suspect the datalogger as well, because that **did** change. 

The new version of it attempts to keep the serial connection open, due to the high amount of data DSMR v5 could generate. It should keep the IO overhead somewhat lower. But since I can only test DSMR v4 at home, I didn't expect it to have any side effect.

So I think I will revert that part of the release back to the dsmr-reader v1.5.x. Which maybe will result in more IO (as it will keep opening and closing the serial connection for each read), but maybe be more stable in the end. Especially when you are sure that you had no datalogger problems on the former version.

Could you check whether the `datalogger` process log indicates any errors at all?

dennissiemensma, 2017-03-15:
I've updated the branch again 4f6343c4c43bf81699afb47344ba55a3696e043d:

* Reverted 1 second backend sleep to 5 seconds.
* Reverted datalogger to dsmr-reader v1.5.x state.
* Reverted some debug logging.

dennissiemensma, 2017-03-15:
I also know now why the 'grouping' of the readings for each minute seems a bit off (selecting a bunch, and handling some after). It's caused by the timestamp the smart meter embeds in the telegram. 

This project's grouping algorithm will delay any telegrams' grouping (default = on) until the 'current minute' has passed, to prevent skipping any telegrams. 
However, I did not take any differences between the system time and the smart meter's internal clock into account. So it's basically thinking it's OK to group all the telegrams so far, but in some cases there might be more coming.

Fortunately this won't matter too much since there are still plenty telegrams grouped each minute. And the consumption in the application is measured by the 'meter positions' between intervals. But still a bug...

FutureCow, 2017-03-15:
I've updated to rc6.

when i look inside the data_logger.log I don't see any errors. The telegrams that are passing by, are exactly the time that DSMR-reader is behind. Now after 1 day, DSMR-reader is 30 minutes behind. Timestamp form the telegrams are at this moment: 0-0:1.0.0(170315182735W), so also 30 minutes behind. But it's not reading al the telegrams(1800!) that are currently in que. I can flush the que again and see what will happen.

dennissiemensma, 2017-03-15:
I think I've overlooked another sleep, as each process defines it's sleep. The **datalogger** has a 1 second sleep as well, but that cannot work for DSMR v5. I will reduce that one, so it will allow to catch up.

dennissiemensma, 2017-03-15:
It's updated again, and should now only sleep for a quarter of a second (to prevent IO hammering).

FutureCow, 2017-03-15:
Sorry, it's still lacking behind 30 minutes. It's not reading the que.

dennissiemensma, 2017-03-15:
You might need to give it a few minutes, but you should see some progress in the end. Even after a few minutes. Does the datalogger log still only read one reading per second?

Are you sure it worked without delay on the previous release? Because I'm [trying to see for whats causing this difference](https://github.com/dennissiemensma/dsmr-reader/compare/v1.5.5...development), even with the datalogger reset to it's previous state.

You could force the processes in Supervisor to reload their code, even though the deployment script should already have done that. `sudo supervisorctl restart all` should do the trick. To be sure.

FutureCow, 2017-03-15:
after the restart all, i see the telegrams flying in :)

dennissiemensma, 2017-03-15:
Good to hear. Did you btw run the deployment script at all? Because it should do all kinds of stuff.

Can you try syncing the static files (images etc) manually? It's for one of the problems you reported about the sidebar missing an image. This should copy all static files to the folder where the webserver (nginx) can serve it:

```
su - dsmr
./manage.py collectstatic --noinput
```

If this does not fix it, can you copy the error your browser states in the console? I'm interested in the path it's using, here it's: `/static/dsmr_frontend/img/GitHub-Mark-32px.png`

FutureCow, 2017-03-15:
Isn't the upgrade.sh script also running deploy?

The image is working now, but the alignment is still a bit off:
[image]

dennissiemensma, 2017-03-15:
Yeah the script is just an alias. THanks for the image, I will try to fix the alignment later. 

So I can asume the datalogger is 'back to normal' now?

FutureCow, 2017-03-15:
For me it's working great. Also the gas graph is getting back to normal. I deleted the 3 wrong value's.

dennissiemensma, 2017-03-15:
Can you make a screenshot? It should [look like the bottom screenshots on this page](http://dsmr-reader.readthedocs.io/en/latest/screenshots.html).

Can you check whether your browser prints any errors about missing files it cannot load in the admin pages?

FutureCow, 2017-03-15:
Screenshot:
[image]

dennissiemensma, 2017-03-16:
I think it's a problem with your static files served by the webserver. Did you run the `collectstatic` command mentioned above? 

If so, you might have to check your webserver configuration. Did you alter the default one that comes with this project?

dennissiemensma, 2017-03-16:
I think many of the other problems you're seeing are related to that as well (missing CSS, javascript, images):

- *"The bottom of the sidebar is missing a picture and also is a bit out of alignement"*
- *"At the status page looking for new version is mixed up"*
- *"When you open the compare page, you already see 2 turning cirkels, like it is loading something. But it's only waiting for you to fill something in..."*

You might want to check for sure, by pressing F12 or enabling the dev console in your browser. It should complain about missing resources.

These were related to invalid gas readings:
- *"The gas graph on the dashboard looks a bit wrong, at the end i see wrong times (5:26,11:29,23:11)"*
- *"The gas graph values looks a bit wrong (almost the same all the time)"*

This is something I didn't account for, I will try to fix it:
- *"The gas graph also doesn't change when you click forward/backward with the arrows (only electricity does)"*

dennissiemensma, 2017-03-16:
@WatskeBart did you switch to the development version and found any weird stuff? I asume you are running a smart meter with DSMR v4, like I am?

WatskeBart, 2017-03-16:
@dennissiemensma I didn't switch to the dev branch, but I withheld from updating because it's a one-way upgrade. I hoped I could run the dev branch parallel.

dennissiemensma, 2017-03-16:
Okay still thanks for trying.

dennissiemensma, 2017-03-16:
I've added the missing scroll buttons, for the gas graph, in the latest rc (8) now.

dennissiemensma, 2017-03-17:
@WatskeBart it seems that it **is** possible to switch between versions. *Data* migrations are not backwards compatible by default, but *schema* changes are. And there are only schema changes between these versions. Django supports going back and it's quite easily. I've just confirmed it here for my own setup.

```
$ ./manage.py migrate dsmr_datalogger 0005_verify_telegram_crc_setting 

Operations to perform:
  Target specific migration: 0005_verify_telegram_crc_setting, from dsmr_datalogger
Running migrations:
  Rendering model states... DONE
  Unapplying dsmr_datalogger.0006_dsmr_firmware_v5... OK
```

So in case you still want to try it out, you can. But please know that the changes are focussed on helping anyone with a DSMR v5 smart meter. For DSMR v4 smart meters, there are limited extra features in this version. Just so you know, and to manage any expectations :]

**Going back**
* `su - dsmr`
* `./manage.py migrate dsmr_datalogger 0005_verify_telegram_crc_setting`
* It should state: `Unapplying dsmr_datalogger.0006_dsmr_firmware_v5... OK`
* `git checkout master` (back to the latest, v1.5.5)
* `./deploy.sh` (reinstalls stuff and reloads the app with the old version code again)

TopdRob, 2017-03-18:
@dennissiemensma I've installed RC8 and so far no issues with DSMR 5.

dennissiemensma, 2017-03-18:
@TopdRob thanks a lot for taking the time for checking it out yourself. 

Can you confirm that you are now able to see gas consumption more often? The limitation of storing gas readings hourly should be removed for DSMR v5 users.

FutureCow, 2017-03-18:
For me all the problems i had were gone after deleting /var/www/static folder, and deploy again.
My gas reading are now 5 minutes, so this is really nice! 
Thank you for implementing this!

dennissiemensma, 2017-03-18:
@TopdRob & @FutureCow thank you both a lot for providing the feedback. In that case, I will make this release candidate final. I didn't see weird things myself either, so we should be good to go.

closed 2017-03-18 by dennissiemensma

dennissiemensma, 2017-03-18:
The version has been released, so you guys can swap back to the master branch:

* `su - dsmr`
*  `git fetch`
*  `git checkout master`
*  `git pull`
*  `./deploy.sh` 

The application interface should now display `v1.6` in stead of `v1.6rcX`