dsmrreader/dsmr-reader #276: Display error compare page on mobile

thommy101, 2017-03-10:
Hi,

The compare page on mobile has some weird display errors with the time range selection. They are hard to select. They are diagonally placed and the day and month have there hit box to low. This is also reproducible on desktop when resizing the window to anything smaller than 1000 px.

A possible fix I've found is removing the float of .sm-st-icon (around line 3744 of style.css) and letting the col-md do it's work.

dennissiemensma, 2017-03-11:
Hi thank you for reporting this issue and the detailedinformation. I will look into this in a future release dedicated to UI fixes.

dennissiemensma, 2017-09-29:
This was caused by the responsive **row** elements used. It seems that the **columns** containing the three buttons are converted to **rows** when the screen is too small. And because the div's of the buttons did not have any height, they were aligned weirdly.

I've fixed this by giving them all a fixed height. The 'mobile view' will now show them on top of each other, allowing the hitbox to be normal again as well. See below for a screenshot, thanks again for reporting.

[image]

closed 2017-09-29 by dennissiemensma