dsmrreader/dsmr-reader #277: Statuspagina en de waarschuwingen met betrekking tot dataverwerking

pyrocumulus, 2017-03-13:
Op de statuspagina zie je per onderdeel de status en de staat van verwerking. Bij mij is heel erg vaak het bovenste element met 'P1 DSMR TELEGRAMMEN' rood ondanks het laatste telegram volgens de tekst slechts een seconde (of twee) geleden verwerkt is. Tot nu toe was het slechts één keer echt terecht (toen het dataverwerking proces van dsmr er bij mij uit lag). 
Misschien zit de oorzaak van de onterechte waarschuwing in het feit dat dsmr bij mij elke seconde een telegram binnenkrijgt vanwege DSMR v5 maar deze niet direct verwerkt. De onverwerkte telegrammen buffer loopt namelijk doorgaans op tot een stuk of 20 om vervolgens weer terug naar ~5 te vallen. Wellicht heb je dat al gefixed in je aanpassingen voor 1.6.

In navolging daarop; het zou denk ik fijn zijn als je een eventuele waarschuwing vanuit de statuspagina ook op het Dashboard zou zien. Toen de dataverwerking bij mij gestopt was, kwam ik daar pas achter toen ik de 'nieuwste' tijden goed bekeek in de grafieken. Toen ik daarop naar de statuspagina ging bleek ik al ruim 1000 telegrammen achter te lopen.

pyrocumulus, 2017-03-13:
Nog een plaatje van de melding op de statuspagina voor de duidelijkheid:

[image]

dennissiemensma, 2017-03-13:
@pyrocumulus bedankt voor het melden van dit issue. Ik heb dit één keer eerder gezien, namelijk in #207. De oorzaak zat daar in de interne klok van de slimme meter die vóórliep. Visueel heb ik dat opgelost, maar pas in versie v1.6. Die zou gisteren klaar zijn voor release, echter voor DSMR v5 meters lijken er wat issues te zijn via #275.

Wat betreft je andere suggestie voor een waarschuwing op het dashboard, ik heb hem toegevoegd aan #235. Bedankt!

pyrocumulus, 2017-03-13:
Oké ik ben benieuwd! Klinkt wel als hetzelfde probleem inderdaad. Ik heb zojuist nog even de tijd van de pi gecontroleerd (kan niet zo eenvoudig een telegram bekijken volgens mij) en dat lijkt goed, dus ik vermoed dat de tijd uit het telegram voorloopt. Ik wacht v1.6 wel af, het is verder geen probleem :)

dennissiemensma, 2017-03-14:
Oke bedankt voor de terugkoppeling. Je hoeft je in ieder geval dus geen zorgen te maken. 

De applicatie gebruikt een ingebouwd filter die een 'aantal seconden' omzet naar tekst. En dat werkt  ook negatief en geeft een vreemde indruk, want het was eigenlijk nooit bedoeld dat die tijd van de meter vóór zou lopen. Puur visueel dingetje dus. :]

closed 2017-03-14 by dennissiemensma