dsmrreader/dsmr-reader #278: Raspberry zero w?

hsleewis, 2017-03-14:
Is de raspberry pi zero (w) sterk genoeg voor de applicatie?

Varsame, 2017-03-14:
Moet lukken, de Zero heeft volgens mij dezelfde SOC als de originele Raspberry Pi model B, met een hogere kloksnelheid (1GHz vs. 700MHz). DSMR draait prima op mijn model B, soms even wachten maar verder geen issue.

dennissiemensma, 2017-03-14:
Ik heb het zelf nog niet getest. Hier heb ik alleen de normale Zero in huis, maar daar kwam ik niet uit met poorten en connectors. 
Voor een datalogger zou een Zero (W) sowieso snel genoeg moeten zijn. Voor het draaien van zowel een database, webserver, applicatie en twee (zware) achtergrondprocessen... lijkt het mij wat te zwak.

Zoals Varsame al aangeeft kán het wel werken, met wat vertraging. Ik zou het zelf niet kunnen aanraden in ieder geval, tenzij je alleen een datalogger er op zet. Je kunt ermee experimenteren. 

Dit project is officieel voor een Pi 2+, maar in principe is het overal te draaien, mits de requirements/installatie maar uitgevoerd kan worden.

closed 2017-03-14 by dennissiemensma