dsmrreader/dsmr-reader #279: Weather report with temperature '-' eventually results in stopped dsmr_backend

fpompert, 2017-03-17:
Version: 1.5.5
Debian 8.7.2

Whenever Buienradar supplies a '-' temperature value instead of the usual float value, and in case this takes a while to recover, the dsmr_backend will fail after about 1,5 hours and processing of data is halted. This is due to an unhandled exception.

Logging from dsmr_backend.log:

2017-03-17 09:00:00.232048+01:00: Calling backend services
 - Processed reading: 2017-03-17 08:59:56.468089+01:00.
 - Performing temperature reading at Buienradar.
 >>> Uncaught exception :: ["'-' value must be a decimal number."]
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_weather>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 29, in run
    raise current_response

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 227, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_weather/apps.py", line 20, in _on_backend_called_signal
    dsmr_weather.services.read_weather()

  File "/home/dsmr/dsmr-reader/dsmr_weather/services.py", line 52, in read_weather
    TemperatureReading.objects.create(read_at=read_at, degrees_celcius=temperature)

supervisord.log:

2017-03-17 10:24:09,988 INFO spawned: 'dsmr_backend' with pid 2411
2017-03-17 10:24:10,577 INFO exited: dsmr_backend (exit status 1; not expected)
2017-03-17 10:25:51,001 INFO spawned: 'dsmr_backend' with pid 2417
2017-03-17 10:25:51,589 INFO exited: dsmr_backend (exit status 1; not expected)
2017-03-17 10:25:52,590 INFO gave up: dsmr_backend entered FATAL state, too many start retries too quickly

I've implemented a quick workaround in dsmr_weather/services.py by checking if the received value from Buienradar is a valid float, otherwise it will just report a 0.0 temperature for now :)

snippet
---
    temperature_element = root.find(xpath)
    temperature = temperature_element.text
>>>
    print (' - Got back temperature: ', str(temperature))

    if not is_number(str(temperature)):
        temperature = 0.0
        print (' - Reverting temperature to 0.0 due to error!')

    print (' - Using temperature: ', str(temperature))
<<<
    # Gas readings trigger these readings, so the 'read at' timestamp should be somewhat in sync.
    # Therefor we align temperature readings with them, having them grouped by hour that is..
    read_at = timezone.now().replace(minute=0, second=0, microsecond=0)
    TemperatureReading.objects.create(read_at=read_at, degrees_celcius=temperature)

    # Push next sync back for an hour.
    weather_settings.next_sync = read_at + timezone.timedelta(hours=1)
    weather_settings.save()
>>>
def is_number(s):
    try:
        complex(s) # for int, long, float and complex
    except ValueError:
        return False

    return True

<<<

dennissiemensma, 2017-03-18:
Hi, thanks a lot for taking the trouble of reporting this. 

A similar issue was found in #251 and fixed for the v1.6 release. I hope to release that version this weekend, as I'm about to checkout some feedback regarding the latest release candidate.

dennissiemensma, 2017-03-20:
Can you try upgrading to the new `v1.6`, it should fix at least on of the causes for this crash.

fpompert, 2017-03-21:
Installed v1.6. I'll report back with results. Hopefully it's solved 👍

fpompert, 2017-03-28:
Unfortunately today Buienradar still suffers from unhandled exceptions in the backend log due to it receiving a - temperature. What should be the expected behaviour in these cases?

dennissiemensma, 2017-03-28:
Thank you for reopening. Can you tell me whether the backend process actually crashes or does it just report the error?

The behaviour changed in v1.6 in a way that it should not crash the process (but log the error anyway).

fpompert, 2017-03-28:
It logs the error and continues. The supervisor log looks cleaner than it did before with the previous version. So it matches your expectation. It does now retry Buienradar requests every 5 seconds? Hope this does not cause problems further down the road :) If this also matches your expectations I can reclose the issue.

dennissiemensma, 2017-03-28:
Thanks for the clarification. I think I will still need to fix it to prevent the many retries it does now, but it's good to hear that it no longer crashes the backend.

dennissiemensma, 2017-09-19:
I've added some additional validation for next release. Also, on any failure the next retry is delayed for 5 minutes now.

closed 2017-09-19 by dennissiemensma