dsmrreader/dsmr-reader #280: Uncaught exception

knetsooj, 2017-03-18:
sinds vannacht klapt de dsmr_backend er continu uit met de volgende melding. 

> Starting INFINITE command loop...
> 2017-03-18 12:54:11.118102+01:00: Calling backend services
 > Uncaught exception :: invalid page in block 239 of relation base/16385/16671
 > <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_consumption>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 29, in run
    raise current_response
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 227, in send_robust
>    response = receiver(signal=self, sender=sender, **named)
>
>  File "/home/dsmr/dsmr-reader/dsmr_consumption/apps.py", line 20, in _on_backend_called_signal
>    dsmr_consumption.services.compact_all()
>
>  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 21, in compact_all
>    compact(dsmr_reading=current_reading)
>
>  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 83, in compact
>    phase_currently_delivered_l3=grouped_reading['avg_phase_delivered_l3'],
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 85, in manager_method
>    return getattr(self.get_queryset(), name)(*args, **kwargs)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 399, in create
>    obj.save(force_insert=True, using=self.db)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 796, in save
>    force_update=force_update, update_fields=update_fields)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 824, in save_base
>    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 908, in _save_table
>    result = self._do_insert(cls._base_manager, using, fields, update_pk, raw)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 947, in _do_insert
>    using=using, raw=raw)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 85, in manager_method
>    return getattr(self.get_queryset(), name)(*args, **kwargs)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 1045, in _insert
>    return query.get_compiler(using=using).execute_sql(return_id)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1054, in execute_sql
>    cursor.execute(sql, params)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
>    return self.cursor.execute(sql, params)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 94, in __exit__
>    six.reraise(dj_exc_type, dj_exc_value, traceback)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/utils/six.py", line 685, in reraise
>    raise value.with_traceback(tb)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
>    return self.cursor.execute(sql, params)
>  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 29, in run
>    raise current_response
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 227, in send_robust
>    response = receiver(signal=self, sender=sender, **named)
>
>  File "/home/dsmr/dsmr-reader/dsmr_consumption/apps.py", line 20, in _on_backend_called_signal
>    dsmr_consumption.services.compact_all()
>
>  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 21, in compact_all
>    compact(dsmr_reading=current_reading)
>
>  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 83, in compact
>    phase_currently_delivered_l3=grouped_reading['avg_phase_delivered_l3'],
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 85, in manager_method
>    return getattr(self.get_queryset(), name)(*args, **kwargs)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 399, in create
>    obj.save(force_insert=True, using=self.db)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 796, in save
>    force_update=force_update, update_fields=update_fields)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 824, in save_base
>    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 908, in _save_table
>    result = self._do_insert(cls._base_manager, using, fields, update_pk, raw)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/base.py", line 947, in _do_insert
>    using=using, raw=raw)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 85, in manager_method
>    return getattr(self.get_queryset(), name)(*args, **kwargs)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 1045, in _insert
>    return query.get_compiler(using=using).execute_sql(return_id)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1054, in execute_sql
>    cursor.execute(sql, params)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
>    return self.cursor.execute(sql, params)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 94, in __exit__
>    six.reraise(dj_exc_type, dj_exc_value, traceback)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/utils/six.py", line 685, in reraise
>    raise value.with_traceback(tb)
>
>  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
>    return self.cursor.execute(sql, params)

dit was ongeveer na 2 uur na de service restart in navolging op mijn melding in ticket #254

dennissiemensma, 2017-03-18:
Bedankt voor je melding. De fout ` invalid page in block 239 of relation base/16385/16671` zegt mij zelf weinig, behalve dat het een directe fout uit de database lijkt. Ik zal even voor je zoeken hoe en wat.

dennissiemensma, 2017-03-18:
Het lijkt erop dat er datacorruptie is opgetreden als ik de foutmelding mag geloven. 

Heb je nog een recente backup? Standaard maakt de applicatie databasebackups in de `backups` map en die kun je optioneel syncen met je Dropbox account. Wellicht goed om die even achter de hand te houden.

Ik neem aan dat je dit op een Raspberry draait. Dan is het zaak om te kijken of je SD-kaart nog goed is. Voor Linux-gebruikers is dat vrij eenvoudig te doen door de kaart in je PC of laptop uit te lezen en `fsck` te doen. Voor Windows-gebruikers weet ik helaas niet hoe en wat. Je kunt nog kijken of je de Raspberry (het OS erop) zelf kan forceren om bij het opnieuw opstarten een `fsck` te doen: https://www.cyberciti.biz/faq/linux-force-fsck-on-the-next-reboot-or-boot-sequence/

Hopelijk geeft dat in ieder geval duidelijkheid over of de kaart nog goed is. Helaas blijft dat een onderdeel dat zeer gevoelig voor slijtage is. Mocht dit het niet zijn dan mogen we even verder kijken naar een oplossing hiervoor.

knetsooj, 2017-03-18:
Allereerst bedankt voor je snelle reactie weer!

Zojuist heb ik de backup van afgelopen nacht teruggezet deze bevatte alleen de verwerkte data tot en met begin deze week. Nu is de RPI bezig om de rest van de 35000+ telegrammen te verwerken. duimen dat het goed blijft gaan.

Ik draai dit inderdaad op een RPI, echter is de SD kaart pas een week of 6 oud. (de vorige was na een jaar versleten)

Ik heb nog een odroid C2 met eMMC geheugen en een nieuwe USB stick liggen, waarop ik eventueel ook de data op zou kunnen zetten. Wat zou de beste keuze zijn mbt de schrijffrequenties en levensduur?

dennissiemensma, 2017-03-18:
Bedankt voor de aanvulling. Ik verwacht niet dat een nieuwe SD-kaart meteen kapot gaat. Ik heb dat zelf 1 keer gehad en na vervanging van die kaart, draait die inmiddels anderhalf jaar. Wat dat betreft zit ik niet heel diep in welke types het beste zijn hiervoor. Uiteindelijk crasht elke kaart 1x en helpen alleen de backups nog.

Houd wel in de achterhoofd dat **als** de kaart toch corrupt is, je op den duur weer zoiets krijgt. Zorg ervoor dat je in ieder geval 1 goede backup achter de hand houd.

dennissiemensma, 2017-09-18:
Nog even opnieuw gecheckt of ik hier kan voor kan fixen, maar is helaas hardware gerelateerd. 

Uiteindelijk kan het ook een oplossing zijn om bijvoorbeeld een PiDrive of server met niet-flash storage te gebruiken. Ik heb inmiddels mijn SD-kaart ook al een keer moeten vervangen afgelopen zomer.

closed 2017-09-18 by dennissiemensma