dsmrreader/dsmr-reader #282: Warning error tijdens upgrade naar 1.6 van 1.5.5

dbmnl, 2017-03-19:
Zojuist geupgrade naar versie 1.6 van 1.5.5 en zag tijdens de installatie de volgende 2 warning foutmeldingen voorbij vliegen. Het zal waarschijnlijk geen kwaad kunnen want even snel er doorheen vliegen zag ik geen fouten verder, maar wellicht toch goed om even te melden zodat je er rekening mee kan houden. 

**Melding 1:**
```
Downloading/unpacking django-colorfield==0.1.12 (from -r dsmrreader/provisioning/requirements/base.txt (line 3))
  Downloading django-colorfield-0.1.12.tar.gz
  Running setup.py (path:/tmp/pip-build-h9d3rc30/django-colorfield/setup.py) egg_info for package django-colorfield

    warning: no previously-included files matching '*.pyc' found under directory '*'
    warning: no previously-included files matching '__pycache__' found under directory '*'
    warning: no previously-included files matching '.DS_Store' found under directory '*'
Downloading/unpacking django-debug-toolbar==1.7 (from -r dsmrreader/provisioning/requirements/base.txt (line 4))
```

**Melding 2:**
```
Found existing installation: Django 1.10.4
    Uninstalling Django:
      Successfully uninstalled Django
  Found existing installation: django-colorfield 0.1.11
    Uninstalling django-colorfield:
      Successfully uninstalled django-colorfield
  Running setup.py install for django-colorfield

    warning: no previously-included files matching '*.pyc' found under directory '*'
    warning: no previously-included files matching '__pycache__' found under directory '*'
    warning: no previously-included files matching '.DS_Store' found under directory '*'
  Found existing installation: django-debug-toolbar 1.6
    Uninstalling django-debug-toolbar:
```

dennissiemensma, 2017-03-19:
Bedankt voor het melden. Dit zijn waarschuwingen die door `pip` gemaakt zijn, de tool die wat afhankelijkheden installeert. Doorgaans kunnen deze geen kwaad.

Ik neem aan dat de laatste versie verder prima draait na de upgrade?

dbmnl, 2017-03-19:
Jup geen enkel probleem verder draait prima.

dennissiemensma, 2017-03-19:
Top dan klopt het allemaal (Y)

closed 2017-03-19 by dennissiemensma