dsmrreader/dsmr-reader #283: Upgrade naar 1.6

goegol, 2017-03-19:
In je documentatie schrijf je (https://dsmr-reader.readthedocs.io/nl/latest/faq.html#how-can-i-update-my-application) dat je als dsmr user moet zijn ingelogd, en dan in de hoofdmap de file ./upgrade.sh moet draaien.

Deze file ontbreekt bij mij.
Volgende files staan in de hoofdmap:
backups      coverage_report  dsmr_api      dsmr_consumption  dsmr_mindergas     dsmr_stats    logs            pre-deploy.sh  README.md
CHANGELOG    deploy.sh        dsmr_backend  dsmr_datalogger   dsmr_notification  dsmr_weather  manage.py       pylama.ini     reload.sh
codecov.yml  docs             dsmr_backup   dsmr_frontend     dsmrreader         LICENCE       post-deploy.sh  quick-test.sh  test-all.sh

dennissiemensma, 2017-03-19:
Bedankt voor het melden, zeer scherp! Ik heb de documentatie wat herschreven en dit klopt inderdaad niet. Ik zal het aanpassen.

Om te upgraden kun je het `./deploy.sh` script draaien.

dennissiemensma, 2017-03-19:
Het is aangepast, bedankt.

closed 2017-03-19 by dennissiemensma