dsmrreader/dsmr-reader #284: Automatische backups geven alleen lege bestanden

michielvisser, 2017-03-19:
Ik heb in DSMR Reader de dagelijkse backups aanstaan en op het tijdstip (02:00) worden de bestanden ook daadwerkelijk aangemaakt in de juiste map. Echter, hoewel ik inmiddels bijna 10000 telegrammen in de database heb zitten (en een handmatige PSQL backup een .SQL-bestand van iets meer dan 1.5 megabyte oplevert) zijn de backupbestanden die gemaakt worden slechts 60 bytes groot. Als ik deze gunzip dan zijn de resulterende bestanden leeg.

Ik heb gezocht naar een configuratie o.i.d. die wellicht niet goed zou staan, maar kan niets afwijkends vinden. De rest van de applicatie werkt wel goed. DSMR Reader was versie 1.5.5, nu 1.6 met een PostgreSQL database versie 9.2.18 (CentOS 7). Is ergens na te zien in bijvoorbeeld logging van de applicatie wat er fout gaat bij het maken van de backup?

Voorlopig heb ik een workaround door een cron-job te maken die de PostgreSQL database exporteert zodat ik de database in de backup kan meenemen.

dennissiemensma, 2017-03-19:
Bedankt voor je melding. Begrijp ik het goed dat dit al langere tijd speelt, dus ook vóór dsmr-reader v1.6?

Je zou kunnen kijken of **handmatig** een backup maken wél lukt (als de dsmr-gebruiker). De [applicatie doet het op deze manier](https://github.com/dennissiemensma/dsmr-reader/blob/v1.6.0/dsmr_backup/services/backup.py#L55). De instellingen die het gebruikt staan binnen het project in `dsmrreader/settings.py` ([hier staan de standaardinstellingen](https://github.com/dennissiemensma/dsmr-reader/blob/v1.6.0/dsmrreader/provisioning/django/postgresql.py).

```
sudo su - dsmr
pg_dump --host=localhost --user=dsmrreader --dbname=dsmrreader | head -n 100
```

Dit vraag wel handmatig het wachtwoord in te voeren, wat iets anders is dan het script. Maar ik zou eerst kijken of dit goed werkt.

michielvisser, 2017-03-22:
Ik dacht in eerste instantie dat dit ook speelde met 1.5.5, maar realiseerde me later dat met een vorige installatie van de DSMR Reader ik een MySQL-database had gebruikt. Deze werkte voor zover ik weet wel goed. Daarna heb ik in het kader van het testen en het gebruik van de API een nieuwe installatie gedaan waarbij ik gelijk naar versie 1.6 (b7) en een PSQL database ben gegaan.

Het commando heb ik geprobeerd, echter geeft ook handmatig een foutmelding. Het issue zit in de parameter --dbname, deze is niet bekend in pg_dump. Als ik het commando ombouw en de DB naam opgeef zonder --dbname, dan werkt het wel goed en krijg ik ook een output.

`pg_dump --host=localhost --user=dsmrreader dsmrreader | head -n 100`

Als ik in de documentatie van PostgreSQL kijk is dit ook de juiste syntax voor versie 9.1 en hoger, zie https://www.postgresql.org/docs/9.2/static/app-pgdump.html.

dennissiemensma, 2017-03-22:
Bedankt voor je toelichting. Ik zal het voor je onderzoeken, want hier werkt het prima (postgresql 9.4 & 9.5). Ik zie de `dbname` zowel als expliciete parameter en argument staan wanneer ik `pg_dump --help` doe.

Mocht de argument-versie overal werken (degene die je voorstelt) dan zal ik het aanpassen zodat het voor iedereen geldt.

michielvisser, 2017-03-23:
Na wat verder onderzoek in de documentatie vanaf mijn kant: de --dbname is in versie 9.3 blijkbaar geintroduceerd, want dat is de eerste verwijzing ernaar die ik kan vinden. De standaard versie van PostgreSQL is voor CentOS 7 nog versie 9.2, hoewel het wel mogelijk is om met een extra repository een nieuwere versie te installeren. Mocht het dus onhandig zijn om de parameter te veranderen dan moet ik in mijn installatie-aantekeningen dus opnemen dat PSQL door een nieuwere versie vervangen dient te worden.

Afgaande op de documentatie zou de argument-versie overigens moeten werken voor alle versies van PSQL, dit zou het dus wel universeler maken.

dennissiemensma, 2017-03-23:
Bedankt voor je onderzoek. Mijn voorkeur gaat ook uit naar een oplossing die zo breed mogelijk werkt, dus ik zal testen of de andere variant hier goed werkt. Zo ja dan voer ik het door in een volgende release. 
Een nieuwere postgresql versie installeren zou niet hoeven, deze backup's moeten uiteindelijk werken in alle 9.x versies van de DB.

dennissiemensma, 2017-09-12:
Lijkt goed te werken. Ik heb het doorgevoerd voor de volgende release. Bedankt voor de tip :]

closed 2017-09-12 by dennissiemensma