dsmrreader/dsmr-reader #285: Verzoek: mogelijkheid in het dashboard om een eigen tijdspanne weer te geven

michielvisser, 2017-03-19:
Op dit moment laat de elektriciteitsgrafiek (ik heb geen teruglevering en gas) standaard het laatste half uur zien in het dashboard. Het is mogelijk terug te kijken d.m.v. de knoppen onder de grafiek. Echter, omdat data pas een dag later geaggregeerd wordt in het archief, zou ik de volgende twee opties voor het bekijken van de 'live' grafiek een goede aanvulling vinden:

- Mogeljkheid tot het bekijken van een grotere tijdspanne naast de standaard 30 minuten, bijvoorbeeld laatste uur, laatste 2 uur, enz. 
- Knop waarmee je na terugscrollen met 1 klik erop weer teruggaat naar de laatste metingen.

Ik weet niet of iets dergelijks te realiseren valt, maar wellicht is het een idee voor toekomstige releases.

dennissiemensma, 2017-03-19:
Bedankt voor je suggesties. Voor je eerste idee staat al een redesign voor later op de planning. Zie #235. Voor je tweede idee, daar is zeker wel iets voor te maken.

dennissiemensma, 2018-03-05:
Redesign van de GUI is momenteel helaas (realistisch gezien) niet mogelijk, dus ik laat dit nu voor wat het is.

closed 2018-03-05 by dennissiemensma