dsmrreader/dsmr-reader #286: Na gebruik admin-pagina's geen (eenvoudige) mogelijkheid voor terugkeren naar de site

michielvisser, 2017-03-19:
Bij het gebruik van de configuratie-pagina's (Django) is er na uitloggen geen mogelijkheid meer om met een link terug te keren naar het dashboard. Uiteraard kan je wel de URL opnieuw ingeven in de browser, maar wellicht is het mooier om bij uitloggen uit Django direct een redirect te geven naar de URL van het dashboard.

dennissiemensma, 2017-03-19:
Bedankt voor je suggestie, lijkt me zeker een goede verbetering.

ThinkPadNL, 2017-03-27:
Dit zit er al in toch? Of het moet onlangs geïmplementeerd zijn, maar ik zie in de menubalk bovenaan 'Bekijk site' en daarmee kom ik weer in dsmr-reader frontend terecht.

michielvisser, 2017-03-27:
De link zoals je meld zit er inderdaad in, maar alleen als je ingelogd bent op Django. Zodra je uitlogt verdwijnt de link en daar heb ik de verbetering op voorgesteld.

dennissiemensma, 2017-03-27:
Het gaat hier inderdaad alleen om het uitlogscherm. Alle schermen, templates, knoppen en linkjes in de Django Admin zijn gemaakt door Django zelf, dus ik moet tzt even kijken hoe ik dat erin krijg. 
Er is wel een mogelijkheid voor mij om de template te kopieren en een eigen versie te maken (met link), maar wellicht dat ik het simpeler oplos door een redirect te doen na het uitloggen.

dennissiemensma, 2017-09-11:
Het 'mooi' oplossen hiervan is te veel werk en vereist aanpassingen in de templates van Django. Dat doe ik liever niet.

Echter vind ik het wel een valide punt dat het lastig is om terug te keren naar de app, zonder zelf te goochelen met de URL. Daarom heb ik een wijziging klaarstaan die de admin-pagina's simpelweg in een nieuw tabblad opent. Wanneer iemand daarmee klaar is, kan die simpelweg de tab sluiten en is die waar die gebleven is.

closed 2017-09-11 by dennissiemensma