dsmrreader/dsmr-reader #287: Maandelijkse meterstanden

RemcoLeonard, 2017-03-20:
Hoi, 

Is het mogelijk om de maandelijkse meterstanden uit te lezen? 
Ik hou per maand mijn standen bij in een excel sheet voor mijn eigen gemoedsrust. 
Ik had voor ik DSMR gebruikte een scriptje draaien in Domoticz die me iedere 1e van de maand de standen pushte. 
In DSMR kan ik echter nergens vinden wat de meterstanden waren op een specifieke datum/dag. 

Thx, en ga ze door! Top applicatie!
Groet, 
Remco

pyrocumulus, 2017-03-20:
Lijkt me iets wat wel in het archief past, al kun je daar ook perioden selecteren (dus dat is wat lastiger wellicht). Wellicht de begin meterstand + eind meterstand weergeven voor de betreffende periode? Is wellicht wat overkill als je naar een enkele dag in het archief kijkt maar klopt wel altijd.

dennissiemensma, 2017-03-20:
Bedankt voor de vriendelijke woorden en jullie input. Er komt ondersteuning voor meterstanden in het archief. Dit is al eerder aangevraagd in #196 (ik houd waarschijnlijk gewoon middernacht aan) en zal ik oppakken in het verzamelticket #236. Ik sluit het huidige ticket gezien er al een voor is.

Echter, de gegevens staan momenteel al in de database, dus je bent vrij om die zelf er uit te halen met een query. In eerdergenoemd ticket en #227 staan wat voorbeelden. 
Mocht je er niet uitkomen qua query, laat het even weten want dan help ik je op weg. Zodat je in ieder geval die standen (voorlopig) handmatig kan uitlezen.

closed 2017-03-20 by dennissiemensma