dsmrreader/dsmr-reader #288: Feature request: Launch full screen on iOS device when opening from homescreen

emilkwaaitaal, 2017-03-20:
I'm launching DSMR Reader from my homescreen on iPhone and iPad (save webpage to homescreen in Safari browser). In 1.6 you added the possibility to add a homescreen icon (thanks!). There are also possibilities to launch full screen (without browser controls) to make it really look like a native iOS app.

See official documentation from Apple on how to get this working:
https://developer.apple.com/library/content/documentation/AppleApplications/Reference/SafariWebContent/ConfiguringWebApplications/ConfiguringWebApplications.html

Would be great if this can be implemented to further improve the user experience on iOS devices.

dennissiemensma, 2017-03-20:
Thanks, I might look into it some day later.

dennissiemensma, 2017-09-29:
I've added `<meta name="apple-mobile-web-app-capable" content="yes">` for the next release. Thank  you again for your suggestion.

closed 2017-09-29 by dennissiemensma