dsmrreader/dsmr-reader #290: Import Solarpanel data from e.g. pvoutput.org

helmo, 2017-03-25:
Similar to #9 it would also be cool to get data about generated power. 
Combining both data sources gives a better insight into usage patterns.

There are obviously more sources for such data, like directly from an inverter or the supplier's api site. But that will be different for every brand.

Docs about the api to use: http://pvoutput.org/help.html#api-getstatus

dennissiemensma, 2017-03-26:
Thank you for providing this idea. I'm not sure at the moment whether there will be any data imported from any external source in the future. The reasoning behind #9 is to keep as much non-DSMR data outside the applicatiion. 
I might reconsider it again after finishing #9, so I'll put this to the backlog for now.

dennissiemensma, 2017-09-11:
It has been a while, but I will (now) definitely not **import** data from pvoutput. Implementing the data export in #9 will be hard enough already to keep the data correctly... :]

closed 2017-09-11 by dennissiemensma