dsmrreader/dsmr-reader #291: API option to get status info

helmo, 2017-03-25:
It would be nice to get the info from the /status page via an API call.

This would make it easier to let a monitoring tool like Nagios check periodically to see if the service is still up and receiving data.

PS: Thanks for doing this... it works/looks pretty cool already.

dennissiemensma, 2017-03-26:
Thank you for your kind words and this excellent idea. I haven't thought about this yet, but it seems a nice addition for when I'm finishing #230 (working on that a.t.m.).

dennissiemensma, 2018-06-28:
Toegevoegd voor de volgende release: 

http://dsmr-reader.readthedocs.io/nl/development/api.html#get-application-status

closed 2018-06-28 by dennissiemensma

dennissiemensma, 2018-06-28:
Bedankt voor je scherpte, ik had de 1.20 branch nog niet op mijn eigen productieomgeving gecontroleerd. Bij deze is het gefixt.

dennissiemensma, 2018-07-04:
Uitgebracht in nieuwe release: https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.20.0