dsmrreader/dsmr-reader #292: Statuspagina uitbreiden: (Dropbox) backup + Mindergas upload

ThinkPadNL, 2017-03-27:
Is het een idee om op de statuspagina nog twee onderdelen toe te voegen:
- Backup
- Upload naar Mindergas
Dit om de status/verwerking van beide onderdelen te kunnen monitoren.

Die laatste had ik gisteren wat problemen mee, maar ik kwam er bij toeval achter (geen meterstand geupload op Mindergas.nl) en moest zelf in de logfiles van `dsmr-reader` spitten om erachter te komen wat er mis was (fout aan mijn kant). Dat laatste houd je toch denk ik, maar als er een blokje rood kleurt dan weet ik iig dat ik op onderzoek uit moet.

Als er een kopje bijkomt voor backup die kijkt of er in de afgelopen 24h een .tgz file is gemaakt + geupload naar Dropbox (indien ingeschakeld) en voor de Mindergas upload checken op de juiste responscode > groene omkadering/fill en anders rood?

dennissiemensma, 2017-03-27:
Goed idee! Sowieso wil ik uiteindelijk foutmeldingen en dergelijke wat proactiever opslaan en tonen (buiten de logfiles om). Ook in combinatie met je suggestie in #208, waar ik laatst al even mee bezig ben geweest maar nog wel wat werk in zit.

dennissiemensma, 2018-07-23:
Gaat mee met volgende release.

closed 2018-07-23 by dennissiemensma