dsmrreader/dsmr-reader #293: dsmr_webinterface                FATAL      Exited too quickly (process log may have details)

Jeltel, 2017-03-27:
Na het overzetten van de image op een andere SD-kaart, krijg ik de webinterface niet meer aan de praat.
Het ligt niet aan de nginx configuratie, die is nog hetzelfde en ook de standaard werkt helaas niet.

Waar kan ik de genoemde log vinden?

En gelukkig werkt de backend en logger nog wel. Dus in theorie zou ik een backup kunnen maken, de dsmr-reader folder weggooien en opnieuw beginnen toch?

dennissiemensma, 2017-03-27:
Ik heb [hier nog wat eerste hulp](http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html) neergezet. Je zou ook nog de Nginx logsfiles kunnen bekijken, die staan in `/var/log/nginx/`.

Als de rest van de applicatie wél draait, vermoed ik dat het een klein dingetje is. Ja, je kunt in principe de de hele `dsmr-reader` checkout kunnen weggooien (dubbelcheck even je database backups voordat je die ook samen weggooit met de checkout). Maar mogelijk is het werkend krijgen ook de moeite waard.

Je kunt ook nog kijken of de applicatie zelf een fout geeft:
```
sudo su - dsmr
./manage.py runserver

Er komt wat tekst te staan zonder foutmeldingen als het goed is. Afbreken kan met CTRL+C
```

Jeltel, 2017-05-05:
Wat het precies was is me niet duidelijk. Na updaten en opnieuw de Git clonen etc werkte het killen van een proces. Er was namelijk de melding:

`[2017-05-05 16:14:17 +0200] [4638] [INFO] Starting gunicorn 19.7.0

Error: Already running on PID 887 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)
`
Leek me dat een proces na herstarten van de PI er niet meer zou zijn. Wel dus. Handmatig PID 887 killen en daarna via de supervisorctl de webserver restarten lostte het op.

closed 2017-05-05 by Jeltel