dsmrreader/dsmr-reader #295: Question: Manual backup

janvanhelvoort, 2017-04-01:
Can i do a manual backup? Export on pi 1, import on pi 2.
I can use the daily backup. but than i have a gap of a day.

Jan.

WatskeBart, 2017-04-01:
You export your database from the RPi1 to a USB drive and import it back on the RPi2. See my ticket here #268 on how to make a backup.

janvanhelvoort, 2017-04-02:
Thank you.

Because of the 'day' in the name of the file. i thought it will only backup the 'sundays'.

dennissiemensma, 2017-04-02:
Hi, you can manually create a fresh backup using either WatskeBart's method or [one of these scripts](http://dsmr-reader.readthedocs.io/nl/development/application.html#data-preservation-backups), or using the commands in those scripts (as `root` or sudo-user):

* PostgreSQL: `sudo -u postgres pg_dump -d dsmrreader -v | gzip --fast > manual-backup.sql.gz`

* MySQL: `mysqldump --defaults-file=/etc/mysql/debian.cnf dsmrreader -v --extended-insert --lock-all-tables --hex-blob | gzip --fast > manual-backup.sql.gz`

Please let me know if you are running into any issues.

closed 2017-04-02 by dennissiemensma