dsmrreader/dsmr-reader #296: 3 fasen teruglevering

kadettilac, 2017-04-04:
Is het mogelijk om de 3 fasen verbruikgrafiek ook terugleverwaarden te laten zien? Eigenlijk negatieve waarden dus in dezelfde grafiek. Zo zie ik welke fasen er actief gebruiken en/of terugleveren. (3 fasen omvormer aanwezig)

BTW super project Dennis, m'n domoticz is al de deur uit :)

dennissiemensma, 2017-04-04:
Bedankt voor je ticket en goed om te horen dat je zo blij ermee bent. Ik zie dat DSMR 4+ telegrammen ook de teruglevering ondersteunen qua fasen, dus dat moet zeker mogelijk zijn.

Het zal wel enige tijd duren voordat dit mogelijk is, gezien er nog een hoop andere zaken zijn die voorrang krijgen. 

Mocht je het nu al willen weten, dan kun je dat indirect [terugzien in de telegrammen](https://github.com/dennissiemensma/dsmr-reader/blob/v1.6.0/dsmr_datalogger/tests/datalogger/test_landisgyr350_dsmr42.py#L48), en zijn het volgens mij deze regels (bij mij 0 kW):
```
...
"1-0:22.7.0(00.000*kW)\r\n",
"1-0:42.7.0(00.000*kW)\r\n",
"1-0:62.7.0(00.000*kW)\r\n",
...
```

Telegrammen zijn in te zien in de logs van de datalogger: 
`tail -f /var/log/supervisor/dsmr_datalogger.log`

kadettilac, 2017-04-05:
Thanks voor het opnemen in de backlog! Ik zal vast even kijken in de telegrammen en de feature request in de gaten houden.

dennissiemensma, 2018-07-06:
@kadettilac is dit nog steeds een actieve wens van je? Of had je destijds genoeg aan de telegrammen om te peilen hoe de teruglevering verdeeld wordt over je fasen?

kadettilac, 2018-07-09:
Hi Dennis,

Jazeker is dat nog een wens. Ik merk dat ik t dashboard regelmatig gebruik (bijna dagelijks), dus de optie om daarin gelijk de teruglevering per fase te zien zou top zijn. Vooral nu de zon vee schijnt (vandaag even wat minder:). Zo kan ik zien hoe de omvormer de stroom verdeeld en kan ik evt groepen op fases aanpassen om het meeste uit de dan opgewekte stroom te halen voor directe consumptie. 

Is dit iets waar je voor de volgende Release van plan bent aan te werken? Mij zou je er iig erg blij mee maken, en vast meer pv eigenaren. Ik maak al dankbaar gebruik van de pvoutput upload functionaliteit en dit zou dsmr reader nog beter maken. 

Als je hulp nodig hebt met testen hoor ik t graag. 

Fijne avond,

Caspar

dennissiemensma, 2018-07-09:
Bedankt voor je update. Ik wilde het even dubbelchecken, omdat er wel wat werk in zit en het een tijd geleden is dat je verzoek is geweest.

Ik weet niet of die voor de volgende release al komt, ik ben nu eerst bezig met het vervangen van de grafieken in het dashboard. Dat ga ik eerst afmaken.

kadettilac, 2018-07-09:
Dank voor je update, mooi dat t nog op je lijstje staat. Als ik tzt iets kan doen hoor ik t wel.

dennissiemensma, 2018-07-21:
@kadettilac ik heb inmiddels een eerste opzet werkend.

Echter, het wordt vrij druk om 6 lijnen in één grafiek te tonen:
[image]

Ik kan ze ook nog boven en onder tegen elkaar weergeven:
[image]

Wat denk jij dat handig is voor het aflezen? Wellicht dat @helmo hier ook nog een idee over heeft?

Ik kan ze overigens weer in **losse grafieken** zetten, maar dan is het verbruik en teruglevering niet perse synchroon aan elkaar bij scrollen/zoomen.

dennissiemensma, 2018-07-21:
Voor meer inspiratie zouden jullie trouwens ook nog de eChart-grafiek-voorbeelden kunnen bekijken: https://ecomfe.github.io/echarts-examples/public/index.html

kadettilac, 2018-07-21:
Cool! Ziet er goed uit, precies wat ik bedoel. 

Hoewel het er idd druk uit ziet op grafiek 1, laat het wel alle relevante info zien in 1 oogopslag. Dat heeft mijn voorkeur over boven/onder. Losse grafieken is de laatste optie wmb, om de reden die je geeft over al dan niet synchroon. 

Is het een optie om bijv teruglevering aan/uit te toggelen in de grafiek zodat je kan kiezen tussen meer of minder drukte op je grafiek?

dennissiemensma, 2018-07-21:
Bedankt voor je terugkoppeling. Een toggle-optie zou kunnen, maar ik zoek liever een optie die duidelijk genoeg is, mits dat mogelijk is.

Ik kan de lijnen ook nog **stapelen**, zodat de fases per type (+/-) nooit met de lijnen door elkaar heen lopen:
[image]

Eigenlijk net zoals het in de laatste release nu al werkt met die lijnen in de fasen-grafiek. Bij het **derde datapunt van links** hierboven kun je dat goed zien. Twee fasen zijn nul, maar blijven bij elkaar. 

Maakt dat het nóg wat duidelijker en minder druk? Sowieso zijn de fasen dan wat beter te onderscheiden zonder mengkleur.

dennissiemensma, 2018-07-21:
@kadettilac je kunt hem overigens nu zelf ook uitproberen:

```
sudo su - dsmr
git fetch
git checkout -b 296-multi-phase-return origin/296-multi-phase-return
./deploy.sh
```

De willekeurige grafieken, die ik gebruik voor testen, zijn namelijk ver van de werkelijkheid. Er zou in het echt veel minder wisseling moeten zijn tussen (netto) verbruiken en terugleveren. Dan kun je wellicht beter beoordelen of het zo lekker leest of juist niet.

Zoals deze grafiek in een ander ticket: https://user-images.githubusercontent.com/28338927/42986508-14aa8180-8bf6-11e8-90b0-1e29b331a59e.JPG

dennissiemensma, 2018-07-21:
Fijn, bedankt voor je snelle uitproberen! Dan laat ik het zo en neem ik het op die manier mee in de volgende release.

Voor nu kun je de versie/branch blijven gebruiken zoals die nu draait (het bevat al alles voor de volgende release). 
Wanneer `v1.22` uitkomt kun je t.z.t. weer terug naar de standaard-aftakking van de code met:
```
sudo su - dsmr
git checkout master
./deploy.sh
```

kadettilac, 2018-07-21:
Heb m geïnstalleerd maar zie nog geen terugleverdata per fase (onderste grafiek), neem aan dat dat is omdat ie die data niet met terugwerkende kracht indexeert) Wel zie ik t terugleveren als totaal (bovenste grafiek). Zal morgen kijken als de zon schijnt :)

Assigned #296 to @kadettilac.

kadettilac, 2018-07-21:
Ha, we postten net langs elkaar. Ik heb m dus eerst ge-update van 1.19 naar 1.21 een half uur geleden en tien minuten later naar 1.22b4 toen ik je bericht las. 

Ik heb dus nog geen uitsluitsel of t goed werkt, even wachten tot de zon schijnt morgen. Nu zie ik de grafiek iig nog niet correct.

kadettilac, 2018-07-21:
Zo ziet de grafiek van vandaag eruit. Veel 0 watt verbruik ipv opbrengst:

[image]

dennissiemensma, 2018-07-21:
Bedankt voor je opheldering!

Ik heb net nog even gecheckt hoe het komt dat die niets toont. Dat lijkt een feature of bijeffect van de nieuwe grafieken lib. Als er geen data is of alle data is 0, dan toont die de dataserie uberhaupt niet (wat opzich wel wenselijk is in mijn situatie zonder panelen, maar niet perse in jouw situatie). 
Zodra ik ook maar 1 Watt teruglevering simuleer, tekent hij de dataserie.Dus als het goed is zie je inderdaad vanaf morgenochtend een grafiek met alle data. 

En aangezien de grafiek altijd de afgelopen 24 uur toont, zou daarmee gegarandeerd moeten zijn dat er altijd (op een moment van de dag) in de grafiek teruglevering is geweest. En dan is dit een eenmalig issue waar men na een upgrade voor een dagdeel last van hebben.

Hoe dan ook hoor ik dan graag van je of het morgen helemaal naar wens werkt!

kadettilac, 2018-07-22:
Het werkt :)

dennissiemensma, 2018-07-22:
Fijn om te horen! Ik neem aan dat deze versie van de grafiek dan duidelijk genoeg is qua aflezen?

dennissiemensma, 2018-07-22:
Gemerged naar `development` voor volgende release.

closed 2018-07-22 by dennissiemensma

dennissiemensma, 2018-07-22:
Uitgebracht in `v1.22`.

helmo, 2018-07-23:
@dennissiemensma Dank ... dat ziet er al mooi uit.

Wat de 'drukte' betreft, als je teruglevert op een fase zal de + lijn daarvan 0 zijn. Althans ik kan me zo geen situatie voorstellen waarbij + en - tegelijk een waarde anders dan 0 hebben.

Als je nou een lijn per fase zou doen, die bij teruglevering onder 0 komt.... het is toch al een negatief getal.

Als voorbewerking moet je dan alleen de waarde kiezen die niet 0 is.

Zo ziet het er uit na de bovenstaande commit. helemaal tevreden ben ik ernog niet mee, maar wie weet inspireert het jullie.
[image]

kadettilac, 2018-07-23:
Toch wel plus en min op L1 bij mij, zie hier een screenshot van vandaag:

Weliswaar zeer sporadisch, maar af en toe gebeurt t als een grote verbruiker inschakelt met een hogere vraag dan de levering op die fase. T is een momentopname, want ik heb dsmr die elke 10 seconden een telegram stuurt, niet de nieuwst die dat elke seconde doet.

kadettilac, 2018-07-28:
@dennissiemensma Screenshot nogmaals toegevoegd. Heb t nu al een paar keer zien gebeuren.

Twee vragen nog:
1) Ik kom niet terug naar de  master branch. Wat doe ik fout?; 
`pi@raspberrypi:~ $ sudo su - dsmr
dsmr@raspberrypi:~ $ git checkout master
fatal: Not a git repository (or any of the parent directories): .git`
2) Ik ben als een kind zo blij met de nieuwe functionaliteit! Hoe kan ik je hiervoor bedanken? Heb je een 'buy-me-a-beer/coffee' paypal/tikkie link?

dennissiemensma, 2018-07-29:
Nogmaals bedankt voor je bijdrage!