dsmrreader/dsmr-reader #297: Water meter support

kadettilac, 2017-04-04:
Is het mogelijk om support voor een s0 pulse watermeter toe te voegen? Dit is de enige verbruiker in huis die nu niet gemeten wordt en Domoticz aan t werk houdt. 

Zoals deze meter (ltron cyble v2): http://www.homewizard.nl/pulse-sensor-voor-watermeter.html 

Domoticz meet die nu via de s0 meter van http://www.smartmeterdashboard.nl/webshop, S0 Pulse Counter Module (5 Poorten) obv arduino. Docs hier: https://docs.google.com/viewer?a=v&pid=sites&srcid=ZGVmYXVsdGRvbWFpbnxudGE4MTMwcDFzbWFydG1ldGVyfGd4OmYzMjc0ZTFjMzFkYzYwMw

dennissiemensma, 2017-04-04:
Bedankt voor je suggestie. Ondanks dat ik zeker begrijp dat er een goede usecase voor is, is de grens van dit project de slimme meter zelf, dus zal ik geen andere bronnen inlezen die niet DSMR-gerelateerd zijn. Dit geldt ook voor eerdere verzoeken van anderen m.b.t. zonnepanelen/omvormers [#145] en een Youless-apparaat [#263]. 

De reden hiervoor is dat het voor mij moeilijk te testen is én dat het (onnodig) complexiteit toevoegt aan het project. Bijvoorbeeld het temperatuur meten via Buienradar had ik achteraf gezien beter niet in het project willen bouwen.
 Anderzijds staat het project zeker open voor het exporteren van de data 'in' dit project naar externe bronnen (MinderGas, PVOutput [#9], CSV, API [#230] en mogelijk zelfs MQTT later), gezien dit jullie 'eigen' gegevens zijn en ik de bruikbaarheid daarvan zoveel mogelijk ondersteun. 

Hopelijk begrijp je de reden, ondanks dat ik je graag een andere oplossing had aangeboden die ervoor zorgt dat je geen extra software hoeft te draaien voor je watermeter.

closed 2017-04-04 by dennissiemensma

kadettilac, 2017-04-05:
Dat houdt de scope lekker duidelijk en de creep minimaal, ik begrijp t zeker. Ik ga op zoek naar een andere oplossing voor water maar zal DSMR Reader zeker blijven gebruiken voor elektra en gas. Thanks!

Martinvdm, 2018-07-15:
@kadattilac heb je het alternatief nog gevonden? Ik heb exact dezelfde watermet en sensor.

kadettilac, 2018-07-16:
@Martinvdm Ik heb van de nood een deugd gemaakt en de P1 kabel gesplitst naar 2 Pi's. 1 met DSMR Reader erop die het water niet kan meten, 1 met Domoticz erop die water en stroomverbruik/opwekking registreert. Zo zijn ze backup voor elkaar qua energiemeting mocht 1 er de brui aan geven. Ik gebruik DSMR reader voor de dagelijkse metingen vanwege de veel fijnere interface.

Martinvdm, 2018-07-16:
Splitsen? Je water meter is toch een wezenlijk andere usb kabel?
Maargoed begrijp dat je domoticz hiervoor hebt draaien.

kadettilac, 2018-07-16:
Klopt, de watermeter is een andere kabel. Ik heb niet de watermeter maar de p1 kabel gesplitst. 

Omdat Dsmr reader (voor energie) en domoticz (voor water en als energie backup) niet op dezelfde pi kunnen draaien ivm de telegrammen die maar door 1 applicatie kunnen worden opgepikt, draai ik 2 pi’s. 

Als ik domoticz alleen water zou laten doen en dsmr reader de energie kan t wel op 1 pi. Helaas dus niet in 1 GUI, tenzij Dennis met z’n hand over z’n hart strijkt en de feature request reopend :)

bartbakels, 2020-08-28:
wellicht toch een idee om de watermeter en of andere meter toetvoegen als een value, die meegenomen wordt in de berekening. En om het simpel te houden deze waarde te lezen vanaf een MQTT broker. Zo hoef je ook geen interfaces bij tehouden. 

Daarnaast thnx voor deze super oplossing :D