dsmrreader/dsmr-reader #302: Archief 2016?

Calimerorulez, 2017-04-18:
Hi Dennis,

Ik draai de meest recente versie 1.6.1, en ik probeer het archief van 2016 te raadplegen, alleen is dat niet klikbaar in 'Archief'. Terugbladeren lukt ook niet helaas.

Wat zie ik over het hoofd?

Calimerorulez, 2017-04-18:
[image]

Bijgaand een screenshot. F12 geeft geen errors in de console.

[image]

Detail per dag gaat ook niet verder terug dan 01-01-2017.

dennissiemensma, 2017-04-18:
Bedankt voor alle info. Op je tweede screenshot lijkt zelfs alles vóór za. 7 januari grijs (dus niet-selecteerbaar). Klopt dat?

Je zou nog even kunnen dubbelchecken welke informatie in de database staat. Ik heb wel een vermoeden waar het mis gaat. Kun je dit uitvoeren en hier plakken? Het geeft alleen ID's en dagen weer, in twee verschillende volgordes:

*Op ID*
```
sudo -u postgres psql dsmrreader -c "select id, day from dsmr_stats_daystatistics order by id limit 10;"
```

*Op dag*
```
sudo -u postgres psql dsmrreader -c "select id, day from dsmr_stats_daystatistics order by day limit 10;"
```

Calimerorulez, 2017-04-18:
Vóór 7 januari is inderdaad niet selecteerbaar.

Output op ID:

 id |    day
----+------------
  1 | 2017-01-07
  2 | 2017-01-08
  3 | 2017-01-09
  4 | 2017-01-10
  5 | 2017-01-11
  6 | 2017-01-12
  7 | 2017-01-13
  8 | 2017-01-14
  9 | 2017-01-15
 10 | 2017-01-16
(10 rows)

Output op dag:

 id |    day
----+------------
  1 | 2017-01-07
  2 | 2017-01-08
  3 | 2017-01-09
  4 | 2017-01-10
  5 | 2017-01-11
  6 | 2017-01-12
  7 | 2017-01-13
  8 | 2017-01-14
  9 | 2017-01-15
 10 | 2017-01-16
(10 rows)

dennissiemensma, 2017-04-18:
Het lijkt erop dat er uberhaupt geen (dag)gegevens zijn van vóór 7 januari. Vandaar dat de interface dat daar ook denkt.

Kun je kijken wanneer je eerste meting is?

```
sudo -u postgres psql dsmrreader -c "select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp limit 5;"
```

Calimerorulez, 2017-04-18:
Dan val ik denk ik door de mand:

 id |       timestamp
----+------------------------
  1 | 2017-01-07 21:53:56+01
  2 | 2017-01-07 21:55:57+01
  3 | 2017-01-07 21:56:07+01
  4 | 2017-01-07 21:56:17+01
  5 | 2017-01-07 21:57:37+01
(5 rows)

Ik heb dsmr-reader al maanden (ook in 2016) draaien zonder onderbrekingen...

Sorry voor de verwarring... verklaren kan ik het niet...

dennissiemensma, 2017-04-18:
Ik denk dan dat de gegevens weg zijn. Alleen heb ik er ook geen verklaring voor. De applicatie gooit zelf in ieder geval geen data weg, sowieso niet totdat #72 geimplementeerd is (en dat duurt nog wel even).

Je kunt twee andere tabellen controleren, maar ik vermoed dat die hetzelfde gaan zeggen:

```
sudo -u postgres psql dsmrreader -c "select id, read_at from dsmr_consumption_electricityconsumption order by read_at limit 5;"
```

```
sudo -u postgres psql dsmrreader -c "select id, read_at from dsmr_consumption_gasconsumption order by read_at limit 5;"
```

Calimerorulez, 2017-04-18:
Klopt, allebei 7 januari... Dank voor je hulp.

closed 2017-04-18 by Calimerorulez

dennissiemensma, 2017-04-18:
Gezien de ID's ook allemaal bij 1 beginnen, lijkt het er in ieder geval op dat er vóór die tijd uberhaupt niets opgeslagen is.  Mocht je nog ergens anders hulp bij nodig hebben of iets vreemds hebben gezien, laat maar weten! 

Sowieso bedankt voor het melden want hierdoor kwam een bug in #303 aan het licht. :]

Calimerorulez, 2017-04-18:
Gelukkig :-))