dsmrreader/dsmr-reader #304: Feature request: Support for additional stream(s) from KWh meters to separete consumption from production

emilkwaaitaal, 2017-04-23:
I have solar panels and would like to to see my actual energy consumption separately from solar panel production. I can either read the production from my invertor or from a separate KWh meter connected to the invertor.

dennissiemensma, 2017-04-23:
Hi, thank you for creating a ticket. Could you elaborate bit more about what you are requesting exactly? I'm not sure whether I understand your request correctly.

emilkwaaitaal, 2017-04-23:
Thanks. Happy to elaborate on this. 

I currently cannot easily analyse my actual energy consumption as the production from solar varies over time. It would be great if there is a possibility to display production KWh and consumption KWh in separate charts. This would mean that we need a second input (in addition from P1 DSMR reader input). Input is KWh meter (S0 connection) coming from invertor solar system. This input will by used as input for the production chart. Input for consumption chart will then be P1 port KWh + KWh meter from solar system.

Example: P1 KWh for a specific day is 10 KWh. Production KWh meter is 15 KWh. Consumption is then 25 KWh and Production 15 KWh.

HomeWizards EnergyLink already supports this via a second input in addition to P1 (called S0). And there are also Raspberry Pi solutions (with Arduino) supporting this.

Hope this clarifies!

dennissiemensma, 2017-04-23:
Thank you for your detailed explanation. These requests (or similar ones) have been made in the past as well: #297, #255, #209 and #145.

Unfortunately, I cannot provide support for other devices of input, as that is beyond the scope of the project. I do understand it would help some users, but it will make this project even more complicated than it already is at the moment. And supporting all kind of different devices/brands will be hard as well, not to mention testing it. So that's why the project scope is limited to the smart meter's DSMR protocol.

However, I do actively support any tools to have you *extract your DSMR data* to external parties or services (such as CSV and MinderGas.nl). 
So, the next feature that might actually help you will be #9 (PVOutput.org). There are some solar panel converters natively supporting PVOutput support, allowing you to actually see your generated power. Adding DSMR data on top of that, will probably give you the information you are looking for.

I will start working on PVOuput support whenever I've finished another data feature #230 (API support).

closed 2017-04-23 by dennissiemensma